AllenCellModeling/aicsimageio #267: Development set up: aiobotocore 1.3.1 requires botocore<1.20.50,>=1.20.49

rcasero, 2021-07-02:
## System and Software
* aicsimageio Version: 4.0.2
* Python Version: 3.8 and 3.9
* Operating System: Ubuntu linux Focal

## Description

Local environment set up with

```
conda create -y --name aicsimageio python=3.8
conda activate aicsimageio
conda install python=3.9
```

Following the instructions in https://allencellmodeling.github.io/aicsimageio/CONTRIBUTING.html#

```
git clone https://rcasero@github.com/rcasero/aicsimageio.git
cd aicsimageio/
pip install -e .[dev]
pip install quilt3
```
Gives the error

```
Installing collected packages: Werkzeug, itsdangerous, botocore, s3transfer, pyrsistent, flask, tenacity, requests-futures, jsonschema, jsonlines, flask-json, flask-cors, boto3, aws-requests-auth, quilt3
  Attempting uninstall: botocore
    Found existing installation: botocore 1.20.49
    Uninstalling botocore-1.20.49:
      Successfully uninstalled botocore-1.20.49
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
aiobotocore 1.3.1 requires botocore<1.20.50,>=1.20.49, but you have botocore 1.20.104 which is incompatible.
Successfully installed Werkzeug-2.0.1 aws-requests-auth-0.4.3 boto3-1.17.104 botocore-1.20.104 flask-2.0.1 flask-cors-3.0.10 flask-json-0.3.4 itsdangerous-2.0.1 jsonlines-1.2.0 jsonschema-3.2.0 pyrsistent-0.18.0 quilt3-3.4.0 requests-futures-1.0.0 s3transfer-0.4.2 tenacity-7.0.0
```

When 

## Expected Behavior
No errors

## Reproduction
See above

## Environment
*Any additional information about your environment*

JacksonMaxfield, 2021-07-02:
Hah. Yea this conflict has just always happened. I should add updated docs or something about this.

Not to worry, everything installed just fine for cloning the data.

The updated docs should say like:

* install quilt3
* download test resources
* install dev dependencies again to remove the conflicting boto

boto / boto3 / botocore / aibotocore continues to be the most annoying dependency of any library I have ever worked with.

JacksonMaxfield, 2021-07-02:
Huh.... apparently we can resolve this by adding the `[boto3]` extra to `s3fs`: https://github.com/dask/s3fs/issues/357

So the setup.py could now include both `s3fs[boto3]` and `quilt3` in the same dev deps. I will look into this this weekend. Unless you want to try it during your PR @rcasero

rcasero, 2021-07-03:
Thanks for the hints, @JacksonMaxfield . I've fixed the issue (and the documentation) in this commit

https://github.com/rcasero/aicsimageio/commit/134a2863fcef5e8c3980d518ad2d5300b38a6450

that will be in my PR

JacksonMaxfield, 2021-07-04:
Wonderful! Thank you so much @rcasero

rcasero, 2021-07-04:
Actually, there's still some problem. Looking into it

```
tox
py38 create: /home/rcasero/Software/aicsimageio/.tox/py38
py38 installdeps: .[test]
ERROR: invocation failed (exit code 2), logfile: /home/rcasero/Software/aicsimageio/.tox/py38/log/py38-1.log
================================== log start ===================================
Processing /home/rcasero/Software/aicsimageio
  DEPRECATION: A future pip version will change local packages to be built in-place without first copying to a temporary directory. We recommend you use --use-feature=in-tree-build to test your packages with this new behavior before it becomes the default.
   pip 21.3 will remove support for this functionality. You can find discussion regarding this at https://github.com/pypa/pip/issues/7555.
Collecting dask[array]>=2021.4.1
  Using cached dask-2021.6.2-py3-none-any.whl (973 kB)
Collecting fsspec>=2021.4.0
  Using cached fsspec-2021.6.1-py3-none-any.whl (115 kB)
Collecting imagecodecs>=2020.5.30
  Using cached imagecodecs-2021.6.8-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (32.0 MB)
Collecting numpy~=1.16
  Using cached numpy-1.21.0-cp38-cp38-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (15.7 MB)
Collecting ome-types~=0.2.4
  Using cached ome_types-0.2.6-py3-none-any.whl (137 kB)
Collecting tifffile>=2021.2.1
  Using cached tifffile-2021.7.2-py3-none-any.whl (169 kB)
Collecting toolz~=0.11.0
  Using cached toolz-0.11.1-py3-none-any.whl (55 kB)
Collecting xarray~=0.16.1
  Using cached xarray-0.16.2-py3-none-any.whl (736 kB)
Collecting xmlschema
  Using cached xmlschema-1.6.4-py3-none-any.whl (254 kB)
Collecting zarr~=2.6.1
  Using cached zarr-2.6.1-py3-none-any.whl (132 kB)
Collecting imageio[ffmpeg]~=2.9.0
  Using cached imageio-2.9.0-py3-none-any.whl (3.3 MB)
Collecting readlif~=0.6.1
  Using cached readlif-0.6.1-py3-none-any.whl (23 kB)
Collecting aicspylibczi~=3.0.2
  Using cached aicspylibczi-3.0.2-cp38-cp38-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (625 kB)
Collecting black>=19.10b0
  Using cached black-21.6b0-py3-none-any.whl (140 kB)
Collecting codecov>=2.1.4
  Using cached codecov-2.1.11-py2.py3-none-any.whl (16 kB)
Collecting distributed>=2021.4.1
  Using cached distributed-2021.6.2-py3-none-any.whl (722 kB)
Collecting docutils<0.16,>=0.10
  Using cached docutils-0.15.2-py3-none-any.whl (547 kB)
Collecting flake8>=3.8.3
  Using cached flake8-3.9.2-py2.py3-none-any.whl (73 kB)
Collecting flake8-debugger>=3.2.1
  Using cached flake8_debugger-4.0.0-py3-none-any.whl (5.0 kB)
Collecting isort>=5.7.0
  Using cached isort-5.9.1-py3-none-any.whl (105 kB)
Collecting mypy>=0.800
  Using cached mypy-0.910-cp38-cp38-manylinux2010_x86_64.whl (22.8 MB)
Collecting psutil>=5.7.0
  Using cached psutil-5.8.0-cp38-cp38-manylinux2010_x86_64.whl (296 kB)
Collecting pytest>=5.4.3
  Using cached pytest-6.2.4-py3-none-any.whl (280 kB)
Collecting pytest-cov>=2.9.0
  Using cached pytest_cov-2.12.1-py2.py3-none-any.whl (20 kB)
Collecting pytest-raises>=0.11
  Using cached pytest_raises-0.11-py2.py3-none-any.whl (9.4 kB)
Collecting s3fs[boto3]>=2021.4.0
  Using cached s3fs-2021.6.1-py3-none-any.whl (25 kB)
Collecting quilt3>=3.4.0
  Using cached quilt3-3.4.0-py3-none-any.whl (111 kB)
Collecting toml>=0.10.1
  Using cached toml-0.10.2-py2.py3-none-any.whl (16 kB)
Collecting click>=7.1.2
  Using cached click-8.0.1-py3-none-any.whl (97 kB)
Collecting appdirs
[... 251 lines cut ...]
  Downloading Pillow-8.1.2-cp38-cp38-manylinux1_x86_64.whl (2.2 MB)
  Downloading Pillow-8.1.1-cp38-cp38-manylinux1_x86_64.whl (2.2 MB)
ERROR: Exception:
Traceback (most recent call last):
  File "/home/rcasero/Software/aicsimageio/.tox/py38/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", line 180, in _main
    status = self.run(options, args)
  File "/home/rcasero/Software/aicsimageio/.tox/py38/lib/python3.8/site-packages/pip/_internal/cli/req_command.py", line 204, in wrapper
    return func(self, options, args)
  File "/home/rcasero/Software/aicsimageio/.tox/py38/lib/python3.8/site-packages/pip/_internal/commands/install.py", line 318, in run
    requirement_set = resolver.resolve(
  File "/home/rcasero/Software/aicsimageio/.tox/py38/lib/python3.8/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 127, in resolve
    result = self._result = resolver.resolve(
  File "/home/rcasero/Software/aicsimageio/.tox/py38/lib/python3.8/site-packages/pip/_vendor/resolvelib/resolvers.py", line 473, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
  File "/home/rcasero/Software/aicsimageio/.tox/py38/lib/python3.8/site-packages/pip/_vendor/resolvelib/resolvers.py", line 384, in resolve
    raise ResolutionTooDeep(max_rounds)
pip._vendor.resolvelib.resolvers.ResolutionTooDeep: 2000000
WARNING: You are using pip version 21.1.2; however, version 21.1.3 is available.
You should consider upgrading via the '/home/rcasero/Software/aicsimageio/.tox/py38/bin/python -m pip install --upgrade pip' command.

=================================== log end ====================================
ERROR: could not install deps [.[test]]; v = InvocationError("/home/rcasero/Software/aicsimageio/.tox/py38/bin/python -m pip install '.[test]'", 2)
py39 create: /home/rcasero/Software/aicsimageio/.tox/py39
py39 installdeps: .[test]
```

rcasero, 2021-07-04:
@JacksonMaxfield Tests fail. I'm going to do this step by step.

Going back to the original `setup.py`, where the only dependency is 

```
test_requirements = [
    "s3fs>=0.4.2",
```

We run things in this order:

```
conda create -y --name aicsimageio python=3.8
conda activate aicsimageio
conda install python=3.9
pip install quilt3
python scripts/download_test_resources.py
pip install -e .[dev]
```

This produces a set up with what looks like a non-critical error

```
Installing collected packages: pycparser, typing-extensions, toolz, pyparsing, multidict, locket, cffi, zipp, yarl, webencodings, pytz, pydantic, partd, packaging, numpy, jeepney, ipython-genutils, fsspec, email-validator, elementpath, cryptography, cloudpickle, async-timeout, xmlschema, wrapt, wcwidth, traitlets, toml, sphinxcontrib-serializinghtml, sphinxcontrib-qthelp, sphinxcontrib-jsmath, sphinxcontrib-htmlhelp, sphinxcontrib-devhelp, sphinxcontrib-applehelp, snowballstemmer, SecretStorage, pygments, pyflakes, pycodestyle, py, ptyprocess, pluggy, Pint, pillow, parso, pandas, numcodecs, mccabe, iniconfig, importlib-metadata, imagesize, heapdict, filelock, fasteners, docutils, distlib, dask, botocore, bleach, babel, asciitree, alabaster, aioitertools, aiohttp, zict, zarr, xarray, virtualenv, tornado, tifffile, tblib, Sphinx, sortedcontainers, rfc3986, requests-toolbelt, regex, readme-renderer, pytest, psutil, prompt-toolkit, pkginfo, pickleshare, pexpect, pathspec, ome-types, mypy-extensions, msgpack, mistune, matplotlib-inline, keyring, jedi, imageio-ffmpeg, imageio, imagecodecs, flake8, decorator, coverage, colorama, backcall, aiobotocore, twine, tox, sphinx-rtd-theme, s3fs, readlif, pytest-runner, pytest-raises, pytest-cov, mypy, m2r2, isort, ipython, gitchangelog, flake8-debugger, distributed, codecov, bump2version, black, asv, aicspylibczi, aicsimageio
  Attempting uninstall: botocore
    Found existing installation: botocore 1.20.105
    Uninstalling botocore-1.20.105:
      Successfully uninstalled botocore-1.20.105
  Running setup.py develop for aicsimageio
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
boto3 1.17.105 requires botocore<1.21.0,>=1.20.105, but you have botocore 1.20.49 which is incompatible.
Successfully installed Pint-0.17 SecretStorage-3.3.1 Sphinx-4.0.2 aicsimageio-4.0.2 aicspylibczi-3.0.2 aiobotocore-1.3.1 aiohttp-3.7.4.post0 aioitertools-0.7.1 alabaster-0.7.12 asciitree-0.3.3 asv-0.4.2 async-timeout-3.0.1 babel-2.9.1 backcall-0.2.0 black-21.6b0 bleach-3.3.0 botocore-1.20.49 bump2version-1.0.1 cffi-1.14.5 cloudpickle-1.6.0 codecov-2.1.11 colorama-0.4.4 coverage-5.5 cryptography-3.4.7 dask-2021.6.2 decorator-5.0.9 distlib-0.3.2 distributed-2021.6.2 docutils-0.15.2 elementpath-2.2.3 email-validator-1.1.3 fasteners-0.16.3 filelock-3.0.12 flake8-3.9.2 flake8-debugger-4.0.0 fsspec-2021.6.1 gitchangelog-3.0.4 heapdict-1.0.1 imagecodecs-2021.6.8 imageio-2.9.0 imageio-ffmpeg-0.4.4 imagesize-1.2.0 importlib-metadata-4.6.0 iniconfig-1.1.1 ipython-7.25.0 ipython-genutils-0.2.0 isort-5.9.1 jedi-0.18.0 jeepney-0.6.0 keyring-23.0.1 locket-0.2.1 m2r2-0.2.8 matplotlib-inline-0.1.2 mccabe-0.6.1 mistune-0.8.4 msgpack-1.0.2 multidict-5.1.0 mypy-0.910 mypy-extensions-0.4.3 numcodecs-0.8.0 numpy-1.21.0 ome-types-0.2.6 packaging-21.0 pandas-1.3.0 parso-0.8.2 partd-1.2.0 pathspec-0.8.1 pexpect-4.8.0 pickleshare-0.7.5 pillow-8.3.0 pkginfo-1.7.0 pluggy-0.13.1 prompt-toolkit-3.0.19 psutil-5.8.0 ptyprocess-0.7.0 py-1.10.0 pycodestyle-2.7.0 pycparser-2.20 pydantic-1.8.2 pyflakes-2.3.1 pygments-2.9.0 pyparsing-2.4.7 pytest-6.2.4 pytest-cov-2.12.1 pytest-raises-0.11 pytest-runner-5.3.1 pytz-2021.1 readlif-0.6.1 readme-renderer-29.0 regex-2021.7.1 requests-toolbelt-0.9.1 rfc3986-1.5.0 s3fs-2021.6.1 snowballstemmer-2.1.0 sortedcontainers-2.4.0 sphinx-rtd-theme-0.5.2 sphinxcontrib-applehelp-1.0.2 sphinxcontrib-devhelp-1.0.2 sphinxcontrib-htmlhelp-2.0.0 sphinxcontrib-jsmath-1.0.1 sphinxcontrib-qthelp-1.0.3 sphinxcontrib-serializinghtml-1.1.5 tblib-1.7.0 tifffile-2021.7.2 toml-0.10.2 toolz-0.11.1 tornado-6.1 tox-3.23.1 traitlets-5.0.5 twine-3.4.1 typing-extensions-3.10.0.0 virtualenv-20.4.7 wcwidth-0.2.5 webencodings-0.5.1 wrapt-1.12.1 xarray-0.16.2 xmlschema-1.4.1 yarl-1.6.3 zarr-2.6.1 zict-2.0.0 zipp-3.5.0
```

JacksonMaxfield, 2021-07-04:
Ugh. Yeah for now I would just update documentation.

rcasero, 2021-07-04:
@JacksonMaxfield  Some of the tests that fail when doing `make build` in the current main project, without any changes:

```
tox
py38 create: /home/rcasero/Software/aicsimageio/.tox/py38
py38 installdeps: .[test]
py38 installed: aicsimageio @ file:///home/rcasero/Software/aicsimageio,aicspylibczi==3.0.2,aiobotocore==1.3.1,aiohttp==3.7.4.post0,aioitertools==0.7.1,appdirs==1.4.4,asciitree==
0.3.3,async-timeout==3.0.1,attrs==21.2.0,black==21.6b0,botocore==1.20.49,certifi==2021.5.30,chardet==4.0.0,click==8.0.1,cloudpickle==1.6.0,codecov==2.1.11,coverage==5.5,dask==202
1.6.2,distributed==2021.6.2,dnspython==2.1.0,docutils==0.15.2,elementpath==2.2.3,email-validator==1.1.3,fasteners==0.16.3,flake8==3.9.2,flake8-debugger==4.0.0,fsspec==2021.6.1,He
apDict==1.0.1,idna==2.10,imagecodecs==2021.6.8,imageio==2.9.0,imageio-ffmpeg==0.4.4,iniconfig==1.1.1,isort==5.9.1,jmespath==0.10.0,locket==0.2.1,mccabe==0.6.1,msgpack==1.0.2,mult
idict==5.1.0,mypy==0.910,mypy-extensions==0.4.3,numcodecs==0.8.0,numpy==1.21.0,ome-types==0.2.6,packaging==21.0,pandas==1.3.0,partd==1.2.0,pathspec==0.8.1,Pillow==8.3.0,Pint==0.1
7,pluggy==0.13.1,psutil==5.8.0,py==1.10.0,pycodestyle==2.7.0,pydantic==1.8.2,pyflakes==2.3.1,pyparsing==2.4.7,pytest==6.2.4,pytest-cov==2.12.1,pytest-raises==0.11,python-dateutil
==2.8.1,pytz==2021.1,PyYAML==5.4.1,readlif==0.6.1,regex==2021.7.1,requests==2.25.1,s3fs==2021.6.1,six==1.16.0,sortedcontainers==2.4.0,tblib==1.7.0,tifffile==2021.7.2,toml==0.10.2
,toolz==0.11.1,tornado==6.1,typing-extensions==3.10.0.0,urllib3==1.26.6,wrapt==1.12.1,xarray==0.16.2,xmlschema==1.4.1,yarl==1.6.3,zarr==2.6.1,zict==2.0.0
py38 run-test-pre: PYTHONHASHSEED='2685928569'
py38 run-test: commands[0] | pytest --basetemp=/home/rcasero/Software/aicsimageio/.tox/py38/tmp --cov-report html --cov=aicsimageio aicsimageio/tests/
============================= test session starts ==============================
platform linux -- Python 3.8.10, pytest-6.2.4, py-1.10.0, pluggy-0.13.1
cachedir: .tox/py38/.pytest_cache
rootdir: /home/rcasero/Software/aicsimageio, configfile: setup.cfg
plugins: cov-2.12.1, raises-0.11
collected 542 items

aicsimageio/tests/test_aics_image.py ................................... [  6%]
......................................................                   [ 16%]
aicsimageio/tests/test_transforms.py ................................... [ 22%]
.................................................                        [ 31%]
aicsimageio/tests/readers/test_array_like_reader.py .................... [ 35%]
..................................................                       [ 44%]
aicsimageio/tests/readers/test_czi_reader.py ......x..x........          [ 48%]
aicsimageio/tests/readers/test_default_reader.py FF.F.F.F.F.F.F.F.       [ 51%]
aicsimageio/tests/readers/test_lif_reader.py .F.F.F.F.F................. [ 56%]
....                                                                     [ 57%]
aicsimageio/tests/readers/test_ome_tiff_reader.py .F.F.F.F.F.F.F.F.F.F.. [ 61%]
..F.F...F.F.F.F.F.F.F.F.                                                 [ 65%]
aicsimageio/tests/readers/test_tiff_reader.py .F.F.F.F.F.F.F.F.F.F.F.F.F [ 70%]
.F.F.......                                                              [ 72%]
aicsimageio/tests/utils/test_io_utils.py .........F.F                    [ 74%]
aicsimageio/tests/writers/test_ome_tiff_writer.py ...................... [ 78%]
................................................                         [ 87%]
aicsimageio/tests/writers/test_timeseries_writer.py ..FF..FF............ [ 91%]
............                                                             [ 93%]
aicsimageio/tests/writers/test_two_d_writer.py ......................... [ 97%]
...........                                                              [100%]
```

Testing takes very long. I'm attaching the `make `

rcasero, 2021-07-04:
[make_build.log](https://github.com/AllenCellModeling/aicsimageio/files/6760287/make_build.log)

PR #270 (2021-07-05): @rcasero-feature/add-python-3.7-support
## Description

See #269 and #268 for prior PRs during the process.

This adds Python 3.7 support back to the library. Also during the process we found out that a new `Pillow` version was released that broke a lot of builds (including our if we didn't patch this today probably). So I changed how we can specify deps for a format, instead of just specifying one dep for a format, you can now provide a list of deps and it will be unpacked. In this case, to specifically ignore a `Pillow` version for `DefaultReader` / `TimeseriesWriter`.

## Pull request recom [... cut]
closed 2021-07-05 by JacksonMaxfield