AllenCellModeling/aicsimageio #277: Reading LIFs, scene names vs. project structure

psobolewskiPhD, 2021-07-08:
## Use Case
I use LAS X (Leica) to image and generate LIF projects of multi-well plate imaging.
The LIFs have either tilescans or multiple RoI per well.
Importing to ImageJ using ReadMyLIFs
https://twitter.com/jmutterer/status/1387740202484092934?s=20
lets me see the full names and select what I need for analysis.
Example:
[image]
This is R1 in well B/2.

Importing this same LIF with aicsimageio just lists R1, but if I have many wells, then I have many R1...
Also, I may have upwards of 60 ROI in a well, that I would like to import at once, but not merge.
Example:
[image]
This well has 60 ROI (not part of a mosaic merge).
Again, in ImageJ ReadMyLIFs I can mark the start and end of an import to get all 60 into a stack.

Here's a link to a representative LIF:
https://www.dropbox.com/s/jk9se8i1kpzqvsn/20210428_24w_L929_Ho_B2C3.lif?dl=0
B is row B, then B/n/ are wells in the B row. R1-R5 are actually mosaic merges, but that's not material to this issue. Same with C/
At the end is B/3/R1-R60 which ideally would be able to be imported together.

## Solution
Ideally, importing via aicsimageio would be aware of the Leica project structure so that tile scans and multiple ROI scans would be 1) distinguishable (most important) and 2) able to be imported together.

## Alternatives
N/A

psobolewskiPhD, 2021-07-12:
I should add that these projects were generated using the Navigator feature.
I see `readlif` has an issue regarding the structure:
https://github.com/nimne/readlif/issues/19

On a related note, with the above linked .LIF, I can't seem to access scenes by index—I thought this would be a way to get around the double scenes names. However:
>>> img.set_scene(1)
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/piotrsobolewski/Dev/miniforge3/envs/TF-StarDist-napari/lib/python3.9/site-packages/aicsimageio/aics_image.py", line 286, in set_scene
    raise IndexError(
IndexError: Scene id: 1 is not present in available image scenes: ('Image002', 'Image003', 'Image004', 'Image005', 'Image006', 'Image 1', 'R1', 'R4', 'R5', 'R2', 'R3', 'R1_Merged', 'R4_Merged', 'R5_Merged', 'R2_Merged', 'R3_Merged', 'R1', 'R4', 'R5', 'R2', 'R3', 'R1_Merged', 'R4_Merged', 'R5_Merged', 'R2_Merged', 'R3_Merged', 'R1', 'R4', 'R5', 'R2', 'R3', 'R1_Merged', 'R4_Merged', 'R5_Merged', 'R2_Merged', 'R3_Merged', 'R1', 'R4', 'R5', 'R2', 'R3', 'R1_Merged', 'R4_Merged', 'R5_Merged', 'R2_Merged', 'R3_Merged', 'B4_no_dye', 'R1', 'R2', 'R3', 'R4', 'R5', 'R6', 'R7', 'R8', 'R9', 'R10', 'R11', 'R12', 'R13', 'R14', 'R15', 'R16', 'R17', 'R18', 'R19', 'R20', 'R21', 'R22', 'R23', 'R24', 'R25', 'R26', 'R27', 'R28', 'R29', 'R30', 'R31', 'R32', 'R33', 'R34', 'R35', 'R36', 'R37', 'R38', 'R39', 'R40', 'R41', 'R42', 'R43', 'R44', 'R45', 'R46', 'R47', 'R48', 'R49', 'R50', 'R51', 'R52', 'R53', 'R54', 'R55', 'R56', 'R57', 'R58', 'R59', 'R60')
```
>>>img.current_scene_index
works...

JacksonMaxfield, 2021-07-12:
Ahhh well if `readlif` has an issue then yes we will as well. I can maybe dive deep into the issue for them but I am unfortunately not a LIF expert :/

---

On the scene index note, that's my bad. I merged those changes in but haven't released them. https://allencellmodeling.github.io/aicsimageio/CHANGELOG.html#unreleased

Sorry! Will do so now!

JacksonMaxfield, 2021-07-12:
aicsimageio v4.0.4 has been released. You should now be able to upgrade and properly use `set_scene(<int>)`

psobolewskiPhD, 2021-07-12:
Oh, shit I'm sorry I didn't realize what was in the docs wasn't for 4.0.3.

Works like a charm btw...

Regarding `readlif` I will link this issue over there. I think there may be more than one aspect at play: scene names and the channel issue observed by that OP.

JacksonMaxfield, 2021-07-12:
No worries! Our docs are aligned with `main` branch and generally `main` is always released but I just totally forgot last time.

psobolewskiPhD, 2021-07-12:
So now I'm a bit unsure if this issue belongs here or over at `readlif`.
I don't want to be bothering the wrong person!

Anyhow, based on your post in a different issue, I exported the xml and looked.
[big-lif.xml.zip](https://github.com/AllenCellModeling/aicsimageio/files/6802920/big-lif.xml.zip)
Here's the start of the last section of images:
`<Attributes><Attribute>ExtendedMemoryBlock</Attribute><Attribute>_DCSERVER_PREVENT_MEMKILL</Attribute></Attributes><Memory Size="2764800" MemoryBlockID="MemBlock_1050" /><Children /></Element><Element Name="B3 60 10x" Visibility="1" CopyOption="1" UniqueID="a6f972e2-a82c-11eb-bc72-a4bb6dca3902"><Data><Collection ChildTypeTest="AcceptAll"><ChildTypeList /></Collection></Data>
				<Memory Size="0" MemoryBlockID="MemBlock_1319" /><Children><Element Name="B" Visibility="1" CopyOption="1" UniqueID="a6f972e3-a82c-11eb-bc72-a4bb6dca3902"><Data><Collection ChildTypeTest="AcceptAll"><ChildTypeList /></Collection></Data>
						<Memory Size="0" MemoryBlockID="MemBlock_1320" /><Children><Element Name="3" Visibility="1" CopyOption="1" UniqueID="a6f972e4-a82c-11eb-bc72-a4bb6dca3902"><Data><Collection ChildTypeTest="AcceptAll"><ChildTypeList /></Collection></Data>
								<Memory Size="0" MemoryBlockID="MemBlock_1321" /><Children><Element Name="R1" Visibility="1" CopyOption="1" UniqueID="a6f972e5-a82c-11eb-bc72-a4bb6dca3902"><Data><Image TextDescription=""><Attachment Name="TileScanInfo" Application="LAS AF" FlipX="0" FlipY="0" SwapXY="0"><Tile FieldX="0" FieldY="0" PosX="0.0534542067" PosY="0.0256040830" />`

You can see 
`<Element Name="B3 60 10x" Visibility="1" CopyOption="1" UniqueID="a6f972e2-a82c-11eb-bc72-a4bb6dca3902">`
And then Children:
`<Children><Element Name="B" Visibility="1" CopyOption="1" UniqueID="a6f972e3-a82c-11eb-bc72-a4bb6dca3902">`
`<Children><Element Name="3" Visibility="1" CopyOption="1" UniqueID="a6f972e4-a82c-11eb-bc72-a4bb6dca3902">`
`<Children><Element Name="R1" Visibility="1" CopyOption="1" UniqueID="a6f972e5-a82c-11eb-bc72-a4bb6dca3902">`

At the moment, `aicsimageio` is just returning the last value, R1, as the scene name (img.scenes).
So to now make the feature request more precise, I'd like to see those parent names, so something like:
"B3 60 10x B3 R1"
In an ideal world, the scene would be "B3 60 10x B3" and then all the R1-60 would be part of that as M values.

JacksonMaxfield, 2021-07-12:
I think this issue should probably be moved over / recreated on `readlif`. All we are doing is parsing the result of `readlif` "image name" metadata: https://github.com/AllenCellModeling/aicsimageio/blob/main/aicsimageio/readers/lif_reader.py#L93

psobolewskiPhD, 2021-07-12:
Yeah, sorry about that, I had no idea about `readlif`.
Thank you for your patience, not to mention the awesome package!

closed 2021-07-12 by psobolewskiPhD