AllenCellModeling/aicsimageio #279: KeyError 'Name' when reading CZI image

zeissmicroscopy, 2021-07-09:
## System and Software
* aicsimageio Version: 4.0.2
* Python Version: 3.8
* Operating System: Win10

## Description

In ZEN I have an image that was created by "joining 2 czi stacks. The resulting image

https://www.dropbox.com/s/v59ip060mttyfrb/mitochondria_train_01_seg_ov_small.czi?dl=0

is the one I try to read using AICSimageIO

```python
filename = "D:\Testdata_Zeiss\Mitochondria_EM_with_DNN\original_data\mitochondria_train_01_seg_ov_small.czi"

img = AICSImage(filename)

# get number of zplanes
numZ = img.dims.Z
```

and here the error message:

```text
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-2-820b58d03155> in <module>
      4 
      5 # get number of zplanes
----> 6 numZ = img.dims.Z

C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\aicsimageio\aics_image.py in dims(self)
    499         if self._dims is None:
    500             self._dims = dimensions.Dimensions(
--> 501                 dims=self.xarray_dask_data.dims, shape=self.shape
    502             )
    503 

C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\aicsimageio\aics_image.py in xarray_dask_data(self)
    363                 self._reconstruct_mosaic
    364                 # Does the data have a tile dim
--> 365                 and dimensions.DimensionNames.MosaicTile in self.reader.dims.order
    366             ):
    367                 try:

C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\aicsimageio\readers\reader.py in dims(self)
    481         """
    482         if self._dims is None:
--> 483             self._dims = Dimensions(dims=self.xarray_dask_data.dims, shape=self.shape)
    484 
    485         return self._dims

C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\aicsimageio\readers\reader.py in xarray_dask_data(self)
    308         """
    309         if self._xarray_dask_data is None:
--> 310             self._xarray_dask_data = self._read_delayed()
    311 
    312         return self._xarray_dask_data

C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\aicsimageio\readers\czi_reader.py in _read_delayed(self)
    524             # Create coordinate planes
    525             meta = czi.meta
--> 526             coords, px_sizes = self._get_coords_and_physical_px_sizes(
    527                 xml=meta,
    528                 scene_index=self.current_scene_index,

C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\aicsimageio\readers\czi_reader.py in _get_coords_and_physical_px_sizes(xml, scene_index, dims_shape)
    443             channels = img.findall("./Channel")
    444             for channel in channels:
--> 445                 channel_name = channel.attrib["Name"]
    446                 scene_channel_list.append(channel_name)
    447 

KeyError: 'Name'
``` 
I looked ito the CZI metadata and insteaf of 'Name' '@Name' is used. This is most likely the cause. 

I know, that "I should know" why inside a CZI 'Name' and '@Name' is used, but I don't (at least right now). I will check. But I have seen this also for other entries. So may be "pragmatic" about it and check for 'Name' and '@Name'.

I have also seen this for 'Index' and '@Index' ...

[image]

JacksonMaxfield, 2021-07-09:
Any chance you can make a PR to resolve this issue? You all are the experts on CZI metadata :sweat_smile:

My time is pretty limited and we got a bunch of issues this past week that I need to look at next week.

chlmarchal, 2022-01-12:
Hello!
I tried to read the example image “8Bit-ZStack.czi” that can be found here: [http://czisamples.blob.core.windows.net/czi-samples/TESTIMAGES.zip](http://czisamples.blob.core.windows.net/czi-samples/TESTIMAGES.zip).

When trying to read this image with AICSImageIO in the following way:
```
img = AICSImage("/home/chloe/Documents/images/8Bit-ZStack.czi")
data = img.data 
```
I encounter the following error:
```
Traceback (most recent call last):
  File "czi_script.py", line 9, in <module>
    img.data  # returns 5D TCZYX numpy array
  File "/home/chloe/.local/lib/python3.8/site-packages/aicsimageio/aics_image.py", line 491, in data
    return self.xarray_data.data
  File "/home/chloe/.local/lib/python3.8/site-packages/aicsimageio/aics_image.py", line 431, in xarray_data
    and dimensions.DimensionNames.MosaicTile in self.reader.dims.order
  File "/home/chloe/.local/lib/python3.8/site-packages/aicsimageio/readers/reader.py", line 511, in dims
    self._dims = Dimensions(dims=self.xarray_dask_data.dims, shape=self.shape)
  File "/home/chloe/.local/lib/python3.8/site-packages/aicsimageio/readers/reader.py", line 338, in xarray_dask_data
    self._xarray_dask_data = self._read_delayed()
  File "/home/chloe/.local/lib/python3.8/site-packages/aicsimageio/readers/czi_reader.py", line 549, in _read_delayed
    coords, px_sizes = self._get_coords_and_physical_px_sizes(
  File "/home/chloe/.local/lib/python3.8/site-packages/aicsimageio/readers/czi_reader.py", line 468, in _get_coords_and_physical_px_sizes
    channel_name = channel.attrib["Name"]
KeyError: 'Name'

```

Any update about this issue?

JacksonMaxfield, 2022-01-12:
I haven't looked into the issue but if you have the time we would be more than happy to accept a PR to fix!

JacksonMaxfield, 2022-01-12:
From the looks of it, seems like the image doesn't have channel names? In which case we should create a channel name based off the channel index? https://github.com/AllenCellModeling/aicsimageio/blob/main/aicsimageio/metadata/utils.py#L123

toloudis, 2022-07-19:
The above closed issue https://github.com/AllenCellModeling/aicspylibczi/issues/108 has some additional important info

closed 2022-08-02 by evamaxfield