AllenCellModeling/aicsimageio #282: AICSImage CZI images is Slow (infinite?) processing compared to CZIReader

MosGeo, 2021-07-13:
## System and Software
* aicsimageio Version: 4.0.4
* Python Version: 3.8
* Operating System: Windows 10

## Description
**Symptom:** Aicsimageio starts processing (cpu + ram) after calling a lot of functions with CZI files (1GB) when using AICSImage and it doesn't stop. For example, when calling `shape`.
**Possible issue:** Compute is being called in any operation.

Calling the CziReader directly works normally.

Test image is multichannel (7), tiled (1000+), RGB image.

## Expected Behavior
Show the shape of the image without processing the image.

## Reproduction
```
from aicsimageio import AICSImage
from aicsimageio.readers.czi_reader import CziReader

# This is slow
image = AICSImage("Test.czi")
print(image.shape)

# This is normal
image = CziReader("Test.czi")
print(image.shape)
```
## Environment
Clean venv environment with only aicsimageio installed.

JacksonMaxfield, 2021-07-13:
This is because `AICSImage` reconstructs the mosaic by default and our tile stitching is very slow for some reason. You can turn this off with `reconstruct_mosaic=False` if you want to use the `AICSImage` object specifically but for now going to close this as it is a duplicate of #274

closed 2021-07-13 by JacksonMaxfield

MosGeo, 2021-07-13:
Ahhh, makes total sense :) I was going crazy trying to figure it out. Thanks!

martinschatz-cz, 2023-09-10:
Thank you! Saved me from improper dimension size loading of large CZI file.