AllenCellModeling/aicsimageio #301: Conversion from CZI to OME TIFF

MosGeo, 2021-08-24:
## System and Software
* aicsimageio Version:  4.1.0
* Python Version: 3.9.6 64bit
* Operating System: Windows 10

## Description
This is my first time trying to save files using aicsimageio and dealing with ome.tiff. I am still getting my head around it. So maybe I am missing something.

When running 
```
from aicsimageio import AICSImage
filename = "./my_czi_file.czi"
img  = AICSImage(filename, reconstruct_mosaic=False)
img.save("my_tiff_file.ome.tiff")
```

the following error is recieved
```
  File "C:\Users\Mustafa\Desktop\CZIConvert\main.py", line 6, in <module>
    img.save("my_file.ome.tiff")
  File "C:\Users\Mustafa\Desktop\CZIConvert\.venv\lib\site-packages\aicsimageio\aics_image.py", line 797, in save
    OmeTiffWriter.save(
  File "C:\Users\Mustafa\Desktop\CZIConvert\.venv\lib\site-packages\aicsimageio\writers\ome_tiff_writer.py", line 214, in save
    ome_xml = OmeTiffWriter.build_ome(
  File "C:\Users\Mustafa\Desktop\CZIConvert\.venv\lib\site-packages\aicsimageio\writers\ome_tiff_writer.py", line 590, in build_ome
    ome_dimension_order, is_rgb = OmeTiffWriter._resolve_OME_dimension_order(
  File "C:\Users\Mustafa\Desktop\CZIConvert\.venv\lib\site-packages\aicsimageio\writers\ome_tiff_writer.py", line 328, in _resolve_OME_dimension_order
    raise ValueError(
ValueError: Data array has unexpected number of dimensions: is_rgb = True and shape is (1102, 1, 7, 1, 1200, 1600, 3)
```
Note that the image is: 1) tiles, 2) multi-cannel, 3) RGB. It is also pyramidal but as I understand it, aicsimageio will just take the 0 level for now.

## Expected Behavior
ome.tiff is saved.

## Reproduction
See description

## Environment
Clean virtual environment with only aicsimageio installed.

JacksonMaxfield, 2021-08-24:
We currently do not support saving OME-TIFF with tiles, just the whole reconstructed YX planes.

The OME model itself only "recently" adopted Tiles (and other dims) as supported so it's just a "tooling needs to catch up" problem. I see that you have 1102 tiles of 1200 x 1600 YX which is massive so I can understand why you wouldn't `reconstruct_mosaic=True` (#274) so for now it seems we may not have a solution for you.... That said. I think under the hood our current implementation of `OmeTiffWriter` would segfault with this code because it tries to pull in the whole data block at once.... (#214)

Basically... a lot going on here, all of it is known, but we don't have a solution for yet.

MosGeo, 2021-08-25:
Thanks for the info @JacksonMaxfield. Yes, we are dealing with very large datasets. This example actually a small file. I am thinking of converting files to a more open format ome.tiff or even ome.zarr that will simplify their viewing and analysis by different tools in the future (including napari).

Yes, reconstructing the mosaic in aicsimageio right now is practically impossible for these files.

Nicholas-Schaub, 2022-03-02:
If you're looking for something that can convert czi to ome tiff, you should check out our `bfio` utility. It is not as fully featured as AICSImageIO, but we are in the process of creating bindings for our reading and writing tools for AICSImageIO.

One of the examples we provide in our documentation is how to create a scalable tiled tiff converter:
https://bfio.readthedocs.io/en/master/Examples/Converter.html#an-efficient-scalable-tiled-tiff-converter

JacksonMaxfield, 2022-03-03:
Honestly, gotta give credit to ya @MosGeo. You really do have a crazy dataset :joy: 

I hope one of the various projects can support your data soon.

---

> I'll add that as you see from the below error, bfio did not detect that the RGB channels (it just lumped everything together (7 channels * RGB = 21)

That is actually how OME stores the data. OME doesn't have an RGB dimension.

Nicholas-Schaub, 2022-03-04:
Ahhh, yeah that's a bug we are actually in the process of fixing. `bfio` doesn't handle interleaved channels properly. I'll let you know when that's fixed. Should be within the next week.

We use `bfio` to convert czi files that are 30,000x50,000x55x1000x1, so it should be able to handle your data once we get that interleaved channel thing fixed.

Nicholas-Schaub, 2022-03-04:
Also, I'm not trying to steal the thunder from AICSImageIO. The only reason I'm here is because we are writing a Reader for `bfio`.

JacksonMaxfield, 2022-03-04:
Oh I don't mind. I would rather have @MosGeo's issues solved than not :joy:

Excited for a reader tho!

Nicholas-Schaub, 2022-03-04:
Just submitted a PR for the tiled tiff reader.

Here is a link for the previously reported issue that we are currently working to resolve. I think this is the same issue @MosGeo ran into.

https://github.com/PolusAI/bfio/issues/10

Nicholas-Schaub, 2022-03-04:
Also, for our writer (I should probably open up a separate discussion), but my thought is that you should be able to submit a delayed array. We can then do chunked writing with our writer inside of the `Writer.save` function.

Nicholas-Schaub, 2022-03-04:
@MosGeo , this may be a silly question, but can you just do chunked reading of your czi file using `img.dask_array` and feed the chunks into the BioWriter instead of using a BioReader?

Nicholas-Schaub, 2022-03-04:
We just fixed the interleaved channel bug. @MosGeo, would you be able to share an example file so we can test against it? I know we can handle interleaved channels, but you have multichannel interleaved rgb. It should work on it, but I'd like to test if you're willing to share.

MosGeo, 2022-03-06:
@Nicholas-Schaub please check your gmail. Thanks! I will test the latest version in a bit.

I am now considering the following for manual conversion:
- Load image with aicsimageio.
- Stitch tiles with m2stitch (zulip napari recently have a discussion about it). I am still not sure if it can handle RGB or not. If not, I might convert to grayscale/stitch to get locations, and then stitch the RGB if that is possible.
- Save in ome.zarr with pyramidal generation if possible. Now I have something that can be handled easily for users.
- Load with aicsimageio (can't handle pyramidal for now) or napari-ome.zarr.
- View in napari.
- Analyze.

As you can see, I don't really need the tiles so I can loose them. In fact, the analysis code will be easier without the tiles. I still have not looked into BioWriter but single channel image would be a deal breaker.

Nicholas-Schaub, 2022-04-18:
@MosGeo We have released `bfio==2.3.0`. We just opened a PR for the writer implementation (#396) which should permit semi-scalable writing if you pass in a dask array into the writer. Can you test?

MosGeo, 2022-05-25:
@Nicholas-Schaub @JacksonMaxfield 

Sorry for the late reply. I seem to only able to pickup this effort intermittently with other work commitments. Thanks for all the work. So here are the results of my tests:

## Test image info:
This is the info from Zen Zeiss. So, it is a tiled image, with multiple channels. You will also note that each channel is 24 bit which means that each channel is an RGB.
[image]

## Reading test:
You can see the code and output below. Notes:
1. Both BioReader and AICSImage picked up the actual size of the image which is exactly as reported by the Zen Zeiss. Overall, each is doing what is expected of them. For example, aicslibczi is reporting the tiles (M). So far so good.
```
bfio_reader = BioReader(image_filename)
ai_reader   = AICSImage(image_filename, reader=CziReader)
czi_reader  = CziFile(image_filename)
print(bfio_reader._DIMS)
print(ai_reader.dims)
print(czi_reader.get_dims_shape())
```
The output is
```
{'X': 3054, 'Y': 1205, 'Z': 1, 'C': 21, 'T': 1}
<Dimensions [T: 1, C: 7, Z: 1, Y: 1205, X: 3054, S: 3]>
[{'A': (0, 3), 'X': (0, 1600), 'Y': (0, 1200), 'C': (0, 7), 'M': (0, 2), 'S': (0, 1)}]
```
# bfio metadata
Let's start by looking at `bfio_reader.metadata`. You will note that it actually picked up meta data for three images. This is because the CZI file actually has two attachments (barcode image, and a macro image of the sample). Cool. I don't think I can retrieve the data for those extra attachments though. `bfio_reader.channel_names` also reports the 7 channel names correctly and the channel info are correct.
```
OME(
   experimenters=[<1 Experimenters>],
   images=[<3 Images>],
   instruments=[<1 Instruments>],
   structured_annotations=[<2810 Structured_Annotations>],
)
```

# Writing tests (bfioreader->bfiowriter->bfioreader)
Let's see what we can do with writing. Starting with bfio, where we pass the reader metadata as input. Bfio warns that `WARNING  - The BioWriter only writes single image files, but the metadata has 3 images. Setting the number of images to 1`. This means that we will loos the metadata of the other images. This is fine for the time being.

```
out_filename = Path(r"../output/out.ome.tif")
bfio_writer = BioWriter(out_filename,metadata=bfio_reader.metadata)
bfio_writer[:]= bfio_reader[:]
bfio_writer.close()
bfio_out_reader = bfio.BioReader(out_filename)
print(bfio_reader._DIMS)
```

Now, reading back the images using a new bfioreader, we are able to retrieve the image. Some notes:
- When retrieving the actual data of the image, bfioreader returns the whole image. I assume bfio just does the transformation defined in the metadata only and overlaps the pixel on top of each other (i.e., no image registration)? The resultant images look good as they are already registered.

## Writing tests (bfioreader->bfiowriter->AICSImage)
Following the same procedure for writing outlined above using bfio and then attempting to read it using AICSImage, there is an error when enquiring about shape which says `conflicting sizes for dimension 'C': length 21 on the data but length 7 on coordinate 'C'` this is because the metadata house 7 channels (correctly) but the actual channels saved are 21 because each is an RGB image.

## Writing tests (AICSImage/CZI -> AICSImage/ome.tiff -> AICSImage/ome.tiff)
Everything is working as expected. It loads the data correctly. Some notes:
- I know that aicsimagio does stitching automatically when retrieving the data so that might take long time. In future tests, I will attempt to stitch the tiles before saving.
```
out_filename = Path(r"../output/out.ome.tif")
OmeTiffWriter.save(ai_reader.data, out_filename, ai_reader.dims.order)
ai_out_reader   = AICSImage(out_filename)
print(ai_out_reader.shape)
```

## Writing tests (AICSImage/CZI -> AICSImage/ome.tiff -> bfiowriter/ome.tiff)
To be continued... need to go now :)

SeanLeRoy, 2023-03-16:
>Seems like we are covering a lot of the bases now thanks to all the contributions from bfio and others ❤️

@evamaxfield Does this seem ready to close then (with the potential to be re-opened if uncovered use cases arise)?

evamaxfield, 2023-03-16:
I think there are still some issues that need to be addressed. Can we talk about this issue on Monday?

SeanLeRoy, 2023-03-20:
Closing this due to inactivity, feel free to re-open with file we can test this with if still desired!

closed 2023-03-20 by SeanLeRoy