AllenCellModeling/aicsimageio #304: Support CZI subblocks

MatteBailey, 2021-08-31:
## Use Case
CZI files have additional metadata embedded in their subblocks that can be relevant to our users. This subblock metadata is accessible with [aicspylibczi](https://github.com/AllenCellModeling/aicspylibczi), but we currently do provide a way to access it directly through AICSImageIO.

## Solution
From the initial conversations I've had with folks about this, it seems like we would want to append the subblock metadata to end of the normal CZI metadata returned by the `.metadata` property. This would also allow the subblock metadata to be handled by the XSLT when using the `.ome_metadata` property.

## Alternatives
Alternatively, we could add a `subblock_metadata` property to the CZI reader. While this might be more clear in some ways, it would make it more complicated to process this metadata with the XSLT.

MatteBailey, 2021-08-31:
@cfrick13, let us know if you have any other thoughts on this!

cfrick13, 2021-08-31:
First, what is the XLST?
I think returning the subblock metadata appended to the end of the normal CZI metadata would be a fine solution. I think the only potential issue would be that this could break the way that users currently work with  `reader.metadata`. But I think that is a pretty unlikely scenario. 
So my vote is to: "append the subblock metadata to end of the normal CZI metadata returned by the .metadata property"

The other potential issue now that I think of it, is that the subblock metadata can be huge! It coule be as large as 14*25*500 (scenes x Z slices x time points) = 175,000 blocks for long time lapse movies. Not sure what kind of issues that presents for computing time or memory.

MatteBailey, 2021-08-31:
@cfrick13 Sorry, the XSLT is the code that we wrote to translate from CZI to OME metadata.

cfrick13, 2021-08-31:
Got it. In that case, I think maybe there should be the following options:
1. reader.metadata returns one metadata field with subblock metadata appended to end of the normal CZI metadata 
2.  reader.normal_metadata returns only the non-subblock containing metadata alone
3. reader.subblock_metadata reutnrs the subblock metadata alone.

then xlst will work and the other metadata fields can be accessed independently by any users who might still want that option.

MatteBailey, 2021-11-10:
@JacksonMaxfield I'm digging into this feature now, and am debating what the most reasonable way to expose this functionality to users. Here are the options I see:

1. For CZIs, automatically append the subblock metadata to the usual `metadata` property. This could be done in like two lines of code. The obvious downsides of this are that there's a chance it could break existing code in some way, and could be pretty slow if there are a lot subblocks for a given file.
2. Add a param like `read_subblock_metadata` to the `__init__` method of `CziReader`, in which case the subblock metadata would be appended to the normal `metadata` property. From what I can tell, it seems like different readers already have specific params in their `__init__` methods, so this would follow our existing patterns.
3. Add extra properties to `CziReader` (and I guess `Reader` to make the typing work) like `subblock_metadata` and `ome_metadata_with_subblocks`. As I'm typing this option out, it seems a bit convoluted, so I don't think I'll go in that direction.

After writing that all out, option 2 seems like the most elegant one. Thoughts?

JacksonMaxfield, 2021-11-10:
I would go with option 2. After option 2 is implemented, if it doesn't meet all the needs, I would bet option 3 is next best.

MatteBailey, 2021-11-10:
Sounds good!

PR #353 (2021-12-02): feature/czi-subblock-metadata
## Description
Resolves #304 

This PR adds a param to `CziReader` that causes subblock metadata to be appended to the other embedded metadata.

## Pull request recommendations:
- [X] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [X] Link to any relevant issue in the PR description. Ex: _Resolves [gh-<number>], adds tiff file format support_
- [X] Provide relevant tests for your feature or bug fix.
- [X] Provide or update documentation for any feature added by your pull request.

Thanks for contributing!
closed 2021-12-02 by MatteBailey