AllenCellModeling/aicsimageio #307: Conflict between formats.FORMAT_IMPLEMENTATIONS and AICSImage.determine_reader

tlambert03, 2021-09-14:
Observation from a newbie to the repo...

as I was tinkering with #306, I started out by putting a single "supported" file format in `FORMAT_IMPLEMENTATIONS`, above all the default readers:

```py
FORMAT_IMPLEMENTATIONS: Dict[str, str] = {
    ...
    "nd2": "aicsimageio.readers.bioformats_reader.BioformatsReader",
    "bmp": "aicsimageio.readers.default_reader.DefaultReader",
    ...
}
```

... but the `_is_supported_image` method of my new reader still returned `True` for all of the other  supported formats.  I was a bit surprised when that broke the AICSImage tests because `.png` (and other cross-compatible formats) were being stolen by the Bioformats reader.  That appears to be this section in `determine_reader`:

https://github.com/AllenCellModeling/aicsimageio/blob/982efbb2abb855d41daee2f659c84e7c4be6d4db/aicsimageio/aics_image.py#L146-L148

... which effectively trumps whatever ordering was assigned in FORMAT_IMPLEMENTATIONS.  It appears that whichever reader appears first in the dict, will take _all_ of its supported formats, regardless of the ordering of keys that come later.

thoughts?

JacksonMaxfield, 2021-09-14:
Ahhhhhh. This is an interesting one. Totally forgot that we do that.... Initial thoughts:

Maybe we just try path suffixes first and if its fails (or no suffix) we roll back to trying all readers?

This would generally result in a faster Reader lookup too.... I.e. using the suffix and getting reader from a dictionary instead of checking bytes in a loop.

cc @toloudis

tlambert03, 2021-09-14:
> Maybe we just try path suffixes first and if its fails (or no suffix) we roll back to trying all readers?

seems reasonable to me

JacksonMaxfield, 2021-09-14:
Now up for discussion that leaves us at:

```python
FORMAT_IMPLEMENTATIONS = {
    ".png": "aicsimageio.readers.default_reader.DefaultReader",
}
```

or

```python
FORMAT_IMPLEMENTATIONS = {
    ".png": [
        "aicsimageio.readers.default_reader.DefaultReader",
        "aicsimageio.readers.bioformats_reader.BioformatsReader",
    ]
}
```

I think the second one. Would be extra attempts but I think this is also the method that base `imageio` handles it.

toloudis, 2021-09-14:
if we are going by extension then one reader per extension should be enough, right?

JacksonMaxfield, 2021-09-14:
I agree that one reader per extension _should_ be enough and if it is, it will only ever hit the first one. But my idea was basically having a priority reader list for each suffix.

toloudis, 2021-09-14:
another way of asking this is: is it a problem if bioformats reads png files?  does it arrange the data totally differently than imageio does?  And how worried should we be about reading png files anyway? :)

toloudis, 2021-09-20:
After a little discussion in a meeting today I think we landed on:
1. a user-provided reader hint takes precedence
2. then check file extension... try a reader that is 1-1 mapped to file extension (taking care of string handling: casing, tiff vs tif etc)
3. use the in-order attempts to load file if the extension does not exist or is not found.

JacksonMaxfield, 2021-09-24:
I have thought of a _very_ good reason to have an ordered list of readers per file extension.

`.tiff` files should always try [OmeTiffReader, TiffReader]... because its _very_ common for users to not have the OME extension tagged on.

toloudis, 2021-09-24:
> I have thought of a _very_ good reason to have an ordered list of readers per file extension.
> 
> `.tiff` files should always try [OmeTiffReader, TiffReader]... because its _very_ common for users to not have the OME extension tagged on.

that is also a good reason to merge those two readers into one!

JacksonMaxfield, 2021-09-24:
That's a larger project and so for now I am proceeding with List of readers :upside_down_face:

PR #318 (2021-09-24): bugfix/resolve-conflict-between-format-impls-and-determine-reader
## Description

Changes our format implementations to a Dict[str, List[str]] where each suffix has a list of supported Readers. This is useful in cases where the user may have forgotten a suffix (i.e. file is saved as `tiff` but is actually and `ome.tiff`) and in cases where the user only installed certain readers (i.e. installed `bioformats` but not `base-imageio`) multiple options for format support.

Additionally some minor typing changes.

## Pull request recommendations:
- [x] Name your pull request _your-development-type/s [... cut]
closed 2021-09-24 by JacksonMaxfield