AllenCellModeling/aicsimageio #309: aicsimageio in requirements.txt

elmbeech, 2021-09-17:
Dear AICSImageIO developers,

First: thank you for this excellent package!

Second:  I have aicsimageio listed in my requirements file. 
+ https://gitlab.com/bue/jinxif/-/blob/master/requirements.txt
Now, when I pip install my package (pip install jinif==0.0.23) all dependencies listed in this requirements.txt file get installed except aicsimageio.
I don't understand why. Do you have any idea why installing aicsimageio as dependencies is tricky? 
Any insights would be appreciated.

Thank you, Elmar Bucher

JacksonMaxfield, 2021-09-17:
Hello @elmbeech!

This is an interesting one. I just tried creating a new environment and installing `aicsimageio[czi]` from a requirements.txt and all worked fine for me.

Can I ask which Python version you are using? `python -V` should get you that information
And which pip version you are using? `pip -V` as well

What OS are you using and are you using an environment manager (conda, pipenv, etc)?

JacksonMaxfield, 2021-09-17:
You also have a `requirements.txt` file _and_ a `setup.py` file: https://gitlab.com/bue/jinxif/-/blob/master/setup.py#L89

I think you need to add `aicsimageio[czi]` to the `setup.py` `install_requires`

elmbeech, 2021-09-17:
Hi @JacksonMaxfield , 

I run python 3.9 (Python 3.9.5 (default, Jun  4 2021, 12:28:51))  but, 
because this tool need cellpose I have to install it in a python 3.8  environment (Python 3.8.11 (default, Aug  3 2021, 15:09:35)).
We use miniconda (conda 4.10.3) on the server runs CentOS 7.7.1908.

JacksonMaxfield, 2021-09-17:
My best bet is to add it to `install_requires` and publish a new version.

elmbeech, 2021-09-18:
Dear @JacksonMaxfield ,

I just updated my setup.py and  requirements.txt file and checked  (pip install jinxif==0.0.24). 
Now everything runs smoothly and aicsimageio gets properly installed.
I completely forgot that I have the libraries to mention in the setup_file too. Sorry,

(As you could see in the old setup.py file, aicsimageio made it possible to get rid of python-bioformats and especially python-javabridge, which was very painfull to install and run.)

Thank you for all you help! It was not an issue with aicsimageio at all. 
I will close this issue.

Best, Elmar

closed 2021-09-18 by elmbeech

JacksonMaxfield, 2021-09-18:
Awesome. Glad to hear the fix was painless :slightly_smiling_face: 

And thanks for the kind words.

elmbeech, 2021-09-18:
Now I completely understand. I just can put my requirements into the setup.
I don't really need the requitements.txt ~ or I can use it for non-essential libraries like [czi] extension and imagecodecs. 
Cool!  : ).