AllenCellModeling/aicsimageio #323: JAVA difficulty with aicsimageio[bioformats] on windows 10

cfrick13, 2021-09-28:
## System and Software
* aicsimageio Version: 4.2.0
* Python Version: 3.8
* Operating System: Windows 10

## Description
*A clear description of the bug*
I can't get the bioformats reader to work. It throws an error (See below). I think the issue is my JAVA_HOME path. 
I tried to figure that out but was unable to. 

I tried to follow the instructions here:
https://stackoverflow.com/questions/52511778/how-to-install-openjdk-11-on-windows

## Expected Behavior
*What did you expect to happen instead?*

I expected it to work. 

## Reproduction
*A minimal example that exhibits the behavior.*

```
from aicsimageio import AICSImage

# bf_reader = AICSImage(r'\\allen\aics\assay-dev\MicroscopyData\Jamie\2021\20210813\20210813_J03_001.sld')
bf_reader = AICSImage(r'C:\Users\chris.frick\Downloads\3.nd2')
```

```---------------------------------------------------------------------------
UnsupportedFileFormatError                Traceback (most recent call last)
C:\Users\CHRIS~1.FRI\AppData\Local\Temp/ipykernel_16464/1564468111.py in <module>
      2 
      3 # bf_reader = AICSImage(r'\\allen\aics\assay-dev\MicroscopyData\Jamie\2021\20210813\20210813_J03_001.sld')
----> 4 bf_reader = AICSImage(r'C:\Users\chris.frick\Downloads\3.nd2')

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\aicsimageio\aics_image.py in __init__(self, image, reader, reconstruct_mosaic, **kwargs)
    214         if reader is None:
    215             # Determine reader class and create dask delayed array
--> 216             ReaderClass = self.determine_reader(image, **kwargs)
    217         else:
    218             # Init reader

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\aicsimageio\aics_image.py in determine_reader(image, **kwargs)
    180                 if path.lower().endswith(f".{format_ext}"):
    181                     installer = READER_TO_INSTALL[readers[0]]
--> 182                     raise exceptions.UnsupportedFileFormatError(
    183                         "AICSImage",
    184                         path,

UnsupportedFileFormatError: AICSImage does not support the image: 'C:\Users\chris.frick\Downloads\3.nd2'. File extension suggests format: 'nd2'. Install extra format dependency with: `pip install aicsimageio[bioformats]`. See all known format extensions and their extra install name with `aicsimageio.formats.FORMAT_IMPLEMENTATIONS`.
```

also tried
```
from aicsimageio.readers import bioformats_reader
# 'sld': ['aicsimageio.readers.bioformats_reader.BioformatsReader'],

bf_reader = bioformats_reader.BioformatsReader('20210813_J03_001.sld')
```

```
---------------------------------------------------------------------------
JVMNotFoundException                      Traceback (most recent call last)
~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\aicsimageio\readers\bioformats_reader.py in __init__(self, path, series, meta, original_meta, memoize, options)
    250         try:
--> 251             loci = get_loci()
    252         except jpype.JVMNotFoundException as e:

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\bioformats_jar\__init__.py in get_loci(log_level)
     73     """
---> 74     start_jvm()
     75     loci = jpype.JPackage("loci")

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\bioformats_jar\__init__.py in start_jvm(classpath, attach_thread, memory, **kwargs)
     50 
---> 51     jpype.startJVM(f"-Xmx{memory}", classpath=classpath, **kwargs)
     52 

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\jpype\_core.py in startJVM(*args, **kwargs)
    191     if not jvmpath:
--> 192         jvmpath = getDefaultJVMPath()
    193 

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\jpype\_jvmfinder.py in getDefaultJVMPath()
     73         finder = LinuxJVMFinder()
---> 74     return finder.get_jvm_path()
     75 

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\jpype\_jvmfinder.py in get_jvm_path(self)
    211             raise jvm_notsupport_ext
--> 212         raise JVMNotFoundException("No JVM shared library file ({0}) "
    213                                    "found. Try setting up the JAVA_HOME "

JVMNotFoundException: No JVM shared library file (jvm.dll) found. Try setting up the JAVA_HOME environment variable properly.

During handling of the above exception, another exception occurred:

JVMNotFoundException                      Traceback (most recent call last)
C:\Users\CHRIS~1.FRI\AppData\Local\Temp/ipykernel_16464/3427452841.py in <module>
      2 # 'sld': ['aicsimageio.readers.bioformats_reader.BioformatsReader'],
      3 
----> 4 bf_reader = bioformats_reader.BioformatsReader(r'\\allen\aics\assay-dev\MicroscopyData\Jamie\2021\20210813\20210813_J03_001.sld')

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\aicsimageio\readers\bioformats_reader.py in __init__(self, image, original_meta, memoize, options)
    107         }
    108         try:
--> 109             with BioFile(self._path, **self._bf_kwargs) as rdr:  # type: ignore
    110                 md = rdr._r.getMetadataStore()
    111                 self._scenes: Tuple[str, ...] = tuple(

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\aicsimageio\readers\bioformats_reader.py in __init__(self, path, series, meta, original_meta, memoize, options)
    251             loci = get_loci()
    252         except jpype.JVMNotFoundException as e:
--> 253             raise type(e)(
    254                 str(e) + "\n\nBioformatsReader requires a java executable to be "
    255                 "available in your environment. If you are using conda, you can "

JVMNotFoundException: No JVM shared library file (jvm.dll) found. Try setting up the JAVA_HOME environment variable properly.

BioformatsReader requires a java executable to be available in your environment. If you are using conda, you can install with `conda install -c conda-forge openjdk`. 
```

## Environment
*Any additional information about your environment*

Here is how I set up my environment

```
conda create --name aics_bf2  python=3.8 -y
conda activate aics_bf2
pip install pywin32
pip install jupyterlab
pip install pandas
pip install aicsimageio[bioformats]
conda install -c conda-forge openjdk
```

toloudis, 2021-09-28:
did you try `export JAVA_HOME=path_to_your_conda_environment/bin` ?

tlambert03, 2021-09-28:
> did you try export JAVA_HOME=path_to_your_conda_environment/bin ?

that would be for *nix... @cfrick13  is on Windows.  Lemme see if I can find a PC to test this on.  if you definitely did `conda install -c conda-forge openjdk`, you might try deactivating/reactivating your env

tlambert03, 2021-09-28:
alas :/ I just followed your exact environment setup instructions (also py38, Win10) and was able to read an ND2 file just fine.  Do try the deactivate/reactive trick and see if it works.  Otherwise we'll need to get a bit more clever here

tlambert03, 2021-09-28:
for what it's worth, in my working `aics_bf2`, the `JAVA_HOME` variable is set to `%CONDA_PREFIX%\Library` so you could try:

```
set JAVA_HOME=%CONDA_PREFIX%\Library
```

cfrick13, 2021-09-28:
thanks for looking into this. I just ran the line below and reactivated my environment and now can read the ND2 file!
```
set JAVA_HOME=%CONDA_PREFIX%\Library
```

closed 2021-09-28 by JacksonMaxfield

toloudis, 2021-09-28:
good thing to add to the readme/docs

tlambert03, 2021-09-28:
yeah, really it _shouldn't_ be necessary.  but adding that one might need to reactivate the environment if they've just installed it is a good thing to do,  will make a PR

tlambert03, 2021-09-28:
actually... where should I add that?  readme?  or somewhere that will show up at https://allencellmodeling.github.io/aicsimageio/?

JacksonMaxfield, 2021-09-28:
I would maybe add it to the error message? That or in the header docstring of BioformatsReader.