AllenCellModeling/aicsimageio #325: Ship type annotations with library

JacksonMaxfield, 2021-09-30:
Currently if a library has aicsimageio in dependencies / uses aicsimageio and runs mypy on their dependent library, mypy will complain that there are no types for aicsimageio.

To resolve this we simply need to add an empty `py.typed` file to the base of the package dir. And additionally add this file to the MANIFEST.ini

Can see this in a PR on my other project: https://github.com/CouncilDataProject/cdp-backend/pull/93/files
(ignore the dependency shuffling)

tlambert03, 2021-10-01:
you _may_ also need this in `setup.py` to get it into both the wheel and the sdist (I can never remember... run `setup.py` locally and check the `dist` folder to make sure it gets in everything

```py
setup(
  package_data={'aicsimageio': ['py.typed']},
)
```

JacksonMaxfield, 2021-10-01:
Last I read, because we use MANIFEST.in, we use `include_package_data`: https://github.com/AllenCellModeling/aicsimageio/blob/main/setup.py#L130

Will check though

PR #397 (2022-05-22): admin/ship-mypy-type-annotations-drop-py37-add-py310-manage-test-dependencies-separate-for-each-tox-env
## Description

Ships type annotations with the library when installed. Now people can define functions based on our types and mypy will correctly resolve them.

This also massively overhauls the testing portions of our library. Extra readers and writers have been moved to their own subdirs in our tests dirs. And each extra reader and writer is tested independently from one another. This is to make it possible to avoid dependency overlap.

Specifically, this relates to how bfio doesn't suppo [... cut]
closed 2022-05-22 by JacksonMaxfield