AllenCellModeling/aicsimageio #349: Process more tiff tags

ianhi, 2021-11-04:
## Use Case
Loading the tiff from a micromanager MDA it seems a large amount of metadata is getting thrown into tag code: `50839` (which is not included in https://www.loc.gov/preservation/digital/formats/content/tiff_tags.shtml) and I can't figure out where it's documented.

But it would be nice to process these and when turning it into xarray split these keys up into attrs.
[image]

## Solution
Break up the `processed` and `unprocessed` attrs into individual items so that it renders like this:

[image]

You can do this breaking up something like this:
```python
unprocessed = []
processed = []
xarr = gr.xarray_dask_data
attrs = xarr.attrs
for k,v in attrs['unprocessed'].items():
    attrs[k] =v 
    unprocessed.append(k)
attrs['unprocessed'] = unprocessed

# also break up code 50839 which is where it seems MM metadata lives
import json
for k,v in json.loads(attrs[50839]['Info']).items():
    attrs[k] =v
attrs.pop(50839)
# attrs
xarr = xarr.assign_attrs(attrs)

```

## Alternatives
*Please describe any alternatives you've considered, even if you've dismissed them*

toloudis, 2023-03-10:
also see #450

SeanLeRoy, 2023-11-29:
AICSImageIO is entering maintenance mode meaning only critical bugfixes will be applied due to `bioio` a new python package taking over in place of AICSImageIO with the same features. I copied this feature over to `bioio-ome-tiff` which is the plugin to `bioio` responsible for reading OME TIFF files. Feel free to close or move that issue as you see fit.

https://github.com/bioio-devs/bioio-ome-tiff/issues/5