AllenCellModeling/aicsimageio #355: transform czi metadata to ometiff metadata and save the ometiff metadata as xml file

elmbeech, 2021-11-15:
## Use Case
I would like to be able to e.g. read czi files. (this is possible.)
transform the metadata to ome.tiff format. (should work, but I did not get it to work.)
and only save the ometiff metadata xml file, not the whole image. (works with ET.) 

It might be that this is already possible, but I could not figure out how.

## Solution
what I would like to be able to do:
```
from aicsimageio import AICSImage
import xml.etree.ElementTree as ET

o_img = AICSImage('my_file.czi')
et_ome = o_img.ome_metadata()    # results in a NotImplementedError !!!
ET.ElementTree(et_ome).write('my_file.ome.xml')
```

## Alternatives
What is possible but overkill:
```
from aicsimageio import AICSImage
import ome_types

AICSImage('my_file.czi').save('my_file.ome.tiff')
x_ome = ome_types.from_tiff('my_file.ome.tiff')
f = open('my_file.ome.xml', 'w')
f.write(x_ome.to_xml())
f.close()
```

JacksonMaxfield, 2021-11-15:
Hey @elmbeech what version of AICSImageIO are you on?

[CziReader.ome_metadata](https://github.com/AllenCellModeling/aicsimageio/blob/main/aicsimageio/readers/czi_reader.py#L758) was added a while ago.

JacksonMaxfield, 2021-11-15:
Note: your second example produces different OME XML...

I am in the process of upgrading it to have a parameter called `use_metadata_translation` that will do what you are trying to do.

Basically `ome_metadata` is some "translated" metadata whereas the current (and default) behavior of `save` "generates" some basic metadata.

JacksonMaxfield, 2021-11-15:
Here is the WIP PR for the `save` function extension: https://github.com/AllenCellModeling/aicsimageio/pull/300

elmbeech, 2021-11-16:
Thank you and sorry @JacksonMaxfield, I should have checked before! 
I was running aicsimageio.get_module_version() '4.1.0'. The current version is 4.5.0 and with this one it works perfectly! 

About the hint,
do I understand you right, the "Solution" uses this xls: transformer https://github.com/AllenCellModeling/czi-to-ome-xslt
and the "Alternatives" generates a simpler, less complete ometiff file?

JacksonMaxfield, 2021-11-16:
> do I understand you right, the "Solution" uses this xls: transformer https://github.com/AllenCellModeling/czi-to-ome-xslt
and the "Alternatives" generates a simpler, less complete ometiff file?

`ome_metadata` returns the XML produced by `czi-to-ome-xslt` and the `save` function by default returns a simpler, less complete ome metadata (and file). Correct!

closed 2021-11-16 by JacksonMaxfield

elmbeech, 2021-11-16:
Yes, this issue can definitely be closed! 
I think this is very cool step into the future.
By us, ometiff became kind of the standard. When we have this xlst transformers we can in a first step whatever we get out of the microscope transform into ometiff metadata and go from there. 
I very much appreciate that you generating this xslt files and making them easy to apply.

elmbeech, 2021-11-16:
Ok. The problem now is that   czi-to-ome-xslt/xslt/czi-to-ome.xsl not transforms exposure time and scene position.
 Namely: 

 ```
o_img = AICSImage(s_ipathfile)
x_root = o_img.metadata
x_root.findall('./Metadata/Information/Image/Dimensions/S/Scenes/Scene/CenterPosition')  # x,y cooordinat x_root.findall('./Metadata/Information/Image/Dimensions/Channels/Channel/ExposureTime')   # divide by 1000000 for ms
```
is not yet translated into 
```
o_img.ome_metadata.images[*].pixels.planes.position_y
o_img.ome_metadata.images[*].pixels.planes.position_y_unit  # pixel
o_img.ome_metadata.images[*].pixels.planes.position_x
o_img.ome_metadata.images[*].pixels.planes.position_x_unit  # pixel
o_img.ome_metadata.images[*].pixels.planes.exposure_time
o_img.ome_metadata.images[*].pixels.planes.exposure_time_unit  # millisecound
```
There is already a plane.xsl file, but it is not used in  czi-to-ome.xsl
+ https://github.com/AllenCellModeling/czi-to-ome-xslt/blob/main/xslt/Plane.xsl
+ https://github.com/AllenCellModeling/czi-to-ome-xslt/blob/main/xslt/czi-to-ome.xsl

JacksonMaxfield, 2021-11-16:
cc @MatteBailey I know you said you wanted to get more info into the translation, are the above attributes part of what you were hoping to add too?