AllenCellModeling/aicsimageio #356: Image Format Readers as Plugins

toloudis, 2021-11-20:
## Use Case
readers for proprietary formats have various licensing requirements which may or may not be compatible with being bundled in aicsimageio.  It could possibly enable more community development of image format plugins if they didn't require pull requests to aicsimageio itself.   

## Solution
Any class conforming to a particular "simple" API can be an aicsimageio Reader.  These classes could be authored and distributed as separate independent python packages.
We would need to work out how plugins register themselves (or are discovered) by core aicsimageio. And also how to identify which file formats they can read.   One technical consideration is how much shared implementation is in our current base Reader class.

## Alternatives
There are many examples of plugin architectures in python (and otherwise) from which we could draw.

toloudis, 2021-11-20:
off the cuff example:
`pip install my_format_reader`
...
`aicsimageio.register_reader(my_format_reader)`
which internally calls something like
`my_format_reader.get_supported_formats` to add itself to aicsimageio.

AetherUnbound, 2021-11-21:
I really like this idea! Here are a few potential options if we wanted to go with an existing plugin system (these were pulled from a talk I remembered from PyCascades 2020[^1]):

- Yapsy (Yet Another Plugin System): http://yapsy.sourceforge.net/
- Plugins project: https://pypi.org/project/Plugins/
- PluginBase: https://github.com/mitsuhiko/pluginbase
- stevedore: https://github.com/openstack/stevedore

[^1]: https://youtu.be/9V7n1SUEl5g?t=618

JacksonMaxfield, 2021-11-21:
I vote we adopt the model of `fsspec` and use their similar interface for registering "plugins" / extensions

eode, 2022-07-09:
Useful official documentation regarding plugins:

https://packaging.python.org/en/latest/guides/creating-and-discovering-plugins/

toloudis, 2022-07-20:
https://github.com/AllenCellModeling/aicsimageio/discussions/372#discussioncomment-3175617

closed 2023-09-26 by SeanLeRoy