AllenCellModeling/aicsimageio #361: imageio 2.11+ breaks tests

MatteBailey, 2021-11-25:
Several tests are currently failing with errors like:
```
RuntimeError: `mp4` can not handle the given uri.
```
This started when we were automatically upgraded to `imageio` 2.11.0. After doing some digging, I believe this issue is related to the regression mentioned in the PR [here](https://github.com/imageio/imageio/pull/680#issuecomment-968006978). I was thinking we could perhaps get around this by passing the string path in our call to `imageio.get_reader()` [here](https://github.com/AllenCellModeling/aicsimageio/blob/main/aicsimageio/readers/default_reader.py#L102), but that unfortunately didn't fix it.

Our path forward is to either pin `imageio` to 2.10.5, or figure out if there's a way to call `imageio.get_reader()` without getting this error. I didn't have time to investigate the latter sufficiently today, so I am just creating this issue so we have context after we get back from the Thanksgiving break.

MatteBailey, 2021-11-30:
@JacksonMaxfield any thoughts on the best way to proceed here?

JacksonMaxfield, 2021-11-30:
See #360. We are pinning it for now. Leaving this issue open as basically: "figure out what is going on with imageio at a later time / until someone else bumps this issue"

FirefoxMetzger, 2021-12-17:
👋 ImageIO dev here. I just came across this by coincidence and thought I leave a note. 

This sounds like you are running into: https://github.com/imageio/imageio/issues/686 As far as I am aware this only affects reading/writing of `bytes` and `BytesIO` via `ffmpeg`. 

Unfortunately, I am still not aware of a good workaround for reading, and it is one of those issues where a (seemingly) niche thing in the library is broken and the fix for it is non-trivial. This means it will require a motivated volunteer to look into it or it will happen after I solved some higher priority issues.

In either case, if you encounter things that break in a release it would help me a lot if you could leave a comment/issue about this over at ImageIO. I currently don't have a good way to gauge how much various features are actually used or how well our CI covers existing use-cases, so reports like this help me both prioritize and build better CI :)

JacksonMaxfield, 2021-12-17:
Hey @FirefoxMetzger thanks for stopping by!

Yes sorry about that. Totally should have opened an issue for it - or at least added to one. This issue has been open long enough that I am starting to forget what effected us. I believe it was the _writing_ of ffmpeg based files but not entirely sure.

JacksonMaxfield, 2022-02-07:
We can now update this thanks to format hinting (hopefully): https://github.com/imageio/imageio/issues/686#issuecomment-1031193498

JacksonMaxfield, 2022-04-19:
We can now truly update: https://github.com/imageio/imageio/issues/686

FirefoxMetzger, 2022-04-20:
I just tested it to be sure. I found a semi-related bug in the process, which will be fixed in https://github.com/imageio/imageio/pull/792. Once that is done, it should indeed work directly (at least through the v3 API).

```python
>>> import imageio.v3 as iio
>>> from pathlib import Path
>>> encoded_video = Path(".test_images/cockatoo.mp4").read_bytes()
>>> isinstance(encoded_video, bytes)
True 
>>> img = iio.imread(encoded_video)
>>> img.shape
(280, 720, 1280, 3)
>>> # (time, height, width, channel)
```

Note that this relies on a new plugin which wraps pyAV, so `av` needs to be installed. Either via `pip install imageio[pyav]` or directly via `pip install av`.

JacksonMaxfield, 2022-04-20:
Thanks @FirefoxMetzger! This helps a ton!

toloudis, 2022-11-15:
is it possible an update to ffmpeg fixed this without having to change imageio version?  our automated builds are working normally again but I haven't looked into it

toloudis, 2023-03-10:
todo reminder: check to see what version we have pinned if any

BrianWhitneyAI, 2023-03-20:
I opened a PR #470 to look into this and was unable to replicate this error with imageio (2.10.5(pinned), 2.11.0,2.12.0, 2.26.0(current)).

PR #470 (2023-03-29): bugfix: removing 2.11.0 threshold
## How did this come about?
 This pull request was opened to resolve #361. At the time of this bug, several tests were failing with:

```RuntimeError: `mp4` can not handle the given uri.```

The quick-fix solution was to pin the dependency to "imageio[ffmpeg]>=2.9.0,<2.11.0".  Returning to these tests, I was unable to replicate these errors, testing with imageio (2.10.5, 2.11.0,2.12.0, 2.26.0(current)). This is likely an indication that this issue was resolved internally by the imageio team.
closed 2023-03-29 by BrianWhitneyAI