AllenCellModeling/aicsimageio #368: import aicsimageio resulting in TypeError

pr4deepr, 2022-01-14:
## System and Software
* aicsimageio Version: 4.5.2
* Python Version: 3.7.12
* Operating System: Linux (Google Colab)

## Description
When importing aicsimageio, I get this error:
`TypeError: load() missing 1 required positional argument: 'Loader'

It appears to be related to the newest version of dask which was just released. 
I've posted an issue there and linking it here just in case
https://github.com/dask/dask/issues/8574

JacksonMaxfield, 2022-01-14:
Hey @pr4deepr nice to see ya around!

While you debug that on the Dask / Distributed side. Could you just keep me posted on what the resolution to this should be?

Do we need to add a "ignore this version of dask / distributed" to the setup.py / deps or something like that?

pr4deepr, 2022-01-14:
Hi @JacksonMaxfield 

 Nice to see you to. Hope you're doing well!

I had to explicitly install, 
`pip install distributed==2022.01.0`
so, maybe in the setup.py files, you could state under requirements to install latest version of
`distributed` or make sure it's above a certain version  `distributed>=2022.01.0` ?

Cheers
Pradeel

JacksonMaxfield, 2022-01-15:
Hmmmm we specifically don't pin `distributed` because we ran into issues before of people needing certain distributed versions :/

pr4deepr, 2022-01-15:
Ahh, ok
This issue appears to be specific to Google Colab.
The version of `distributed` I had on colab was 1.25.3 , which is from ~ 2 years ago.. Perhaps, it's worth having a warning or message if it detects an older version of `distributed`?

JacksonMaxfield, 2022-01-15:
Hm. I think this should actually be reported to xarray. After looking at the stack trace in the dask issue you posted, we are only getting this issue because we import xarray which does some interesting dask / distributed stuff. You import aicsimageio, aicsimageio imports xarray, xarray is the one causing the problem.

closed 2022-01-17 by JacksonMaxfield

pr4deepr, 2022-01-17:
Thanks for letting me know, I'll post it there..