AllenCellModeling/aicsimageio #369: lxml 4.7.1 on pypi is broken for arm64 macOS

psobolewskiPhD, 2022-01-19:
## System and Software
* aicsimageio Version: 4.5.2
* Python Version: 3.9.9
* Operating System: macOS 12.1

## Description

Installing `aicsimageio` works but fails in places because the `lxml` package latest version 4.7.1 from pypi is broken on arm64 (m1) macOS. version 4.6.5 works fine.

## Expected Behavior
*What did you expect to happen instead?*

## Reproduction

Install `aicsimageio`
Open python REPL
```
import lxml.etree
```
This will yield:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dlopen(/Users/piotrsobolewski/Dev/miniforge3/envs/napari-env/lib/python3.9/site-packages/lxml/etree.cpython-39-darwin.so, 0x0002): tried: '/Users/piotrsobolewski/Dev/miniforge3/envs/napari-env/lib/python3.9/site-packages/lxml/etree.cpython-39-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e')), '/usr/local/lib/etree.cpython-39-darwin.so' (no such file), '/usr/lib/etree.cpython-39-darwin.so' (no such file)
```

## Environment
See above

Note: I've reported this upstream in December, but it was marked a duplicate and is still broken so 🤷‍♂️

JacksonMaxfield, 2022-01-19:
Can confirm, this seems like they have a known issue (https://bugs.launchpad.net/lxml/+bug/1913032).

To be honest, the solution to this would be to move them over to cibuildwheel as well... just like we did for aicspylibczi: https://github.com/lxml/lxml/blob/master/.github/workflows/wheels.yml

Who wants to make a PR to lxml :upside_down_face: ?

I will report to lxml bug tracker that they should consider moving over to cibuildwheel. each projects build is a bit different and I am always worried about touching build infra but I will at least tell them it is available. In the meantime, if you wanted to make a PR to this lib that changes the `lxml>=4.6,<5` to simply `lxml>=4.6,<4.7` I will personally allow it :shrug: https://github.com/AllenCellModeling/aicsimageio/blob/main/setup.py#L92

toloudis, 2022-01-19:
if such a pr comes here, please make sure it links to the known issue so we know why the version is constrained

psobolewskiPhD, 2022-01-19:
I *think* it's possible to use:
```
platform_machine!="arm64"'
```
and combine with `<4.7`

JacksonMaxfield, 2022-01-19:
> I _think_ it's possible to use:
> 
> ```
> platform_machine!="arm64"'
> ```
> 
> and combine with `<4.7`

If that is possible, that would be preferred.

psobolewskiPhD, 2022-01-19:
You know me: google and guess! So i'm not sure, but I think it should work?
I will try and find some time to give it a whirl. But if someone knows this stuff and it's obvious, then go for it and let me know to test.

closed 2022-01-22 by psobolewskiPhD