AllenCellModeling/aicsimageio #370: Incorrect unsupported file format error

JacksonMaxfield, 2022-01-21:
When trying to read a LIF file without `readlif` installed, users are receiving this error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/rex/anaconda3/envs/segmentation/lib/python3.7/site-packages/aicsimageio/aics_image.py", line 226, in __init__
    ReaderClass = self.determine_reader(image, **kwargs)
  File "/home/rex/anaconda3/envs/segmentation/lib/python3.7/site-packages/aicsimageio/aics_image.py", line 196, in determine_reader
    f"File extension suggests format: '{format_ext}'. "
aicsimageio.exceptions.UnsupportedFileFormatError: AICSImage does not support the image: 'wb1-56-microscopy.lif'. File extension suggests format: 'lif'. Install extra format dependency with: `pip install aicsimageio[lif]`. See all known format extensions and their extra install name with `aicsimageio.formats.FORMAT_IMPLEMENTATIONS`
```

This suggestion is actually incorrect after the 4.3.0 patch: https://github.com/AllenCellModeling/aicsimageio/releases/tag/v4.3.0

This should be updated to recommend installing `readlif` separately.

BMP-TUD, 2022-02-03:
For me it is the same problem, do you have already something on that? Thank you.

JacksonMaxfield, 2022-02-03:
I can probably add something to fix this into this patch: https://github.com/AllenCellModeling/aicsimageio/pull/367

commit e50335f1 (2022-02-14): Resolve #370, bad recommended format installer
PR #367 (2022-02-14): bugfix/reader-selection-fixes
## Description

Cleans up a bunch of problems with reader selection. Errors are now passed back up to the user under specific cases, specifically `FileNotFoundError` is properly raised before any readers are attempted.

## Pull request recommendations:
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [gh-<number>], adds tiff file format support_

Resolves #352 
Resolves #351 
Resolves #337 
Resolves #334
Resolves #370 

- [x] Provide relevant  [... cut]
commit f8ac5749 (2022-02-14): bugfix/reader-selection-fixes (#367)

* Fix path non reference and file not found as error

* Allow some errors to pass through on reader attempts

* Add file not found test to aicsimage tests

* Handle reader to install key error

* Lint

* Update ome tiff tests for change to error handle

* Resolve multi-error log messages to single

* Try and except for loading files from path

* Resolve #370, bad recommended format installer

* PR resolutions
closed 2022-02-14 by JacksonMaxfield