AllenCellModeling/aicsimageio #373: Installation failing with file not found for aicsimageio[czi]

timdnewman, 2022-01-26:
Hi,

## System and Software
* aicsimageio Version: 4.5.2
* Python Version: 3.10
* Operating System: Windows 10

## Description
aiscimageio installs fine but when we try pip install aiscimageio[czi]<sup>1</sup> it fails with the following error

ERROR: Command errored out with exit status 1:
     command: 'C:\Program Files\Python310\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\******\\AppData\\Local\\Temp\\pip-install-hn2i25y0\\aicspylibczi_984b1879ced24206bf93091d1a973dbe\\setup.py'"'"'; __file__='"'"'C:\\Users\\******\\AppData\\Local\\Temp\\pip-install-hn2i25y0\\aicspylibczi_984b1879ced24206bf93091d1a973dbe\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\******\AppData\Local\Temp\pip-record-tns6ahqq\install-record.txt' --single-version-externally-managed --user --prefix= --compile --install-headers 'C:\Users\******\AppData\Roaming\Python\Python310\Include\aicspylibczi'
         cwd: C:\Users\******\AppData\Local\Temp\pip-install-hn2i25y0\aicspylibczi_984b1879ced24206bf93091d1a973dbe\
    Complete output (21 lines):
    WARNING: The wheel package is not available.
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-3.10
    creating build\lib.win-amd64-3.10\aicspylibczi
    copying aicspylibczi\CziFile.py -> build\lib.win-amd64-3.10\aicspylibczi
    copying aicspylibczi\types.py -> build\lib.win-amd64-3.10\aicspylibczi
    copying aicspylibczi\_version.py -> build\lib.win-amd64-3.10\aicspylibczi
    copying aicspylibczi\__init__.py -> build\lib.win-amd64-3.10\aicspylibczi
    running egg_info
    writing aicspylibczi.egg-info\PKG-INFO
    writing dependency_links to aicspylibczi.egg-info\dependency_links.txt
    writing requirements to aicspylibczi.egg-info\requires.txt
    writing top-level names to aicspylibczi.egg-info\top_level.txt
    reading manifest file 'aicspylibczi.egg-info\SOURCES.txt'
    adding license file 'LICENSE'
    writing manifest file 'aicspylibczi.egg-info\SOURCES.txt'
    running build_ext
    error: [WinError 2] The system cannot find the file specified
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Program Files\Python310\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\******\\AppData\\Local\\Temp\\pip-install-hn2i25y0\\aicspylibczi_984b1879ced24206bf93091d1a973dbe\\setup.py'"'"'; __file__='"'"'C:\\Users\\******\\AppData\\Local\\Temp\\pip-install-hn2i25y0\\aicspylibczi_984b1879ced24206bf93091d1a973dbe\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\******\AppData\Local\Temp\pip-record-tns6ahqq\install-record.txt' --single-version-externally-managed --user --prefix= --compile --install-headers 'C:\Users\******\AppData\Roaming\Python\Python310\Include\aicspylibczi' Check the logs for full command output.
WARNING: You are using pip version 21.2.3; however, version 21.3.1 is available.
You should consider upgrading via the 'C:\Program Files\Python310\python.exe -m pip install --upgrade pip' command.

1. This might not be the exact command but it's not on my machine and I'm doing it from memory. It's the one that comes up on a failed file load.

## Expected Behavior
It installs

## Reproduction
clean install of 3.10
pip install aiscimageio
*try to import a czi in a script*
pip install aiscimageio[czi]

## Environment
Is this a 3.10 issue?

Thanks,

Tim

JacksonMaxfield, 2022-01-26:
Hey @timdnewman!

This is only because we do not yet build on python 3.10.

cc @toloudis think we can just bump all the builds here and on aicsimageio to 3.10?

toloudis, 2022-01-26:
Yes I think we should definitely add 3.10 to aicspylibczi and go from there.

timdnewman, 2022-02-02:
Brilliant thanks.

JacksonMaxfield, 2022-02-28:
Update on this. We are waiting on some upstream dependencies to release 3.10 support.

tlambert03, 2022-02-28:
let me know if any of those upstreams are me! :)

JacksonMaxfield, 2022-02-28:
I think the only one is Jpype for windows. Everything else passes on this PR: https://github.com/AllenCellModeling/aicsimageio/pull/377

JacksonMaxfield, 2022-02-28:
We pinged them / bumped the issue: https://github.com/jpype-project/jpype/issues/1028

But as of today it looks like they are working on it! https://github.com/jpype-project/jpype/issues/1034

JacksonMaxfield, 2022-05-20:
After many months, this is soon to be closed!! Sorry for the delay but the upstreams have been resolved!

closed 2022-05-22 by JacksonMaxfield

JacksonMaxfield, 2022-05-27:
Python 3.10 support is now included in `aicsimageio==4.8.0`

PR #397 (2022-05-22): admin/ship-mypy-type-annotations-drop-py37-add-py310-manage-test-dependencies-separate-for-each-tox-env
## Description

Ships type annotations with the library when installed. Now people can define functions based on our types and mypy will correctly resolve them.

This also massively overhauls the testing portions of our library. Extra readers and writers have been moved to their own subdirs in our tests dirs. And each extra reader and writer is tested independently from one another. This is to make it possible to avoid dependency overlap.

Specifically, this relates to how bfio doesn't suppo [... cut]