AllenCellModeling/aicsimageio #378: Not loading CZI files but install requirement satisfied

timdnewman, 2022-02-18:
## System and Software
* aicsimageio Version:4.5.2
* Python Version:3.8
* Operating System:Ubuntu 18.04 (ish)

## Description
I'm trying to load an image on an azure vm.  It is throwing the CZI loader not installed exception. But it is also reporting the requirement is satisfied

Requirement already satisfied: aicsimageio[czi] in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (4.5.2)
Requirement already satisfied: imagecodecs>=2020.5.30 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (2021.11.20)
Requirement already satisfied: numpy<2,>=1.16 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (1.21.2)
Requirement already satisfied: tifffile>=2021.8.30 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (2022.2.9)
Requirement already satisfied: xmlschema in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (1.9.2)
Requirement already satisfied: dask[array]>=2021.4.1 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (2022.2.0)
Requirement already satisfied: lxml<5,>=4.6 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (4.8.0)
Requirement already satisfied: xarray>=0.16.1 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (0.21.1)
Requirement already satisfied: zarr<3,>=2.6 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (2.11.0)
Requirement already satisfied: wrapt>=1.12 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (1.13.3)
Requirement already satisfied: ome-types>=0.2 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (0.2.10)
Requirement already satisfied: fsspec>=2021.4.0 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (2022.1.0)
Requirement already satisfied: aicspylibczi>=3.0.4; extra == "czi" in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from aicsimageio[czi]) (3.0.5)
Requirement already satisfied: elementpath<3.0.0,>=2.4.0 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from xmlschema->aicsimageio[czi]) (2.4.0)
Requirement already satisfied: pyyaml>=5.3.1 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from dask[array]>=2021.4.1->aicsimageio[czi]) (6.0)
Requirement already satisfied: partd>=0.3.10 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from dask[array]>=2021.4.1->aicsimageio[czi]) (1.2.0)
Requirement already satisfied: packaging>=20.0 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from dask[array]>=2021.4.1->aicsimageio[czi]) (21.3)
Requirement already satisfied: toolz>=0.8.2 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from dask[array]>=2021.4.1->aicsimageio[czi]) (0.11.2)
Requirement already satisfied: cloudpickle>=1.1.1 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from dask[array]>=2021.4.1->aicsimageio[czi]) (2.0.0)
Requirement already satisfied: pandas>=1.1 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from xarray>=0.16.1->aicsimageio[czi]) (1.4.1)
Requirement already satisfied: asciitree in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from zarr<3,>=2.6->aicsimageio[czi]) (0.3.3)
Requirement already satisfied: numcodecs>=0.6.4 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from zarr<3,>=2.6->aicsimageio[czi]) (0.9.1)
Requirement already satisfied: fasteners in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from zarr<3,>=2.6->aicsimageio[czi]) (0.17.3)
Requirement already satisfied: Pint>=0.15 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from ome-types>=0.2->aicsimageio[czi]) (0.18)
Requirement already satisfied: pydantic[email]>=1.0 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from ome-types>=0.2->aicsimageio[czi]) (1.9.0)
Requirement already satisfied: locket in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from partd>=0.3.10->dask[array]>=2021.4.1->aicsimageio[czi]) (0.2.1)
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from packaging>=20.0->dask[array]>=2021.4.1->aicsimageio[czi]) (3.0.4)
Requirement already satisfied: pytz>=2020.1 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from pandas>=1.1->xarray>=0.16.1->aicsimageio[czi]) (2021.3)
Requirement already satisfied: python-dateutil>=2.8.1 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from pandas>=1.1->xarray>=0.16.1->aicsimageio[czi]) (2.8.2)
Requirement already satisfied: typing-extensions>=3.7.4.3 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from pydantic[email]>=1.0->ome-types>=0.2->aicsimageio[czi]) (3.10.0.2)
Requirement already satisfied: email-validator>=1.0.3; extra == "email" in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from pydantic[email]>=1.0->ome-types>=0.2->aicsimageio[czi]) (1.1.3)
Requirement already satisfied: six>=1.5 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from python-dateutil>=2.8.1->pandas>=1.1->xarray>=0.16.1->aicsimageio[czi]) (1.16.0)
Requirement already satisfied: idna>=2.0.0 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from email-validator>=1.0.3; extra == "email"->pydantic[email]>=1.0->ome-types>=0.2->aicsimageio[czi]) (3.3)
Requirement already satisfied: dnspython>=1.15.0 in /azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages (from email-validator>=1.0.3; extra == "email"->pydantic[email]>=1.0->ome-types>=0.2->aicsimageio[czi]) (2.2.0)
https://sasharedqa.blob.core.windows.net/iris-backup/inputs/150mm/10 U2 1.czi
Cleaning up all outstanding Run operations, waiting 300.0 seconds
1 items cleaning up...
Cleanup took 0.06345367431640625 seconds
[stderr]Traceback (most recent call last):
[stderr]  File "infer.py", line 54, in <module>
[stderr]    img = load_image(name)
[stderr]  File "/mnt/azureml/cr/j/75d1a10d79504c2c951fa99d59989d44/exe/wd/chunking_logic.py", line 22, in load_image
[stderr]    return IM.fromarray(AICSImage(p).get_image_data("YX"))
[stderr]  File "/azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages/aicsimageio/aics_image.py", line 226, in __init__
[stderr]    ReaderClass = self.determine_reader(image, **kwargs)
[stderr]  File "/azureml-envs/azureml_6554f305cf6768f9432d65f408b2c9b2/lib/python3.8/site-packages/aicsimageio/aics_image.py", line 192, in determine_reader
[stderr]    raise exceptions.UnsupportedFileFormatError(
[stderr]aicsimageio.exceptions.UnsupportedFileFormatError: AICSImage does not support the image: '10 U2 1.czi'. File extension suggests format: 'czi'. Install extra format dependency with: `pip install aicsimageio[czi]`. See all known format extensions and their extra install name with `aicsimageio.formats.FORMAT_IMPLEMENTATIONS`.
[stderr]

I build the VM using the wheel on Pypi and it looks to me like it worked.

## Expected Behavior
I expected it to pick up the CZI image loader and work. My key question is, is there anything else that could cause the same error, i.e. a silent fail. Missing package etc

## Reproduction
Built an image on top of the prebuilt docker image with 

Parent Image
mcr.microsoft.com/azureml/openmpi4.1.0-cuda11.1-cudnn8-ubuntu18.04:20211221.v1

channels:
[... 8 lines cut ...]
  - torchaudio=0.10.0
  - cudatoolkit=11.1.1
  - nvidia-apex=0.1.0
  - gxx_linux-64
  - pip
  - pip:
      - azureml-sdk
      - Pillow
      - >-
        aicsimageio[czi]
[... cut ...] 09e9efd38b72d246fda40049a791d5e6d60cbcb7eb/aicsimageio-4.5.2-py2.py3-none-any.whl

Then run:
from PIL import Image as IM
from aicsimageio import AICSImage

#optionally include these which doesn't affect the error but does show the requirements are satisfied as above
#import os
#os.system('pip install aicsimageio[czi]')

p = pathlib.Path(filepath)
img = IM.fromarray(AICSImage(p).get_image_data("YX"))

## Environment
This is an azure ml workspace but its just docker so shouldn't make a difference.

timdnewman, 2022-02-18:
Hi All,

There is something else that will cause this error - it not being a valid czi file. So the error is not that it wasn't installing it, but that the file was corrupt and it gave this error instead. I think it was the file is coming from the blobstore. So my bad.

Do you want a separate feature request around a better error - it clearly wasn't a valid CZI and the loader was installed. So perhaps the error should check if the loader is there before suggesting it?

Thanks

Tim

JacksonMaxfield, 2022-02-18:
Hey @timdnewman!

Sorry for the delay! All of us on the dev team are US pacific coast time.

I was just about to say that it may not be an install error but a corrupt file. We just merged a fix for those issues: https://github.com/AllenCellModeling/aicsimageio/pull/367

We just haven't released a new version yet! Likely next week.

If you could, and you have the time, wanna try installing the repo from current main head just to make sure a more appropriate error is returned?

Thanks! And sorry about the confusion in bad error messages!

timdnewman, 2022-02-18:
Ha! No apologies necessary @JacksonMaxfield the tiny price you pay for the magic of open source. Thanks for all the great work.

Unfortunately, I went to the wheel because I was getting the same behavior from:

"aicsimageio[czi] @ git+https://github.com/AllenCellModeling/aicsimageio.git"

This translates to pip install "aicsimageio[czi] @ git+https://github.com/AllenCellModeling/aicsimageio.git" and that is the dev head, right? So I think I tested it already and it was still giving me the same error.

timdnewman, 2022-02-18:
I've just tried it again and it failed to install - that's why I went to the wheel files. Is there a dev wheel I can point it at?

(I'm not sure the Azure machine learning product has a health relationship with git)

timdnewman, 2022-02-18:
No. No - I had an unhealthy relationship with a trailing ".

I've checked back through the logs though and I did have a build that did this and it didn't give me meaningful errors.

timdnewman, 2022-02-21:
Hi @JacksonMaxfield the actual fil is a no on sharing - but I think that is a moot point. When I access the data store by another method that image is good and works. So it has to do with the method I was using to download it.

I managed to extract what it was doing and it is just piping the error 409 into a file then calling it a CZI. 

See attached.

I've actually got my bit working now so this is about giving back, I'm easy.

[dummy.zip](https://github.com/AllenCellModeling/aicsimageio/files/8108013/dummy.zip)

PR #380 (2022-02-22): bugfix/more-info-and-help-on-corrupt-file
## Description

Adding more help / info to the error message returned on AICSImage general image read attempt.
@toloudis stepped through the file reader attempt process and there isn't much we can do to make this error clearer so more documentation is the best path for now.

## Pull request recommendations:
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [gh-<number>], adds tiff file format support_

Resolves #378

- [x] Provide r [... cut]
closed 2022-02-22 by JacksonMaxfield