AllenCellModeling/aicsimageio #390: Loading single Zstack from large timelapse with AICSImage (aicsimageio) is very slow compared to loading same image with CziFile (aicspylibczi)

cfrick13, 2022-03-18:
## System and Software
* aicsimageio Version: aicsimageio==4.6.0
* Python Version: 3.8.12
* Operating System: Windows 10
* aicspylibczi==3.0.5

## Description
*A clear description of the bug*
Loading single Zstack from large timelapse with AICSImage (aicsimageio) is very slow compared to loading same image with CziFile (aicspylibczi). When AICSImage runs it also seems to be filling up my RAM. Perhaps it is loading subblock metadata?

## Expected Behavior
*What did you expect to happen instead?*
I expect the load speeds to be identical...at worst slightly different

## Reproduction
*A minimal example that exhibits the behavior.*
```
import os
import numpy as np
from aicsimageio import AICSImage
from aicspylibczi import CziFile

###############################################################################

# 300 GB time lapse movie
file_path0 = r" ... MicroscopyData\Frick\2022\20220224\ZSD0\AD00000207\20220221_F04_001.czi" #ask chris for full filepath

file_path = '/'.join(file_path0.split('\\')) #make compatible for other OS

############################################
# this block (CziFile) is fast 
############################################

#read with aicspylibczi
czi = CziFile(file_path)

dims = czi.dims
kwargs={}
for dim in dims:
    kwargs[f"{dim}"]=0
print(kwargs)

kwg = {key:value for key,value in kwargs.items() if key not in ['X','Y']}
print(kwg)

img,_ = czi.read_image(**kwg,)
print(img.shape)

############################################
# this block (AICSImage) is incredibly slow 
############################################
scene=9
tframe=0

raw_reader = AICSImage(file_path)
raw_reader.set_scene(scene)
channel_names = raw_reader.channel_names #ChA
dims_dict =  raw_reader.dims #<Dimensions [T: 60, C: 1, Z: 35, Y: 700, X: 700]>
print(dims_dict)

raw_img = raw_reader.get_image_data('ZYX',T=tframe,C=0) #(1, 35, 700, 700)
print(raw_img.shape)
```

## Environment
*Any additional information about your environment*
```

name: lls_viewer
channels:
  - bioconda
  - conda-forge
  - defaults
  - r
dependencies:
  - ca-certificates=2021.10.8=h5b45459_0
  - openssl=3.0.0=h8ffe710_2
  - pip=22.0.3=pyhd8ed1ab_0
  - python=3.8.12=h900ac77_2_cpython
  - python_abi=3.8=2_cp38
  - setuptools=60.9.3=py38haa244fe_0
  - sqlite=3.37.0=h8ffe710_0
  - ucrt=10.0.20348.0=h57928b3_0
  - vc=14.2=hb210afc_6
  - vs2015_runtime=14.29.30037=h902a5da_6
  - wheel=0.37.1=pyhd8ed1ab_0
  - pip:
    - aicsfiles==5.0.0.dev16
    - aicsimageio==4.6.0
    - aicspylibczi==3.0.5
    - alabaster==0.7.12
    - anyio==3.5.0
    - appdirs==1.4.4
    - argon2-cffi==21.3.0
    - argon2-cffi-bindings==21.2.0
    - asciitree==0.3.3
    - asttokens==2.0.5
    - attrs==21.4.0
    - babel==2.9.1
    - backcall==0.2.0
    - backoff==1.11.1
    - black==22.1.0
    - bleach==4.1.0
    - cachey==0.2.1
    - certifi==2021.10.8
    - cffi==1.15.0
    - charset-normalizer==2.0.12
    - click==8.0.4
    - cloudpickle==2.0.0
    - colorama==0.4.4
    - cycler==0.11.0
    - dask==2022.2.0
    - debugpy==1.5.1
    - decorator==5.1.1
    - defusedxml==0.7.1
    - dnspython==2.2.0
    - docstring-parser==0.13
    - docutils==0.17.1
    - elementpath==2.4.0
    - email-validator==1.1.3
    - entrypoints==0.4
    - executing==0.8.2
    - fasteners==0.17.3
    - fonttools==4.29.1
    - freetype-py==2.2.0
    - fsspec==2022.2.0
    - heapdict==1.0.1
    - hsluv==5.0.2
    - idna==3.3
    - imagecodecs==2022.2.22
    - imageio==2.16.0
    - imagesize==1.3.0
    - importlib-metadata==4.11.1
    - importlib-resources==5.4.0
    - intervaltree==3.1.0
    - ipykernel==6.9.1
    - ipython==8.0.1
    - ipython-genutils==0.2.0
    - jedi==0.18.1
    - jinja2==3.0.3
    - json5==0.9.6
    - jsonschema==4.4.0
    - jupyter-client==7.1.2
    - jupyter-core==4.9.2
    - jupyter-server==1.13.5
    - jupyterlab==3.2.9
    - jupyterlab-pygments==0.1.2
    - jupyterlab-server==2.10.3
    - kiwisolver==1.3.2
    - labkey==1.4.1
    - lkaccess==1.4.25
    - locket==0.2.1
    - lxml==4.8.0
    - magicgui==0.3.7
    - markupsafe==2.1.0
    - matplotlib==3.5.1
    - matplotlib-inline==0.1.3
    - mistune==0.8.4
    - mypy-extensions==0.4.3
    - napari==0.4.14
    - napari-console==0.0.4
    - napari-plugin-engine==0.2.0
    - napari-svg==0.1.6
    - nbclassic==0.3.5
    - nbclient==0.5.11
    - nbconvert==6.4.2
    - nbformat==5.1.3
    - nest-asyncio==1.5.4
    - networkx==2.6.3
    - notebook==6.4.8
    - npe2==0.1.2
    - numcodecs==0.9.1
    - numpy==1.22.2
    - numpydoc==1.2
    - ome-types==0.2.10
    - packaging==21.3
    - pandas==1.4.1
    - pandocfilters==1.5.0
    - parso==0.8.3
    - partd==1.2.0
    - pathspec==0.9.0
    - pickleshare==0.7.5
    - pillow==9.0.1
    - pint==0.18
    - platformdirs==2.5.1
    - pooch==1.6.0
    - prometheus-client==0.13.1
    - prompt-toolkit==3.0.28
    - psutil==5.9.0
    - psygnal==0.3.3
    - pure-eval==0.2.2
    - pyaml==20.4.0
    - pycparser==2.21
    - pydantic==1.9.0
    - pygments==2.11.2
    - pyopengl==3.1.6
    - pyparsing==3.0.7
    - pyqt5==5.15.6
    - pyqt5-qt5==5.15.2
    - pyqt5-sip==12.9.1
    - pyrsistent==0.18.1
    - python-dateutil==2.8.2
    - pytomlpp==1.0.10
    - pytz==2021.3
    - pywavelets==1.2.0
    - pywin32==303
    - pywinpty==1.1.6
    - pyyaml==6.0
    - pyzmq==22.3.0
    - qtconsole==5.2.2
    - qtpy==2.0.1
    - requests==2.27.1
    - scikit-image==0.19.2
    - scipy==1.8.0
    - send2trash==1.8.0
    - six==1.16.0
    - sniffio==1.2.0
    - snowballstemmer==2.2.0
    - sortedcontainers==2.4.0
    - sphinx==4.4.0
    - sphinxcontrib-applehelp==1.0.2
    - sphinxcontrib-devhelp==1.0.2
    - sphinxcontrib-htmlhelp==2.0.0
    - sphinxcontrib-jsmath==1.0.1
    - sphinxcontrib-qthelp==1.0.3
    - sphinxcontrib-serializinghtml==1.1.5
    - stack-data==0.2.0
    - superqt==0.3.0
    - terminado==0.13.1
    - testpath==0.6.0
    - tifffile==2022.2.9
    - tomli==2.0.1
    - toolz==0.11.2
    - tornado==6.1
    - tqdm==4.62.3
    - traitlets==5.1.1
    - typer==0.4.0
    - typing-extensions==4.1.1
    - urllib3==1.26.8
    - vispy==0.9.6
    - wcwidth==0.2.5
    - webencodings==0.5.1
    - websocket-client==1.2.3
    - wrapt==1.13.3
    - xarray==0.21.1
    - xmlschema==1.9.2
    - zarr==2.11.0
    - zipp==3.7.0

```

cfrick13, 2022-03-18:
found an error in my code (was only loading a single Zslice with aicspylibczi) but the issue still stands.
here is the corrected code
import os
import numpy as np
from aicsimageio import AICSImage
from aicspylibczi import CziFile

###############################################################################

# 300 GB time lapse movie
file_path0 = r" ... MicroscopyData\Frick\2022\20220224\ZSD0\AD00000207\20220221_F04_001.czi" #ask chris for full filepath

file_path = '/'.join(file_path0.split('\\')) #make compatible for other OS

############################################
# this block (CziFile) is fast 
############################################

#read with aicspylibczi
czi = CziFile(file_path)

dims = czi.dims
kwargs={}
for dim in dims:
    kwargs[f"{dim}"]=0
print(kwargs)

kwg = {key:value for key,value in kwargs.items() if key not in ['X','Y','Z']}
print(kwg)

img,_ = czi.read_image(**kwg,)
print(img.shape)

############################################
# this block (AICSImage) is incredibly slow 
############################################
scene=9
tframe=0

raw_reader = AICSImage(file_path)
raw_reader.set_scene(scene)
channel_names = raw_reader.channel_names #ChA
dims_dict =  raw_reader.dims #<Dimensions [T: 60, C: 1, Z: 35, Y: 700, X: 700]>
print(dims_dict)

raw_img = raw_reader.get_image_data('ZYX',T=tframe,C=0) #(1, 35, 700, 700)
print(raw_img.shape)

JacksonMaxfield, 2022-03-18:
Short and sweet? are you trying to load a single z plane at a time?

JacksonMaxfield, 2022-03-18:
Or rather, are you trying to load a single t frame?

JacksonMaxfield, 2022-03-18:
Because if so, please see the README again:

> The `.data` and `.xarray_data` properties will load the whole scene into memory. The `.get_image_data` function will load the whole scene into memory and then retrieve the specified chunk.

What you should do instead is:

```python
delayed_chunk = img.get_image_dask_data("ZYX", T=0, C=0)
in_memory_chunk = delayed_chunk.compute()
```

cfrick13, 2022-03-18:
I just tested this and it works beautifully! Thanks, Jackson!

closed 2022-03-18 by JacksonMaxfield

cfrick13, 2022-03-18:
I did not catch that comment in the README previously. Not sure why I didn't see it...