AllenCellModeling/aicsimageio #391: ValueError: shape mismatch

jo-mueller, 2022-03-23:
## System and Software
* aicsimageio Version: 4.6.4
* Python Version: 3.8.12
* Operating System: Windows 10

## Description
Hi all, I am receiving an error when I try to open a common .czi-image, such as [this one ](https://sharing.biotec.tu-dresden.de/index.php/s/YE6d04uLH71pjbO) (Credit to @Theresa-S for the data). I open it with the following code:

```Python
from aicsimageio import AICSImage
img = AICSImage(r"D:\P2E_GFAP-Nestin__0001.czi")  # selects the first scene found
img.shape
```

and receive the following error:
```

  File "C:\Users\johamuel\AppData\Local\Temp\ipykernel_1184\4260751373.py", line 1, in <cell line: 1>
    img.shape

  File "C:\Users\johamuel\Anaconda3\envs\devbio-napari-env\lib\site-packages\aicsimageio\aics_image.py", line 533, in shape
    return self.xarray_dask_data.shape

  File "C:\Users\johamuel\Anaconda3\envs\devbio-napari-env\lib\site-packages\aicsimageio\aics_image.py", line 414, in xarray_dask_data
    self.reader.mosaic_xarray_dask_data

  File "C:\Users\johamuel\Anaconda3\envs\devbio-napari-env\lib\site-packages\aicsimageio\readers\reader.py", line 410, in mosaic_xarray_dask_data
    self._mosaic_xarray_dask_data = self._get_stitched_dask_mosaic()

  File "C:\Users\johamuel\Anaconda3\envs\devbio-napari-env\lib\site-packages\aicsimageio\readers\czi_reader.py", line 777, in _get_stitched_dask_mosaic
    return self._construct_mosaic_xarray(self.dask_data)

  File "C:\Users\johamuel\Anaconda3\envs\devbio-napari-env\lib\site-packages\aicsimageio\readers\czi_reader.py", line 730, in _construct_mosaic_xarray
    stitched = self._stitch_tiles(

  File "C:\Users\johamuel\Anaconda3\envs\devbio-napari-env\lib\site-packages\aicsimageio\readers\czi_reader.py", line 710, in _stitch_tiles
    ans[tuple(ans_indexes)] = data[tuple(data_indexes)]

  File "C:\Users\johamuel\Anaconda3\envs\devbio-napari-env\lib\site-packages\dask\array\core.py", line 1828, in __setitem__
    dsk = setitem_array(out, self, key, value)

  File "C:\Users\johamuel\Anaconda3\envs\devbio-napari-env\lib\site-packages\dask\array\slicing.py", line 1868, in setitem_array
    raise ValueError(

ValueError: shape mismatch: value array of shape (2236, 2236) could not be broadcast to indexing result of shape (2236, 586)
```

## Expected Behavior
Data cannot be provided by the loaded `img` object (commands like `img.data`, `img.get_dask_data()` all fail with the above error.

I assume that aicsimageio tries to chunk up the image somehow but fails to reproduce the original image dimensions in case the original image size is some odd number (here: `18495 x 18495)`.

JacksonMaxfield, 2022-03-23:
Ugh. More mosaic bugs. So sorry about this @jo-mueller.

We know there is a Zeiss created tiling API coming soon so hopefully when that comes out we can resolve a whole host of these issues :shrug: 

---

Do you need it prestitched or can you handle the stitching yourself? If so I would turn off `reconstruct_mosaic`

```python
img = AICSImage("...", reconstruct_mosaic=False)
```

I wonder how m2stitch deals with this... https://github.com/AllenCellModeling/aicsimageio/issues/383 :thinking:

jo-mueller, 2022-03-30:
Hey @JacksonMaxfield ,

thanks for the fast reply. The spatial relation of single mosaic tiles to the whole image is quite important on this one, but the proplem is not super-urgent. If Zeiss brings out some major updates I think it's feasible to wait for the new API to fix these issues rather than self-engineering a fix that may not be worthwhile. 

Do you mind if I keep this issue open or could you ping me when there's any updates on this?

BrianWhitneyAI, 2023-03-28:
We have added a new maintenance feature to clean up stale issues and PRs. Adding this comment to set a baseline for ‘Stale’

BrianWhitneyAI, 2023-04-04:
I beleive that the example file's link has expired and I cant seem to replicate this bug with an alternative file.  Feel free to reopen the issue!