AllenCellModeling/aicsimageio #392: caching of s3 files

satra, 2022-03-28:
## System and Software
* aicsimageio Version: 4.6.4
* Python Version: 3.9.10
* Operating System: ubuntu 20.04

## Description
*A clear description of the bug*

i'm using the following snippet to read a file from an s3 bucket and then i pass it onto a 
neuroglancer viewer via python inside a jupyter notebook. This works the first time i do 
this but not the second time. The first time it provides the appropriate data range, but 
the second time it says all values are 0. I suspect this has something to do with s3fs 
caching, but i don't think i have direct access to the s3fs object to invalidate the cache.

```
s3_url = 's3://dandiarchive/blobs/817/ae2/817ae22f-9e54-410e-8d5a-b392d489bfb4'
img = AICSImage(s3_url)
ds = img.xarray_dask_data
```

the url refers to an ome.tiff file.

## Expected Behavior
*What did you expect to happen instead?*

it should work multiple times, not just the first time.

## Reproduction
*A minimal example that exhibits the behavior.*

```
from aicsimageio import AICSImage
import neuroglancer

s3_url = 's3://dandiarchive/blobs/817/ae2/817ae22f-9e54-410e-8d5a-b392d489bfb4'
img = AICSImage(s3_url)
ds = img.xarray_dask_data

viewer = neuroglancer.Viewer()
# This volume handle can be used to notify the viewer that the data has changed.
volume = neuroglancer.LocalVolume(
    ds,
    dimensions=neuroglancer.CoordinateSpace(
        names=['t', 'c', 'z', 'y', 'x'],
        units=['', '', 'um', 'um', 'um'],
        scales=[1, 1, img.physical_pixel_sizes.Z, img.physical_pixel_sizes.Y, img.physical_pixel_sizes.X],
    ),
    voxel_offset=[0,0,0,0,0]
    )
with viewer.txn() as s:
    s.layers['volume'] = neuroglancer.ImageLayer(
        source=volume,
        # Define a custom shader to display this mask array as red+alpha.
        shader="""
#uicontrol invlerp normalized(range=[0,3500], window=[0,65500], clamp=true)
void main() {
  emitGrayscale(normalized());
}
""",
        
    )
    s.position = np.array(ds.shape)/2
    s.display_dimensions = ["x", "y", "z"]
    s.dimensions = volume.dimensions
    s.show_slices = False

print("Neuroglancer URL:", viewer.get_viewer_url())
```

## Environment
*Any additional information about your environment*

i'm running this on a jupyterhub. the new image is rebuilding, but should be available as `dandiarchive/dandihub:latest`

JacksonMaxfield, 2022-03-28:
Hey @satra! Sorry you are experiencing the issue but hopefully we can get it working quickly!

Quick questions:
1. I am assuming this is an OME-TIFF file? Or just raw TIFF? If not, what file format?
2. Can the image be loaded into memory / how large is the image? Why use dask data? If using a dask cluster have you tried [`persist`ing the data to the cluster](https://docs.dask.org/en/stable/generated/dask.array.Array.persist.html)?
3. Can you provide a minimal example without neuroglancer? I would have to dig around but I could have sworn we had tests where we called `dask_data.compute` multiple times and then do an array equals comparison but that may have only been on a local read and not an s3 read.

satra, 2022-03-29:
thank you @JacksonMaxfield. i will try to not pull you into a rabbit hole, especially if it seems like it's more of a neuroglancer specific issue.

> I am assuming this is an OME-TIFF file? Or just raw TIFF? If not, what file format?

indeed this is an ome.tiff file, but a non-pyramidal one.

```python
import tifffile
import s3fs

fs = s3fs.S3FileSystem(anon=True)
f = fs.open(s3_url, 'rb')
tiff = tifffile.TiffFile(f)
print(tiff.is_ome)
meta = tiff.ome_metadata
```

> Can the image be loaded into memory / how large is the image? 

in this particular case yes (600MB compressed, becomes 1.5 GB uncompressed), but after this initial case is solved we are hoping to scale up with some stitching across many of these files. (will also be looking into kerchunk for this).

> Why use dask data? If using a dask cluster have you tried [persisting the data to the cluster](https://docs.dask.org/en/stable/generated/dask.array.Array.persist.html)?

my hope was the neuroglancer would only fetch the necessary pieces as the user navigates the view, which in turn would do any necessary fetches via s3fs (i'm assuming this is under the hood). i have a different zarr dataset, which i can pass to neuroglancer and it simply loads as necessary. i was hoping to achieve the same for this ome.tiff dataset. 

> Can you provide a minimal example without neuroglancer? I would have to dig around but I could have sworn we had tests where we called dask_data.compute multiple times and then do an array equals comparison but that may have only been on a local read and not an s3 read.

i'm seeing the issue only when the two intersect, hence trying to figure out the weirdness of the issue. if i do `np.array(ds.max())` it will return the relevant value or if i pass `ds.compute()` or `ds.data` to neuroglancer. so somewhere between the dask layer and reuse of the same url results in the issue.

JacksonMaxfield, 2022-03-29:
Thanks for the extra info!

I am really not sure what to say. I just wrote this test script and for each read, the in-memory arrays were equal:

```python
from aicsimageio import AICSImage
import numpy as np

print("Init AICSImage")
img = AICSImage(
    "s3://allencell/aics/pipeline_integrated_cell/fovs/"
    "000801ac_3500002379_100X_20181005_3-Scene-26-aligned_cropped-P85-G07.ome.tiff"
)
print("Completed init")

print(f"img dims: {img.dims}")

print("Pulling first")
first_get = img.xarray_dask_data.data.compute()
print(f"first_get shape: {first_get.shape}")

print("Pulling second")
second_get = img.xarray_dask_data.data.compute()
print(f"second_get shape: {second_get.shape}")

print(f"arrays are equal: {np.array_equal(first_get, second_get)}")

print("Pulling third")
third_get = img.xarray_dask_data.data.compute()
print(f"third_get shape: {third_get.shape}")

print(f"arrays are equal: {np.array_equal(first_get, third_get)}")
```

This seems like it may be a neuroglancer issue? I really have no idea. Any chance we can ping the neuroglancer devs?

JacksonMaxfield, 2022-05-18:
This issue is pretty stale and I could never figure out if this was an AICSImageIO issue or not. Closing.

closed 2022-05-18 by JacksonMaxfield

satra, 2022-05-18:
thanks @JacksonMaxfield - yes i couldn't figure it out either.