AllenCellModeling/aicsimageio #393: local variable 'check_selection_max' referenced before assignment

gummz, 2022-03-30:
## System and Software
* aicsimageio Version: 4.6.3
* Python Version: 3.8.2
* Operating System: Ubuntu 20.04

## Description
```
File "/appl/python/3.8.2/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/appl/python/3.8.2/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/appl/python/3.8.2/lib/python3.8/cProfile.py", line 196, in <module>
    main()
  File "/appl/python/3.8.2/lib/python3.8/cProfile.py", line 189, in main
    runctx(code, globs, None, options.outfile, options.sort)
  File "/appl/python/3.8.2/lib/python3.8/cProfile.py", line 19, in runctx
    return _pyprofile._Utils(Profile).runctx(statement, globals, locals,
  File "/appl/python/3.8.2/lib/python3.8/profile.py", line 62, in runctx
    prof.runctx(statement, globals, locals)
  File "/appl/python/3.8.2/lib/python3.8/cProfile.py", line 100, in runctx
    exec(cmd, globals, locals)
  File "../src/data/make_dataset_test.py", line 60, in <module>
    timepoints = data.get_image_dask_data(
  File "/zhome/e2/e/154260/cell/venv_1/lib/python3.8/site-packages/aicsimageio/aics_image.py", line 620, in get_image_dask_data
    return transforms.reshape_data(
  File "/zhome/e2/e/154260/cell/venv_1/lib/python3.8/site-packages/aicsimageio/transforms.py", line 235, in reshape_data
    if check_selection_max > data.shape[dim_index]:
UnboundLocalError: local variable 'check_selection_max' referenced before assignment
```

## Expected Behavior
*What did you expect to happen instead?*
I was subsetting my data via the `T`, `C`, and `Z` arguments.

## Environment
*Any additional information about your environment*

JacksonMaxfield, 2022-03-30:
Wow. This is surprising... I can't remember the last time we had a bug in that function. Can I get what exactly your arguments were for `get_image_dask_data`? And the dims of your image?

> Would check_selection_max=0 in transforms.py above line 188 do the trick?

I think you may want to call `max` on whatever tuple or range you provided?

JacksonMaxfield, 2022-03-30:
You really just need to find the max value of whatever slice, range, tuple, list, etc of the chunk you are selecting.

gummz, 2022-03-30:
The function call was as follows:

`data.get_image_dask_data(
        'TZYX', T=time_idx, C=0)`

where `time_idx=np.random.randint(0, 280, 10)`. The dims of the image are (ZYX): (38, 1024, 1024)

Do I understand correctly that passing `T=[10,20,30]` will return timepoint indices 10, 20, and 30?

JacksonMaxfield, 2022-03-30:
> I was trying to pass in a list of indices. Do I understand correctly that passing T=[10,20,30] will return timepoint indices 10, 20, and 30?

I believe that is correct.

JacksonMaxfield, 2022-04-01:
Had a quick second to look more in-depth. This isn't actually a bug. We just don't support providing ArrayLike as a param, just a list, which is not the same. So it's just not reaching the list clause.

Replication of your bug then solution by adding a `tolist()` to the end of the `np.random.randint`.

```python
In [1]: import numpy as np

In [2]: from aicsimageio import AICSImage

In [3]: r = AICSImage(np.random.rand(280, 38, 20, 20), dim_order="TZYX").get_image_data("TZYX", T=np.random.randint(0, 280, 10), C=0)
---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
<ipython-input-3-52d4f98d672e> in <module>
----> 1 r = AICSImage(np.random.rand(280, 38, 20, 20), dim_order="TZYX").get_image_data("TZYX", T=np.random.randint(0, 280, 10), C=0)

~/active/cell/aicsimageio/aicsimageio/aics_image.py in get_image_data(self, dimension_order_out, **kwargs)
    696 
    697         # Transform and return
--> 698         return transforms.reshape_data(
    699             data=self.data,
    700             given_dims=self.dims.order,

~/active/cell/aicsimageio/aicsimageio/transforms.py in reshape_data(data, given_dims, return_dims, **kwargs)
    233 
    234         # Check that fixed integer request isn't outside of request
--> 235         if check_selection_max > data.shape[dim_index]:
    236             raise IndexError(
    237                 f"Dimension specified with {dim}={display_dim_spec} "

UnboundLocalError: local variable 'check_selection_max' referenced before assignment

In [4]: r = AICSImage(np.random.rand(280, 38, 20, 20), dim_order="TZYX").get_image_data("TZYX", T=np.random.randint(0, 280, 10).tolist(), C=0)

In [5]: r.shape
Out[5]: (10, 38, 20, 20)
```

closed 2022-04-01 by JacksonMaxfield