AllenCellModeling/aicsimageio #395: Transpose dims to OME standard as a part of normal OME-TIFF writing options

JacksonMaxfield, 2022-04-04:
https://forum.image.sc/t/imsave-tifffile-imwrite-returns-tiff-files-of-wrong-shape-in-imagej-cellprofiler/65332/9?u=jacksonmaxfield

The `dim_order` parameter should be used as the parameter to `aicsimageio.transforms.reshape_data` with `TCZYX` as the return order (optional S)

toloudis, 2022-04-04:
I'm a little bit torn on this one.  It's debatable whether we should be transposing people's data on them.  I would greatly prefer if they do the necessary transposes on their end.  One reason for this is that OME-XML allows any permutation of [T,C,Z]YX as long as YX are the last two.  What if a user really _intends_ ZCTYX instead of TCZYX? I suppose if we wanted the writer to be smarter it could look for Y and X and auto-transpose them to the end of the list.  Thoughts?

toloudis, 2022-04-05:
I would also find it acceptable if we just state that OmeTiffWriter ONLY writes TCZYX order and will shuffle your data into that order if you tell us what dims you are passing in.  But it seems restrictive over the actual options that the underlying spec supports.

JacksonMaxfield, 2022-04-05:
That's entirely fair. I think this does affect downstream read a _tiny_ bit. So totally agree that we may not want to shuffle the data too much. I think I like the idea of simply always moving YX(S) to the end though.

toloudis, 2022-04-05:
so rather than being totally general, we can have some logic like: if X is not the last entry, then find X dim index and transpose to get X as last entry.  if Y is not second-to-last entry, then transpose to get Y into that position. That can probably be done with a single transpose step fairly easily.  Same goes for S if present. (Although I'm less certain about S handling)

closed 2023-05-05 by github-actions[bot]