AllenCellModeling/aicsimageio #399: Support for Pyramidal OME Tiffs?

rjgiedt, 2022-04-27:
Hi,

First, thank you for the very helpful product. 

Secondly, I wanted to ask about support for creating multi-channel, pyramidal tile images in the OME.Tiff format. With tifffile, pyramidal formats are generally done via describing a tile parameter (i.e. tile=(512, 512) etc.) but tifffile to my knowledge can not complete this operation as it isn't capable of writing subIFDs. In the documentation for ome.tiff writer I don't see any "pyramidal formatting"/ tile functionality, but I wanted to check and see if there is already a way in aics to construct these images or if not, if there were plans to support this in the future?  Happy to provide additional information, example arrays etc. if that's helpful.  Thanks so much!

JacksonMaxfield, 2022-04-27:
Hey @rjgiedt! :wave: 

I have two things in progress for you. Not sure if either is a perfect match for what you are looking for but potentially useful.

First is the `OmeTiledTiffWriter` from @Nicholas-Schaub / the `bfio` crew. This will write out tiled tiffs but I am not sure if it will write out pyramids attached to it... On that note, sorry @Nicholas-Schaub, I will try to review that PR today!!

Second, we already have some in progress work to add an OME-Zarr Writer #381 and Zarr is looking like the next generation format for large tiled data.

Our review process is generally pretty long but I will try to have an update for you at some point. Maybe @Nicholas-Schaub can point to a better resource too.

cgohlke, 2022-04-27:
> tifffile to my knowledge can not complete this operation as it isn't capable of writing subIFDs

That should work: https://github.com/cgohlke/tifffile/blob/8fdbf302292782f000c11231dd885a0afb6c0a75/tifffile/tifffile.py#L585-L595

JacksonMaxfield, 2022-04-27:
@cgohlke always has something ready :heart:

rjgiedt, 2022-04-28:
Hey guys, nice to meet both of you and thank you for the quick replies! 

Just to clarify, what I'm trying to do is basically take a numpy array in the format of (18, 1500, 2000) ie. (Channels, Ydim, Xdim) and convert this into a pyramidal ome.tiff. I'm positive I'm a little mixed up on how the subIFDs work in such a case, but just wanted to clarify that @cgohlke 's solution, while perfect for a typical RGB image, probably won't be applicable without additional modification for the multi-channel microscope images I'm working with? 

Again, really appreciate your thoughts and time!

Nicholas-Schaub, 2022-04-28:
`tifffile` is generally going to be your best best, as @cgohlke has already referenced :)

We do have a pretty optimized OME TIFF writer, but it doesn't support pyramids yet. We have hired a full time developer to really flesh out the details we haven't implemented yet, so it should be ready by the end of the year.

Nicholas-Schaub, 2022-04-28:
Actually, we do have a solution for OME Zarr pyramids. We just haven't wrapped our reader/writer for that yet. That should be coming in the next couple weeks.

rjgiedt, 2022-04-28:
Great Nicholas, really appreciate the update and I'll look into OME Zarr as I don't have familiarity with that format. Unfortunately I'm trying to put ome.tiffs into a 3rd party software package which likely won't have ome.zarr compatibility, but I'll put in a word with our vendor to attempt to gain compatibility if that is the direction the field seems to be moving in.

Again, very much appreciate your help.

closed 2022-04-28 by rjgiedt

cgohlke, 2022-04-28:
> just wanted to clarify that @cgohlke 's solution, while perfect for a typical RGB image, probably won't be applicable without additional modification for the multi-channel microscope images I'm working with?

Do you have an example that fails? It should just work with any dimensions that the OME model supports. The only dimensions that can be scaled are X and Y, the other dimensions must be constant for all resolutions.

rjgiedt, 2022-04-28:
Hi @cgohlke,

Thank you very much for your help. I was able to confirm that your code does work on my data, albeit via the creation of a huge file (I've went from ~ 1 GB to 18 in this process). The only issue I'll have from this is needing fully validated ome xml data which I'll need to look further into here... 

Thanks again!