AllenCellModeling/aicsimageio #400: set_scene does not change scene when scene ids are duplicated

jil24, 2022-05-11:
## System and Software
* aicsimageio Version: 4.7.0
* Python Version 3.10.4
* Operating System: Mac OS 12.3.1 / M1

## Description
When loading a file via bioformats (a leica LOF) with multiple scenes which are named identically, changing the scene via scene index does not work; no error occurs, but the active scene does not change.

I needed to update the bio formats jar to a more recent version manually since the packaged version does not have LOF support. I have no ability to control scene naming.

```
>>> import aicsimageio
>>> img = aicsimageio.AICSImage('example.lof', reader=aicsimageio.readers.bioformats_reader.BioformatsReader)
>>> img.scenes
('example-c1', 'example-c1', 'example-c1', 'example-c1')
>>> img.current_scene_index
0
img.set_scene(3)
img.current_scene_index
0
```

JacksonMaxfield, 2022-05-11:
I think this is because the function is simply misnamed? `set_series` instead of `set_scene`? cc @tlambert03 should I just quickly swap the function names?

JacksonMaxfield, 2022-05-11:
https://github.com/AllenCellModeling/aicsimageio/blob/main/aicsimageio/readers/bioformats_reader.py#L331

JacksonMaxfield, 2022-05-11:
We have tests but they only check shape 😬 https://github.com/AllenCellModeling/aicsimageio/blob/main/aicsimageio/tests/readers/test_bioformats_reader.py#L385

tlambert03, 2022-05-11:
> cc @tlambert03 should I just quickly swap the function names?

works for me!

toloudis, 2022-05-11:
maybe run_image_file_checks should test scene index too?

jil24, 2022-05-11:
So this has nothing to do with my scene/series name duplication at all?

JacksonMaxfield, 2022-05-11:
At the current moment, I do not believe so because you are providing an index and not name.

jil24, 2022-05-11:
Hmm, I tried making the changes in those commits; there is no difference in the behavior

JacksonMaxfield, 2022-05-11:
Correction: this is likely because bioformats itself does not support multiple images (read: scenes in an AICSImageIO context) for Leica LOF: https://docs.openmicroscopy.org/bio-formats/6.9.1/supported-formats.html

jil24, 2022-05-11:
Yet I can load multiple scenes via bio formats into FIJI from the same file?

JacksonMaxfield, 2022-05-11:
cc @joshmoore / someone from bioformats? Is the table out of date? Are LOF files actually supported?

dgault, 2022-05-12:
Support for XLEF and LOF files was added in 6.9.0 as an external contribution from Leica Microsystems. So it should be supported if you are using the latest 6.9.1 release. The LOF reader will only read individual files
(https://docs.openmicroscopy.org/bio-formats/6.9.1/formats/leica-lof.html), whereas the XLEF reader will read the entire dataset (https://docs.openmicroscopy.org/bio-formats/6.9.1/formats/leica-xlef.html). So if you have an XLEF file you may want to point at it when opening the dataset.

JacksonMaxfield, 2022-05-19:
@tlambert03 would it be possible that the bioformats jar that ships with the python package simply doesn't have the latest version? Do you package that manually or is it auto pulled / built?

tlambert03, 2022-05-19:
needs updating.  but I've dragged my feet on that, since we should really be using scyjava. (see https://github.com/tlambert03/bioformats_jar/issues/2)

I'll try to get on it soon

tlambert03, 2022-05-19:
(it's '6.7.0')

JacksonMaxfield, 2022-05-19:
> Support for XLEF and LOF files was added in 6.9.0 as an external contribution from Leica Microsystems. So it should be supported if you are using the latest 6.9.1 release.

> [bioformats_jar is] '6.7.0'

Well now I am confused... It is opening a file format that it didn't officially support at that point

JacksonMaxfield, 2022-05-19:
> needs updating. but I've dragged my feet on that, since we should really be using scyjava. (see [tlambert03/bioformats_jar#2](https://github.com/tlambert03/bioformats_jar/issues/2))

It never ceases to amaze me how complicated some issues are. As per usual, thanks for all you do

JacksonMaxfield, 2022-05-20:
I have found the bug and updated the PR. @jil24 could you attempt to install and use the branch again. There is a single bug that is maybe related but I will look into next week.

closed 2022-05-24 by JacksonMaxfield

JacksonMaxfield, 2022-05-27:
This should now be fixed and released. `pip install --upgrade aicsimageio` should install 4.8.0

PR #401 (2022-05-24): bugfix/pass-series-index-in-biofile-init
## Description

~Somehow slipped through multiple layers of testing and review. The `set_scene` function in the `BioformatsReader` was named `set_series` (which makes sense from the Bioformats API but not consistent with AICSImageIO standard.~

Above was incorrect. This PR now actually fixes this. The only problem was that the current scene index of the Reader wasn't being passed down to the `BioFile` object init during reads.

It also adds a parameter to our multi scene checking function that will check if the data between the two comparison scenes is  [... cut]