AllenCellModeling/aicsimageio #416: Unable to run compute() on 2017 czi files

smishra3, 2022-07-19:
## System and Software
* aicsimageio Version: 4.7.0
* Python Version:
* Operating System:

## Description
*A clear description of the bug*
Can't run this code 

reader = AICSImage("/allen/aics/assay-dev/MicroscopyData/Irina/2017/20171106/20171106_I02_002(2).czi")
img1_t = reader.get_image_dask_data("ZYX", T=0, C=0).compute()

## Expected Behavior
*What did you expect to happen instead?*

## Reproduction
*A minimal example that exhibits the behavior.*
reader = AICSImage("/allen/aics/assay-dev/MicroscopyData/Irina/2017/20171106/20171106_I02_002(2).czi")
img1_t = reader.get_image_dask_data("ZYX", T=0, C=0).compute()

work around is to explicitly use reader.set_scene(1) before compute()

## Environment
*Any additional information about your environment*

toloudis, 2022-07-19:
Here's the issue.
CZI files can have scenes enumerated by index and the indices don't have to start at 0.
AICSImageIO uses zero-based indexing for its `set_scene` api.  So we were missing a step to offset that value when we pass it down to the lower level code that actually will read from the czi file.

toloudis, 2022-07-19:
For example, if the czi file reports that it contains scenes 1,2,3, then when you `set_scene(0)` using aicsimageio, you are supposed to get scene 1 according to the czi file's internals.  The bug was that we were just literally sending the `0` in and there IS no scene 0 as far as the czi file is concerned.

evamaxfield, 2022-07-19:
Interesting... So in the file that found the bug, when I asked for `reader.scenes` and it looked like there were 6 scenes in the file, are those all there just with different offsets?

toloudis, 2022-07-28:
I believe this is now fixed by #417

closed 2022-08-02 by evamaxfield