AllenCellModeling/aicsimageio #428: SLDY file format reader

toloudis, 2022-08-09:
## Use Case

SLDY is a newer 3i slidebook file format which AICS will be using.  From my understanding it is a chunked format stored as numpy arrays.  Not sure yet if the metadata format is documented or where.

## Solution

SLDYReader class

cfrick13, 2022-08-09:
I've done a tiny bit of work with the metadata (yaml files in the img directory) looking for pixel sizes
[image]

evamaxfield, 2022-08-09:
Thanks @cfrick13 I think that hits most of the metadata we want? Only problem is: "what do we return from the `metadata` property of the `AICSImage` or `Reader` objects. I am tempted to say a list of files....?? Or a single big dictionary with the file names as the keys:

```json
{
   "ChannelRecord": {...},
   "ImageRecord": {...},
}
```

:woman_shrugging: :woman_shrugging:

toloudis, 2022-08-09:
I kinda like the big dictionary idea

cfrick13, 2022-08-09:
a big dictionary would be cool as a starting point.

closed 2023-05-18 by SeanLeRoy