AllenCellModeling/aicsimageio #430: can't open valid scene from czi file

cfrick13, 2022-08-18:
## System and Software
* aicsimageio Version: 4.9.1 
* Python Version: 3.9
* Operating System: Windows or Linux

## Description
the image data for a valid scene can't get opened. AICSImage seems to believe the scene index is 28 when I've specified it to be 14. 

VERY IMPORTANT! this code (below) works when I run it with a previous verison of `aicsimageio` (an old environment I created)
If I try to install an earlier version of `aicsimageio` in this current working environment, then I get an error saying that I need to install `aicspylibczi>=3.0.5` (even though I have repeatedly reinstalled aicspylibczi>=3.0.5)

## Expected Behavior
the image should load as the scene is valid

## Reproduction
*A minimal example that exhibits the behavior.*
```import os
from aicsimageio import AICSImage
import sys
print(sys.version_info)
filename = '{contactChrisForFullPath}500000724/ZSD1/Round 6/5500000724_Round_6_20X_P1_to_P16.czi'
print(os.path.abspath(filename))

reader = AICSImage(filename)
T=0
ci=0
reader.set_scene(14)
delayed_chunk = reader.get_image_dask_data("ZYX", T=T, C=ci)

print(reader.current_scene)

print(reader.current_scene_index)

imgstack = delayed_chunk.compute()

print(imgstack.shape)
```

output
```
{contactChrisForFullPath}5500000724\ZSD1\Round 6\5500000724_Round_6_20X_P1_to_P16.czi
P15-F7
14
Traceback (most recent call last):
  File "{contactChris}Assessment\4iProcessing-\multiplex_immuno_processing\exectue_round_alignment_test.py", line 18, in <module>
    imgstack = delayed_chunk.compute()
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\base.py", line 315, in compute
    (result,) = compute(self, traverse=False, **kwargs)
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\base.py", line 598, in compute
    results = schedule(dsk, keys, **kwargs)
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\threaded.py", line 89, in get
    results = get_async(
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\local.py", line 511, in get_async
    raise_exception(exc, tb)
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\local.py", line 319, in reraise
    raise exc
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\local.py", line 224, in execute_task
    result = _execute_task(task, data)
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\core.py", line 119, in _execute_task
    return func(*(_execute_task(a, cache) for a in args))
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\core.py", line 119, in <genexpr>
    return func(*(_execute_task(a, cache) for a in args))
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\core.py", line 119, in _execute_task
    return func(*(_execute_task(a, cache) for a in args))
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\core.py", line 119, in <genexpr>
    return func(*(_execute_task(a, cache) for a in args))
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\core.py", line 119, in _execute_task
    return func(*(_execute_task(a, cache) for a in args))
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\core.py", line 119, in <genexpr>
    return func(*(_execute_task(a, cache) for a in args))
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\core.py", line 119, in _execute_task
    return func(*(_execute_task(a, cache) for a in args))
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\core.py", line 119, in <genexpr>
    return func(*(_execute_task(a, cache) for a in args))
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\core.py", line 119, in _execute_task
    return func(*(_execute_task(a, cache) for a in args))
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\dask\utils.py", line 71, in apply
    return func(*args, **kwargs)
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\aicsimageio\readers\czi_reader.py", line 281, in _read_chunk_from_image
    return CziReader._get_image_data(
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\aicsimageio\readers\czi_reader.py", line 330, in _get_image_data
    data, dims = czi.read_image(**read_dims)
  File "C:\Users\chris.frick\AppData\Local\Continuum\anaconda3\envs\frick_multiplex_test\lib\site-packages\aicspylibczi\CziFile.py", line 552, in read_image
    image, shape = self.reader.read_selected(plane_constraints, m_index, cores)
_aicspylibczi.PylibCZI_CDimCoordinatesOverspecifiedException: The coordinates are overspecified = you have specified a Dimension or Dimension value that is not valid. S value 28invalid, ∉ [0, 16)
```

## Environment
*Any additional information about your environment*

```
aicsimageio==4.9.1
aicspylibczi==3.0.5
asciitree==0.3.3
cloudpickle==2.1.0
dask==2022.8.0
dnspython==2.2.1
elementpath==2.5.3
email-validator==1.2.1
entrypoints==0.4
fasteners==0.17.3
fsspec==2022.7.1
idna==3.3
imagecodecs==2022.8.8
imageio==2.21.1
locket==1.0.0
lxml==4.9.1
networkx==2.8.5
numcodecs==0.10.2
numpy==1.23.2
ome-types==0.3.0
packaging==21.3
pandas==1.4.3
partd==1.3.0
Pillow==9.2.0
Pint==0.19.2
pydantic==1.9.2
pyparsing==3.0.9
python-dateutil==2.8.2
pytz==2022.2.1
PyWavelets==1.3.0
PyYAML==6.0
resource-backed-dask-array==0.1.0
scikit-image==0.19.3
scipy==1.9.0
six==1.16.0
tifffile==2022.8.12
toolz==0.12.0
typing_extensions==4.3.0
wrapt==1.14.1
xarray==2022.6.0
xmlschema==1.11.3
zarr==2.12.0
```

cfrick13, 2022-08-18:
note: to install an earlier version of `aicsimageio` to test this I had to do the following:
```
pip uninstall aicsimageio
pip install aicsimageio~=4.7.0

pip uninstall fsspec
pip install fsspec=2022.05.0
```
but 4.7.0 was successful in opening this image on python 3.9

toloudis, 2022-08-19:
I will take a look at this asap.  (Was on vacation this week) 
First check: 
What does 
`reader.scenes`
return?

toloudis, 2022-08-22:
scenes:
('P1-F6', 'P2-F6', 'P3-F6', 'P4-F6', 'P5-F6', 'P6-F6', 'P7-F6', 'P8-F6', 'P9-F6', 'P10-F6', 'P11-F7', 'P12-F7', 'P13-F7', 'P14-F7', 'P15-F7', 'P16-F7', 'P17-F7', 'P18-F7', 'P19-F7', 'P20-F7', 'P21-F8', 'P22-F8', 'P23-F8', 'P24-F8', 
'P25-F8', 'P26-F8', 'P27-F8', 'P28-F8', 'P29-F8', 'P30-F8', 'P31-F9', 'P32-F9', 'P33-F9', 'P34-F9', 'P35-F9', 'P36-F9', 'P37-F9', 'P38-F9', 'P39-F9', 'P40-F9')

toloudis, 2022-08-22:
tracked it down - this is a regression from a very recent other bug fix.  working on a fix now

toloudis, 2022-08-22:
This is a case in which not every scene in the file has the same dims.  I don't think we have a multiscene czi file in the test suite that has that property. @evamaxfield not sure if you know whether that is true?  Anyway I have a fix here without unit tests.

evamaxfield, 2022-08-22:
> @evamaxfield not sure if you know whether that is true? 

To my knowledge we do not a file that meets those conditions in the test suite. If we can get a tiny test file that would be great.

toloudis, 2022-08-22:
Possible future work: use the new pylibczirw to generate test czi?

closed 2022-08-22 by evamaxfield

cfrick13, 2022-08-29:
thanks for fixing this while I was away!