AllenCellModeling/aicsimageio #431: Unable to import .CZI after update to 4.9.1

psobolewskiPhD, 2022-08-21:
## System and Software
* aicsimageio Version: 4.9.1
* Python Version: 3.9
* Operating System: macOS 12.4 (arm64)

## Description
```
In [1]: from pathlib import Path

In [2]: import aicsimageio

In [3]: path = Path(Path.home() / "Dev/napari-aicsimageio/napari_aicsimageio/tests/resources/s_1_t_1_c_1_z_1.czi")

In [4]: img = aicsimageio.AICSImage(path)
```
In 4.9.0 this works fine (and napari-aicsimageio works)
After updating to 4.9.1 this results in an error:
```
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
Input In [3], in <cell line: 1>()
----> 1 img = aicsimageio.AICSImage(path)

NameError: name 'path' is not defined

In [4]: path = Path("/Users/piotrsobolewski/Dev/napari-aicsimageio/napari_aicsim
   ...: ageio/tests/resources/s_1_t_1_c_1_z_1.czi")

In [5]: img = aicsimageio.AICSImage(path)
21-Aug-22 13:50:21 - aicsimageio.aics_image - WARNING  - Attempted file (/Users/piotrsobolewski/Dev/napari-aicsimageio/napari_aicsimageio/tests/resources/s_1_t_1_c_1_z_1.czi) load with reader: aicsimageio.readers.czi_reader.CziReader failed with error: '_io.BufferedReader' object has no attribute 'f'
---------------------------------------------------------------------------
UnsupportedFileFormatError                Traceback (most recent call last)
Input In [5], in <cell line: 1>()
----> 1 img = aicsimageio.AICSImage(path)

File ~/Dev/miniforge3/envs/apple-TF29/lib/python3.9/site-packages/aicsimageio/aics_image.py:264, in AICSImage.__init__(self, image, reader, reconstruct_mosaic, fs_kwargs, **kwargs)
    254 def __init__(
    255     self,
    256     image: types.ImageLike,
   (...)
    260     **kwargs: Any,
    261 ):
    262     if reader is None:
    263         # Determine reader class and create dask delayed array
--> 264         ReaderClass = self.determine_reader(image, fs_kwargs=fs_kwargs, **kwargs)
    265     else:
    266         # Init reader
    267         ReaderClass = reader

File ~/Dev/miniforge3/envs/apple-TF29/lib/python3.9/site-packages/aicsimageio/aics_image.py:218, in AICSImage.determine_reader(image, fs_kwargs, **kwargs)
    216 if readers[0] in READER_TO_INSTALL:
    217     installer = READER_TO_INSTALL[readers[0]]
--> 218     raise exceptions.UnsupportedFileFormatError(
    219         "AICSImage",
    220         path,
    221         msg_extra=(
    222             f"File extension suggests format: '{format_ext}'. "
    223             f"Install extra format dependency with: "
    224             f"`pip install {installer}`. "
    225             f"See all known format extensions and their "
    226             f"extra install name with "
    227             f"`aicsimageio.formats.FORMAT_IMPLEMENTATIONS`. "
    228             f"If the extra dependency is already installed this "
    229             f"error may have raised because the file is "
    230             f"corrupt or similar issue. For potentially more "
    231             f"information and to help debug, try loading the file "
    232             f"directly with the desired file format reader "
    233             f"instead of with the AICSImage object."
    234         ),
    235     )
    236 else:
    237     raise exceptions.UnsupportedFileFormatError(
    238         "AICSImage",
    239         path,
    240     )

UnsupportedFileFormatError: AICSImage does not support the image: '/Users/piotrsobolewski/Dev/napari-aicsimageio/napari_aicsimageio/tests/resources/s_1_t_1_c_1_z_1.czi'. File extension suggests format: 'czi'. Install extra format dependency with: `pip install aicspylibczi>=3.0.5`. See all known format extensions and their extra install name with `aicsimageio.formats.FORMAT_IMPLEMENTATIONS`. If the extra dependency is already installed this error may have raised because the file is corrupt or similar issue. For potentially more information and to help debug, try loading the file directly with the desired file format reader instead of with the AICSImage object.
```
and likewise `napari-aicsimageio` fails to import.

## Expected Behavior
Import the file, just as with 4.9.0

## Reproduction
See above

## Environment
No changes other than updating to 4.9.1, same `aicspylibczi`

evamaxfield, 2022-08-21:
We have been having issues with this since an fsspec update. Can you let us know which version of fsspec you are using / was installed?

psobolewskiPhD, 2022-08-21:
fsspec 2022.5.0

psobolewskiPhD, 2022-08-21:
Updating fsspec (to 2022.7.1) resolved the issue.

evamaxfield, 2022-08-22:
Got it. Thanks for debugging. I think I know a patch to fix.

pr4deepr, 2022-09-02:
Apologies for hijacking this thread as it seems to be a fsspec issue. 

SPECS:
- Windows 10 machine, with Miniconda
- aicsimageio 4.9.4
- python 3.9.13
- fsspec 2022.5.0

I'm getting another error when opening czi files.. 

> WARNING Attempted file ([C:/RBC_tiny.czi](https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/blob/master/sample_data/RBC_tiny.czi)) load with reader: aicsimageio.readers.czi_reader.CziReader failed with error: '_io.BufferedReader' object has no attribute 'f'

Upgrading fsspec to 2022.8.2 fixed it.

evamaxfield, 2022-09-19:
I think we have encountered enough of these bug reports to warrant a change of the dependency minimal version to be bumped to 2022.8.0.

closed 2023-03-23 by SeanLeRoy