AllenCellModeling/aicsimageio #434: AICSImage Initialization with 6 Dimensions (RGB)

MosGeo, 2022-08-28:
## System and Software
* aicsimageio Version: 4.9.2
* Python Version: 3.10
* Operating System: Windows 11 64Bit

## Description
This is a 'bug' related to the fuzzy world of RGB  images and the six dimension of AICSImage :)
  
The following works as expected
```python
data = da.random.random((1, 1,1,512,512))*256
data = data.astype(int)
img = AICSImage(data)
````
However, this following raises an error `ValueError: different number of dimensions on data and dims: 6 vs 1`
```python
data = da.random.random((1, 1,1,512,512,3))*256
data = data.astype(int)
img = AICSImage(data)
````

## Expected Behavior
Image is created.

## Reproduction
```python
from aicsimageio.aics_image import AICSImage
import dask.array as da

data = da.random.random((1, 1,1,512,512,3))*256
data = data.astype(int)
img = AICSImage(data)
```

evamaxfield, 2022-08-29:
@MosGeo we are all a bit busy right now. Feel free to make a PR!

MosGeo, 2022-08-30:
@evamaxfield no worries, I will check it out. Thanks!

@toloudis dims didn't help.

SeanLeRoy, 2023-03-14:
I opened a PR to ideally fix this without having to change your sample code, but for what it is worth it this does not raise an error for (I added the `dim_order` parameter).

Using: 
* python 3.10
* most current aicsimagio (at time of writing) 4.9.4
* MacOS
```
from aicsimageio.aics_image import AICSImage
import dask.array as da

data = da.random.random((1, 1,1,512,512,3))*256
data = data.astype(int)
img = AICSImage(data, dim_order="TCZYXS")
```

closed 2023-03-21 by SeanLeRoy