AllenCellModeling/aicsimageio #436: Default tiff reader missing planes (with specific file)

tlambert03, 2022-09-12:
## System and Software
* aicsimageio Version:  4.9.2
* Python Version:  3.10.5
* Operating System:  macos

## Description
Not sure what it is about this file, and I haven't looked too deeply at the metadata yet, but the default `TiffReader` class seems to miss the z planes in the linked file, whereas the BioformatsReader and tiffile both see it:

https://www.dropbox.com/s/ls6ci1cxyocqbgr/2MF1P2_glia.tif?dl=0

```python
In [28]: im = AICSImage('2MF1P2_glia.tif')

In [29]: bim = AICSImage('2MF1P2_glia.tif', reader=BioformatsReader)

In [30]: tim = tf.imread('2MF1P2_glia.tif')

In [31]: im.shape
Out[31]: (1, 1, 1, 1024, 1024)

In [32]: bim.shape
Out[32]: (343, 1, 1, 1024, 1024)

In [33]: tim.shape
Out[33]: (343, 1024, 1024)
```

and to confirm, they're also not detected as scenes:

```python
In [38]: im.scenes
Out[38]: ('Image:0',)
```

## Expected Behavior
I expected a shape of 343 planes

cgohlke, 2022-09-13:
Interesting case. I think `(1, 1, 1, 1024, 1024)` might be the expected shape for dimension order `'TCZYX'`. The file `2MF1P2_glia.tif` was written by `SCIFIO`. It uses a [provisional, undocumented metadata format](https://github.com/scifio/scifio/blob/b16216ab74929803ab5127e0919b42c7ad146432/src/main/java/io/scif/formats/TIFFFormat.java#L1690-L1726), which is not supported by Bio-Formats or tifffile:

```
SCIFIO=0.42.0
axes=X,Y,Unknown
lengths=1024,1024,343
scales=1.0,1.0,1.0
units=null,null,null
bitsPerPixel=8
images=343
channels=1
slices=1
frames=1
hyperstack=true
mode=composite
unit=null
```

Bio-Formats handles the unknown axis as `time`, while tifffile uses the generic axis `"I"`, which is pruned by aicsimageio...

toloudis, 2022-09-13:
I haven't dived into the code around this yet, but aicsimageio definitely wants to put the planes with their proper semantic dimension, with as little guessing as possible.  That's a big reason why `"I"` gets pruned.

Maybe there is a way for us to say if slices=1 (Z) and channels=1 (C) then the only thing left is T.  Speculating because the implementation wraps tifffile so those may not be exposed in the way i'm suggesting.

evamaxfield, 2022-11-15:
This should be fixed by #445 and will be released in 4.9.3 shortly.

closed 2022-11-15 by evamaxfield