AllenCellModeling/aicsimageio #441: czi files with missing data crash with segfault

tlambert03, 2022-10-08:
## System and Software
* aicsimageio Version:  `main (4.9.2)`
* Python Version: 3.10
* Operating System: macos

## Description
Someone sent me the file linked below wondering why it crashes their script when they open with AICSImage:

https://www.dropbox.com/s/kl25qr3bkae9l0q/demo.czi?dl=0

It appears that the experiment might have been stopped in the middle:  Out of 4 timepoints, all but the last have 15 z planes, but the last timepoint only has 8 zplanes.

when you access `.data`, it results in a segfault.  If you use `.dask_data` then you get a slightly more useful error saying that an array of (8, 512, 512) can't be broadcast into one of (15, 512, 512)...

The same image opens fine in BioFormats/Fiji ... where it appears that they simply add frames of zero to fill out the missing shape.

## Expected Behavior
*What did you expect to happen instead?*

ideally: missing data would be filled in with zeros as is done with Bioformats?
minimally, avoiding a segfault would be great for in-memory data, and for delayed data, the existing message is more or less interpretable for me, but for a casual user of aicsimageio, an error that indicates the bug is in the data itself might be nice.

moli42, 2022-11-27:
Hi! I encountered the same segmentation fault as well, I solved by using `np.array(.dask_data)`, hope it can help you.

aicsimageio Version: 4.9.3
python version: 3.8.13
os: centOS

BrianWhitneyAI, 2023-03-28:
We have added a new maintenance feature to clean up stale issues and PRs. Adding this comment to set a baseline for ‘Stale’

BrianWhitneyAI, 2023-04-05:
Hey! I have been looking into this issue. When attempting to replicate this bug I didn't get an error using `.data` or `.dask_data`. when looking at the dimension specifics of the image I see:

Bytes | 60.00 MiB | 60.00 MiB
-- | -- | --
Shape| (4, 2, 15, 512, 512) | (4, 2, 15, 512, 512)
Dtype| uint16 numpy.ndarray|uint16 numpy.ndarray

Specifically, when looking through timepoints the reported dimension I get are (15, 512, 512) (ZYX) where higher z stacks are filled with zeros. I believe this is the expected behavior you were looking for! There is some possibility that this error was due to an old version of aicspylibczi. I am going to close this issue for now, feel free to reopen it.

closed 2023-04-05 by BrianWhitneyAI