AllenCellModeling/aicsimageio #444: installation error on Mac M1

jxchen01, 2022-11-10:
## System and Software
* aicsimageio Version: latest as of Nov 10, 2022
* Python Version: 3.9
* Operating System: MacOS 12.5 (with Apple M1 chip)

## Description
`pip install aicsimageio` failed.

```
running build_ext
      building 'imagecodecs._aec' extension
      creating build/temp.macosx-11.1-arm64-cpython-39
      creating build/temp.macosx-11.1-arm64-cpython-39/imagecodecs
      clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /opt/homebrew/Caskroom/miniforge/base/envs/val/include -arch arm64 -I/opt/homebrew/Caskroom/miniforge/base/envs/val/include -fPIC -O2 -isystem /opt/homebrew/Caskroom/miniforge/base/envs/val/include -arch arm64 -Iimagecodecs -I/opt/homebrew/Caskroom/miniforge/base/envs/val/include/python3.9 -I/opt/homebrew/Caskroom/miniforge/base/envs/val/lib/python3.9/site-packages/numpy/core/include -c imagecodecs/_aec.c -o build/temp.macosx-11.1-arm64-cpython-39/imagecodecs/_aec.o
      In file included from imagecodecs/_aec.c:746:
      In file included from /opt/homebrew/Caskroom/miniforge/base/envs/val/lib/python3.9/site-packages/numpy/core/include/numpy/arrayobject.h:5:
      In file included from /opt/homebrew/Caskroom/miniforge/base/envs/val/lib/python3.9/site-packages/numpy/core/include/numpy/ndarrayobject.h:12:
      In file included from /opt/homebrew/Caskroom/miniforge/base/envs/val/lib/python3.9/site-packages/numpy/core/include/numpy/ndarraytypes.h:1948:
      /opt/homebrew/Caskroom/miniforge/base/envs/val/lib/python3.9/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: "Using deprecated NumPy API, disable it with "          "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-W#warnings]
      #warning "Using deprecated NumPy API, disable it with " \
       ^
      imagecodecs/_aec.c:756:10: fatal error: 'libaec.h' file not found
      #include "libaec.h"
               ^~~~~~~~~~
      1 warning and 1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> imagecodecs

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

## Expected Behavior
Not failure

## Reproduction
keep failing when doing pip install on fresh conda environment.

## Environment
*Any additional information about your environment*

toloudis, 2022-11-10:
At first glance, the error is coming from a dependency, `imagecodecs`.  Probably a version mismatch or recent incompatibility.  I'll see if I can find more info.

jxchen01, 2022-11-10:
```
Collecting aicsimageio
  Using cached aicsimageio-4.9.2-py2.py3-none-any.whl (119 kB)
Collecting xmlschema
  Using cached xmlschema-2.1.1-py3-none-any.whl (354 kB)
Collecting xarray>=0.16.1
  Using cached xarray-2022.11.0-py3-none-any.whl (963 kB)
Collecting tifffile>=2021.8.30
  Using cached tifffile-2022.10.10-py3-none-any.whl (210 kB)
Collecting numpy<2,>=1.16
  Using cached numpy-1.23.4-cp39-cp39-macosx_11_0_arm64.whl (13.4 MB)
Collecting imagecodecs>=2020.5.30
  Using cached imagecodecs-2022.9.26.tar.gz (13.5 MB)
  Preparing metadata (setup.py) ... done
Collecting resource-backed-dask-array>=0.1.0
  Using cached resource_backed_dask_array-0.1.0-py2.py3-none-any.whl (8.0 kB)
Collecting zarr<3,>=2.6
  Using cached zarr-2.13.3-py3-none-any.whl (191 kB)
Collecting wrapt>=1.12
  Using cached wrapt-1.14.1-cp39-cp39-macosx_11_0_arm64.whl (35 kB)
Collecting ome-types>=0.2
  Using cached ome_types-0.3.1-py3-none-any.whl (142 kB)
Collecting fsspec!=2022.7.0,>=2021.4.0
  Using cached fsspec-2022.11.0-py3-none-any.whl (139 kB)
Collecting lxml<5,>=4.6
  Using cached lxml-4.9.1-cp39-cp39-macosx_11_0_arm64.whl
Collecting dask[array]>=2021.4.1
  Using cached dask-2022.10.2-py3-none-any.whl (1.1 MB)
Collecting click>=7.0
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting toolz>=0.8.2
  Using cached toolz-0.12.0-py3-none-any.whl (55 kB)
Collecting cloudpickle>=1.1.1
  Using cached cloudpickle-2.2.0-py3-none-any.whl (25 kB)
Collecting packaging>=20.0
  Using cached packaging-21.3-py3-none-any.whl (40 kB)
Collecting pyyaml>=5.3.1
  Using cached PyYAML-6.0-cp39-cp39-macosx_11_0_arm64.whl (173 kB)
Collecting partd>=0.3.10
  Using cached partd-1.3.0-py3-none-any.whl (18 kB)
Collecting Pint>=0.15
  Using cached Pint-0.20.1-py3-none-any.whl (269 kB)
Collecting pydantic[email]>=1.0
  Using cached pydantic-1.10.2-cp39-cp39-macosx_11_0_arm64.whl (2.6 MB)
Collecting xmlschema
  Using cached xmlschema-1.11.3-py3-none-any.whl (356 kB)
Collecting typing-extensions
  Using cached typing_extensions-4.4.0-py3-none-any.whl (26 kB)
Collecting pandas>=1.3
  Using cached pandas-1.5.1-cp39-cp39-macosx_11_0_arm64.whl (10.9 MB)
Collecting elementpath<3.0.0,>=2.5.0
  Using cached elementpath-2.5.3-py3-none-any.whl (181 kB)
Collecting numcodecs>=0.10.0
  Using cached numcodecs-0.10.2-cp39-cp39-macosx_11_0_arm64.whl
Collecting asciitree
  Using cached asciitree-0.3.3-py3-none-any.whl
Collecting fasteners
  Using cached fasteners-0.18-py3-none-any.whl (18 kB)
Collecting entrypoints
  Using cached entrypoints-0.4-py3-none-any.whl (5.3 kB)
Collecting pyparsing!=3.0.5,>=2.0.2
  Using cached pyparsing-3.0.9-py3-none-any.whl (98 kB)
Collecting python-dateutil>=2.8.1
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Collecting pytz>=2020.1
  Using cached pytz-2022.6-py2.py3-none-any.whl (498 kB)
Collecting locket
  Using cached locket-1.0.0-py2.py3-none-any.whl (4.4 kB)
Collecting email-validator>=1.0.3
  Using cached email_validator-1.3.0-py2.py3-none-any.whl (22 kB)
[... 159 lines cut ...]
      clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O2 -Wall -fPIC -O2 -isystem /opt/homebrew/Caskroom/miniforge/base/envs/vals/include -arch arm64 -fPIC -O2 -isystem /opt/homebrew/Caskroom/miniforge/base/envs/vals/include -arch arm64 -Iimagecodecs -I/opt/homebrew/Caskroom/miniforge/base/envs/vals/include/python3.9 -I/opt/homebrew/Caskroom/miniforge/base/envs/vals/lib/python3.9/site-packages/numpy/core/include -c imagecodecs/_aec.c -o build/temp.macosx-11.0-arm64-cpython-39/imagecodecs/_aec.o
      In file included from imagecodecs/_aec.c:746:
      In file included from /opt/homebrew/Caskroom/miniforge/base/envs/vals/lib/python3.9/site-packages/numpy/core/include/numpy/arrayobject.h:5:
      In file included from /opt/homebrew/Caskroom/miniforge/base/envs/vals/lib/python3.9/site-packages/numpy/core/include/numpy/ndarrayobject.h:12:
      In file included from /opt/homebrew/Caskroom/miniforge/base/envs/vals/lib/python3.9/site-packages/numpy/core/include/numpy/ndarraytypes.h:1948:
      /opt/homebrew/Caskroom/miniforge/base/envs/vals/lib/python3.9/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: "Using deprecated NumPy API, disable it with "          "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-W#warnings]
      #warning "Using deprecated NumPy API, disable it with " \
       ^
      imagecodecs/_aec.c:756:10: fatal error: 'libaec.h' file not found
      #include "libaec.h"
               ^~~~~~~~~~
      1 warning and 1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> imagecodecs

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

```

jxchen01, 2022-11-10:
here is the full error message. On the same laptop, just an older conda environment I built several months ago, I have aicsimageio installed. It seems like something wrong on my local environment ...

jxchen01, 2022-11-10:
I have found the solution:

I need to pre-install imagecodecs with conda first (pre-install imagecodecs with pip still not good).  `conda install -c conda-forge imagecodecs`. Then, pip install aicsimageio works fine

closed 2022-11-10 by jxchen01