AllenCellModeling/aicsimageio #447: current_scene_index() always return 0 when scene_id's are identical across scene

LegendLukaz, 2022-11-22:
## System and Software
* aicsimageio Version: 4.9.3
* aicspylibczi Version: 3.0.5
* fsspec Version: 2022.11.0
* Python Version: 3.9.15
* Operating System: Windows 10

## Description
When opening czi files with unnamed scenes, the `scene_id` would be `None`, and thus we would obtain a list of `None` that is indistinguishable between scenes. This means that the function to obtain current scene index (`AICSImage.current_scene_index()`) will always return scene `0`. Same behavior can be observed when scenes have identical names.

This is the code snippet that obtains the name of the scene (if available):
https://github.com/AllenCellModeling/aicsimageio/blob/f7896662fe3190f7be1992365dcae7f0a17717e4/aicsimageio/readers/czi_reader.py#L185
In the described situation, `.get("name")` would return `None` as the `"Name"` attribute is missing from the scene's metadata.

This is the code snippet that obtains the current scene index for the image:
https://github.com/AllenCellModeling/aicsimageio/blob/f7896662fe3190f7be1992365dcae7f0a17717e4/aicsimageio/aics_image.py#L332
This works fine when `self.current_scene` is uniquely labelled, but when it is identical, the code here will always return the first match. This means that later scenes with the same name would NOT be accessible. In the above case, all the scenes are labelled `None`, thus this will always return the index `0`.

## Expected Behavior

`AICSImage.current_scene_index` should return the current selected scene as set by `AICSImage.set_scene()` regardless of how scenes are named.

I.e.
```
img = AICSImage(...)
print(f"Before: {img.current_scene_index}")
img .set_scene(2)
print(f"After: {img.current_scene_index}")
```
should give
```
Before: 0
After: 2
```
even when `img.scenes == (None, None)` or `img.scenes == ('a', 'a')`

## Reproduction
Example CZI File: https://www.dropbox.com/s/xhoroq5kfgcwo8h/NoSceneNames.czi?dl=0

Code:
```Python
from pathlib import Path
from aicsimageio import AICSImage

test_img = AICSImage(Path(...*path to the czi file*...))

# List of None scene names
print(f"Scene names: {test_img.scenes}")

# Current index should be 0 by default
print(f"Current scene index: {test_img.current_scene_index()}")

# Change scene index to 2
test_img.set_scene(2)

# Current index is still 0
print(f"Current scene index: {test_img.current_scene_index}")
```
Output:
```
Scene names: (None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None)
Current scene index: 0
Current scene index: 0
```

## Environment
The code was ran in conda environment with the following items installed:

<details>
  <summary>conda list</summary>
[... 116 lines cut ...]
sqlite                    3.39.4               hcfcfb64_0    conda-forge
svt-av1                   1.3.0                h63175ca_0    conda-forge
tbb                       2021.6.0             h91493d7_1    conda-forge
tifffile                  2022.10.10               pypi_0    pypi
toml                      0.10.2             pyhd8ed1ab_0    conda-forge
tomli                     2.0.1              pyhd8ed1ab_0    conda-forge
toolz                     0.12.0                   pypi_0    pypi
typing_extensions         4.4.0              pyha770c72_0    conda-forge
tzdata                    2022f                h191b570_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
vc                        14.3                 h3d8a991_9    conda-forge
vs2015_runtime            14.32.31332          h1d6e394_9    conda-forge
wheel                     0.38.4             pyhd8ed1ab_0    conda-forge
wrapt                     1.14.1                   pypi_0    pypi
x264                      1!164.3095           h8ffe710_2    conda-forge
x265                      3.5                  h2d74725_3    conda-forge
xarray                    2022.11.0                pypi_0    pypi
xmlschema                 1.11.3                   pypi_0    pypi
xz                        5.2.6                h8d14728_0    conda-forge
zarr                      2.13.3                   pypi_0    pypi
zipp                      3.10.0             pyhd8ed1ab_0    conda-forge
zstd                      1.5.2                h7755175_4    conda-forge
  ```

</details>

LegendLukaz, 2022-11-22:
I have thought up two different solutions to this problem right now:

## Solution 1:  Name `None` scenes by scene index and name of file

In `czi_read.py`, we fill in the name of the scene with the scene's index and the file's name

## Solution 2: Reference `_current_scene_index` in `Reader` instead

`_current_scene_index` is set directly by `Reader.set_scene()`, thus circumvent the scene naming issue, but this requires exposing private memeber of the class which is not desired. The better implementation would be to cache the new scene index in `AICSImage.set_scene()` once we checked that `reader.set_scene()` did not raise errors.

Regardless, solution 1 seems to be a needed feature anyways as scene id having identical names is not a desired behavior.

evamaxfield, 2022-11-22:
Hey @LegendLukaz!

First: Sorry you are experiencing the issue. This one is interesting to me because I thought we had pretty well tested our scene management but it continues to be an annoyance.

Second: Thanks for putting together a wonderful issue with reproducible example.

Third: I can try to take a look later today!

LegendLukaz, 2022-11-22:
@evamaxfield thanks for the prompt reply! #448 is my attempt at fixing the issue, though still need to cover the identical name cases. I will continue at it tomorrow.

BrianWhitneyAI, 2023-03-28:
We have added a new maintenance feature to clean up stale issues and PRs. Adding this comment to set a baseline for ‘Stale’

SeanLeRoy, 2023-04-05:
Closed from #477