AllenCellModeling/aicsimageio #450: Process TIFF Tags for Metadata on TiffReader

evamaxfield, 2022-12-08:
## Use Case
*Please provide a use case to help us understand your request in context*

https://forum.image.sc/t/reading-pixel-size-from-image-file-with-python/74798/20

## Solution
*Please describe your ideal solution*

During initial setup of the `TiffReader` object, process TIFF Tags for additional metadata (ImageJ puts physical pixel sizes in the Tiff Tags for example)

## Alternatives
*Please describe any alternatives you've considered, even if you've dismissed them*

toloudis, 2022-12-09:
Really?  tifffile is usually great about this - we should try to get it from that library since we are using it for everything else.

tlambert03, 2022-12-09:
https://github.com/tlambert03/aicsimageio/tree/tiff-res

evamaxfield, 2022-12-09:
That branch looks good to me. Feel free to open a PR.