AllenCellModeling/aicsimageio #452: Feature: Add units to PhysicalPixelSize tuple?

tlambert03, 2022-12-09:
I found myself wondering what the units of `PhysicalPixelSizes` was.  I see that there's no docstring, and if I look around the repo a bit, it doesn't seem very standardized.  (where we pull from ome-types, including in my bioformats reader, it looks like we pretty much just grab the physical_pixel_size for X and Y, but don't look at the units value there :joy:)  

It would be fine by me either way if `PhysicalPixelSize` was just always assumed to be microns or meters or something (provided it was documented with that assumption and reader plugins knew they should do the conversion)... but we could also have a `unit` member of the named tuple?   Problem of course there is that it's a bit breaking, if anyone did something like `dz, dy, dx = file.physical_pixel_sizes`.  so very open to any other suggestions!

evamaxfield, 2022-12-09:
Last time this was mentioned I think we said: "lets use `pint` in v5" but v5 / bioio isn't anywhere close so maybe we should just rollout a release with `pint` now? Would that solve your problem or would you want the units as an additional property. Both are fine imo.

cc @toloudis

tlambert03, 2022-12-09:
sure, pint is fine, but it's slightly orthogonal.  pint will give you a nice object to do conversions and stuff, but you still need to decide where in the model the new info lives no?  If you want pint to be a breaking change (such that  PhysicalPixelSize   becomes a tuple of pint Quantities instead of floats), then a short-term solution would just be the document-it-better approach, and double check that any of the readers that implement a pixel size are converting to microns (which isn't so hard that it requires the full pint library)

toloudis, 2022-12-09:
In general, I think string unit is better than an object out of some library, if you can get away with it (e.g. if the string can be used to initialize a pint object correctly).  Keeps things as simple as possible.  What does xarray do?

toloudis, 2022-12-09:
as for where in the model, I think it's fine to have a PhysicalPixelSize have new fields called x_unit, y_unit, z_unit... or just unit? Do most file formats tend to assume the same unit for all three?

evamaxfield, 2022-12-09:
Ya not introducing pint is fine with me. If the user can take the units and construct pint objects if they want, thats totally fine.

SeanLeRoy, 2023-11-08:
Moving this issue over to [bioio](https://github.com/bioio-devs/bioio/issues/25). `aicsimageio` is moving into "maintenance" mode where only easy to implement bugfixes deemed critical (or community contributed) work will be done in `aicsimageio`. Instead of `aicsimageio`, we are creating a package soon to be released called bioio. [See the reason for this change here](https://github.com/AllenCellModeling/aicsimageio/discussions/372).

closed 2023-11-08 by SeanLeRoy