AllenCellModeling/aicsimageio #454: XMLSchemaValidationError when reading xml/h5 dataset

grinic, 2022-12-13:
## System and Software
* aicsimageio Version: 4.9.2
* Python Version: Python 3.9.15
* Operating System: Windows 10

## Description
I am trying to open the xml/h5 dataset here:
[xml](https://github.com/NEUBIAS/training-resources/raw/master/image_data/xyzct_16bit__mitosis.xml)
[h5](https://github.com/NEUBIAS/training-resources/raw/master/image_data/xyzct_16bit__mitosis.h5)

using:

```
image = AICSImage(xmlfile, reader= BioformatsReader)
img = image.data
meta = image.metadata
```
and I obtain the following error:
```
---------------------------------------------------------------------------
XMLSchemaValidationError                  Traceback (most recent call last)
Cell In[1], line 2
      1 from load_from_url import load_from_url
----> 2 image_url = load_from_url('https://github.com/NEUBIAS/training-resources/raw/master/image_data/xyzct_16bit__mitosis.xml')

File ~\Documents\image_analysis_training\test_notebook\load_from_url.py:36, in load_from_url(url, verbose)
     34     # load image using AICSImage to retrieve metadata
     35     image = AICSImage(fname, reader=BioformatsReader)
---> 36     _ = image.data # this is to make sure the image is loaded into memory
     38 return image

File ~\AppData\Roaming\Python\Python39\site-packages\aicsimageio\aics_image.py:529, in AICSImage.data(self)
    516 @property
    517 def data(self) -> np.ndarray:
    518     """
    519     Returns
    520     -------
   (...)
    527     Recommended to use `dask_data` for large mosaic images.
    528     """
--> 529     return self.xarray_data.data

File ~\AppData\Roaming\Python\Python39\site-packages\aicsimageio\aics_image.py:469, in AICSImage.xarray_data(self)
    453 """
    454 Returns
    455 -------
   (...)
    462 Recommended to use `xarray_dask_data` for large mosaic images.
    463 """
    464 if self._xarray_data is None:
    465     if (
    466         # Does the user want to get stitched mosaic
    467         self._reconstruct_mosaic
    468         # Does the data have a tile dim
--> 469         and dimensions.DimensionNames.MosaicTile in self.reader.dims.order
    470     ):
    471         try:
    472             self._xarray_data = (
    473                 self._transform_data_array_to_aics_image_standard(
    474                     self.reader.mosaic_xarray_data
    475                 )
    476             )

File ~\AppData\Roaming\Python\Python39\site-packages\aicsimageio\readers\reader.py:532, in Reader.dims(self)
    525 """
    526 Returns
    527 -------
    528 dims: Dimensions
    529     Object with the paired dimension names and their sizes.
    530 """
[... 194 lines cut ...]
    ...
    ...
  </SpimData></Value></XMLAnnotation>

Path: /OME/StructuredAnnotations/XMLAnnotation
```
The same error appears when I try to acces `image.data` and `image.metadata`.

## Expected Behavior
I am not sure if the problem is on the dataset side (XMLAnnotation error is raise), however I am able to open the dataset using `python-bioformats` directly and with the FIJI Bioformats importer.

I am aware of this previous [issue](https://github.com/AllenCellModeling/aicsimageio/issues/195), however I am not sure what should I do to solve my problem. I think I already have the latest version of aicsimageio?

Thank you so much for your help!

## Reproduction
After downloading the xml/h5 dataset above, the error can be reproduced with:
```
from aicsimageio import AICSImage
from aicsimageio.readers import BioformatsReader

xmlfile = 'path-to-xml-file'
image = AICSImage(xmlfile, reader= BioformatsReader)
img = image.data
```

evamaxfield, 2022-12-13:
Sorry to hear you are having an issue! Hopefully we can come up with a fix.

Pinging @tlambert03.

From my view this is a problem with the metadata of the dataset itself, but we have made changes to our metadata processing to allow for "common metadata problems" -- exactly like the mircomanager PR you linked. Do you know how this dataset was created?

grinic, 2022-12-14:
Thanks @evamaxfield for the reply.
The dataset was created using FIJI: `Plugins › BigDataViewer › Export Current Image as XML/HDF5`.
I don't know the file format of the original dataset.

Some additional information that might be useful. I tried:
- open the xml/h5 dataset with `FIJI > Plugins > Bio-Formats > Bio-Formats Importer`
- save image as tif (it then retains all metadata)
- open the tif image and re-save it using `Plugins › BigDataViewer › Export Current Image as XML/HDF5`

At this point, using 
```
image = AICSImage(<tif-file>, reader=BioformatsReader)
print(image.data.shape)
print(image.physical_pixel_sizes)
```
works as expected. However:
```
image = AICSImage(<new-xml-h5-file>, reader=BioformatsReader)
print(image.data.shape)
print(image.physical_pixel_sizes)
```
generates the same `XMLSchemaValidationError` error as before.
I guess this suggests that the problem lies in the way AICSImage reads the output of BigDataViewer xml/h5 export?

tlambert03, 2022-12-15:
Yeah, looks like the issue here is that BigDataViewer is outputting xml that somehow doesn't validate against the OME schema.  This is more of a problem with ome-types than aicsimageio.  We've discussed having more graceful validation error handling there.  I think I can probably get this one loading.  Give me a few days and I'll try to release an update to ome-types that works here.

tlambert03, 2022-12-15:
the core problem here is that, somehow/somewhere, the XMLAnnotation is getting written with an ID of `<XMLAnnotation ID="XMLAnnotation:0">` .... which does not validate against the [AnnotationID regex](https://www.openmicroscopy.org/Schemas/Documentation/Generated/OME-2016-06/ome_xsd.html#AnnotationID) of `((urn:lsid:([\w\-\.]+\.[\w\-\.]+)+:\S+:\S+)|(\S+:\S+)) & ((urn:lsid:([\w\-\.]+\.[\w\-\.]+)+:Annotation:\S+)|(Annotation:\S+))`  (it appears that it should be `Annotation:0`)

I haven't been able to figure out exactly where in the process this is happening yet (i.e. whether it's in the big dataviewer exporter, or the java bioformats reader that merges the xml file with the hdf5 file).  But at least it's clear what the problem is, and we could certainly special case this in `clean_ome_xml_for_known_issues`

tlambert03, 2022-12-15:
@ctrueden ... if I may bother/summon you :)  Maybe you can help me figure this out?

Quick summary:  When I load the h5 file linked in the first post in Fiji with the Bioformats importer, i see this metadata:

[image]

note the id of the XMLAnnotation `<XMLAnnotation ID="XMLAnnotation:0">`, which doesn't validate against the AnnotationID regex in the OME model. The method dumping the xml is `loci.formats.ome.OMEPyramidStore.dumpXML`, but I haven't been able to determine exactly where in the inheritance chain the faulty id is getting written?

*side-note, I will of course fix stuff on my end to fail more gracefully with minor problems like that... just curious if you had run into that ID before*

tlambert03, 2022-12-15:
patch for this special case in #455

grinic, 2022-12-15:
Thanks a lot for the quick help!

Now I get this new error (only pasting the last XMLSchemaValidationError). Seems like the `<XMLAnnotation ID>` is completely missing now:

```
XMLSchemaValidationError: failed validating {} with XsdAttributeGroup(['ID', 'Name', 'SamplesPerPixel', 'IlluminationType', 'PinholeSize', 'PinholeSizeUnit', 'AcquisitionMode', 'ContrastMethod', 'ExcitationWavelength', 'ExcitationWavelengthUnit', 'EmissionWavelength', 'EmissionWavelengthUnit', 'Fluor', 'NDFilter', 'PockelCellSetting', 'Color']):

Reason: missing required attribute 'ID'

Schema:

  <xsd:complexType xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <xsd:sequence>
      <xsd:element ref="LightSourceSettings" minOccurs="0" maxOccurs="1" />
      <xsd:element ref="DetectorSettings" minOccurs="0" maxOccurs="1" />
      <xsd:element ref="FilterSetRef" minOccurs="0" maxOccurs="1" />
      <xsd:element ref="AnnotationRef" minOccurs="0" maxOccurs="unbounded">
        <xsd:annotation>
          <xsd:appinfo>
            <xsdfu>
              <manytomany />
            </xsdfu>
          </xsd:appinfo>
        </xsd:annotation>
      </xsd:element>
      <xsd:element ref="LightPath" minOccurs="0" maxOccurs="1" />
    </xsd:sequence>
    <xsd:attribute name="ID" use="required" type="ChannelID" />
    <xsd:attribute name="Name" use="optional" type="xsd:string">
      <xsd:annotation>
        <xsd:documentation>
    ...
    ...
  </xsd:complexType>

Instance:

  <Channel xmlns="http://www.openmicroscopy.org/Schemas/OME/2016-06">
    <id>1</id>
    <name>1</name>
  </Channel>

Path: /OME/StructuredAnnotations/XMLAnnotation/Value/SpimData/SequenceDescription/ViewSetups/Attributes/Channel[1]
```

Maybe I am doing something wrong? I installed following patch #455 using the fix-xmlannotation-id [branch](https://github.com/tlambert03/aicsimageio/tree/fix-xmlannotation-id) via:
`pip install git+https://github.com/tlambert03/aicsimageio.git@fix-xmlannotation-id`

tlambert03, 2022-12-15:
can you show me the exact code that precedes the error?

grinic, 2022-12-15:
Sure, here it is:

```
---------------------------------------------------------------------------
XMLSchemaValidationError                  Traceback (most recent call last)
Cell In[3], line 1
----> 1 print(image2.data.shape)

File ~\Anaconda3\envs\sk-nap-tut\lib\site-packages\aicsimageio\aics_image.py:529, in AICSImage.data(self)
    516 @property
    517 def data(self) -> np.ndarray:
    518     """
    519     Returns
    520     -------
   (...)
    527     Recommended to use `dask_data` for large mosaic images.
    528     """
--> 529     return self.xarray_data.data

File ~\Anaconda3\envs\sk-nap-tut\lib\site-packages\aicsimageio\aics_image.py:469, in AICSImage.xarray_data(self)
    453 """
    454 Returns
    455 -------
   (...)
    462 Recommended to use `xarray_dask_data` for large mosaic images.
    463 """
    464 if self._xarray_data is None:
    465     if (
    466         # Does the user want to get stitched mosaic
    467         self._reconstruct_mosaic
    468         # Does the data have a tile dim
--> 469         and dimensions.DimensionNames.MosaicTile in self.reader.dims.order
    470     ):
    471         try:
    472             self._xarray_data = (
    473                 self._transform_data_array_to_aics_image_standard(
    474                     self.reader.mosaic_xarray_data
    475                 )
    476             )

File ~\Anaconda3\envs\sk-nap-tut\lib\site-packages\aicsimageio\readers\reader.py:532, in Reader.dims(self)
    525 """
    526 Returns
    527 -------
    528 dims: Dimensions
    529     Object with the paired dimension names and their sizes.
    530 """
    531 if self._dims is None:
--> 532     self._dims = Dimensions(dims=self.xarray_dask_data.dims, shape=self.shape)
    534 return self._dims

File ~\Anaconda3\envs\sk-nap-tut\lib\site-packages\aicsimageio\readers\reader.py:359, in Reader.xarray_dask_data(self)
    352 """
    353 Returns
    354 -------
    355 xarray_dask_data: xr.DataArray
    356     The delayed image and metadata as an annotated data array.
    357 """
    358 if self._xarray_dask_data is None:
--> 359     self._xarray_dask_data = self._read_delayed()
    361 return self._xarray_dask_data

File ~\Anaconda3\envs\sk-nap-tut\lib\site-packages\aicsimageio\readers\bioformats_reader.py:164, in BioformatsReader._read_delayed(self)
    163 def _read_delayed(self) -> xr.DataArray:
--> 164     return self._to_xarray(delayed=True)

File ~\Anaconda3\envs\sk-nap-tut\lib\site-packages\aicsimageio\readers\bioformats_reader.py:202, in BioformatsReader._to_xarray(self, delayed)
    195 with BioFile(
    196     self._path,
    197     series=self.current_scene_index,
[... 202 lines cut ...]
            <xsdfu>
              <manytomany />
            </xsdfu>
          </xsd:appinfo>
        </xsd:annotation>
      </xsd:element>
      <xsd:element ref="LightPath" minOccurs="0" maxOccurs="1" />
    </xsd:sequence>
    <xsd:attribute name="ID" use="required" type="ChannelID" />
    <xsd:attribute name="Name" use="optional" type="xsd:string">
      <xsd:annotation>
        <xsd:documentation>
    ...
    ...
  </xsd:complexType>

Instance:

  <Channel xmlns="http://www.openmicroscopy.org/Schemas/OME/2016-06">
    <id>1</id>
    <name>1</name>
  </Channel>

Path: /OME/StructuredAnnotations/XMLAnnotation/Value/SpimData/SequenceDescription/ViewSetups/Attributes/Channel[1]
```

tlambert03, 2022-12-15:
sorry... I'm looking more for what `image2` was.   Like, which file did you open, (xml or h5), how did you open (bioformats reader vs aicsimage), etc...  also can you give me the output of:

```python
from aicsimageio.readers import BioformatsReader
BioformatsReader.bioformats_version()
```

grinic, 2022-12-15:
Ah, sorry I didn't get it.
I am opening the xml file that I posted in the first post, both xml and h5 files are in the same folder, with AICSImage:
```
from aicsimageio import AICSImage
from aicsimageio.readers import BioformatsReader
fname1 = 'test_dataset\\xyzct_16bit__mitosis.tif'
image1 = AICSImage(fname1, reader=BioformatsReader)
fname2 = 'test_dataset\\xyzct_16bit__mitosis.xml'
image2 = AICSImage(fname2, reader=BioformatsReader)
```
In this case, the tif file was generated opening the xml/h5 dataset with Bioformats Importer in FIJI and saving the dataset as tif.
`print(image1.metadata)` gives the expected result, while `print(image2.metadata)` gives the above error.
The output of the code:
```
from aicsimageio.readers import BioformatsReader
BioformatsReader.bioformats_version()
```
is:
`'6.12.0-SNAPSHOT'`

tlambert03, 2022-12-15:
huh... bummer, I can't reproduce.  will need to try windows and python 3.9