AllenCellModeling/aicsimageio #460: Getting all mosaic position at once

yfukai, 2023-02-15:
## Use Case
For example, when we perform mosaic stitching, we need to read all mosaic positions. 

## Solution
It would be great if we have a function like `get_all_mosaic_tile_position` that returns positions for all mosaic images.

## Alternatives
Currently, if I understand correctly, the only solution is iterating over tile positions as:
```python
# img = AICSImage(image_path,reconstruct_mosaic=False)
mosaic_positions = [list(img.get_mosaic_tile_position(i)) for i in range(img.dims.M)]
```
which took 4min. 10sec. for a 10x10 Zeiss CZI file. However, if I use the API in the `aicspylibczi` as 
```python
with open(image_path) as f:
    czi = CziFile(f)
    bboxes = czi.get_all_mosaic_tile_bounding_boxes()
    bbox = list(bboxes.values())
mosic_positions = [(b.y, b.x) for b in bbox]
```
it finishes in 5sec. 

The above method is CZI-file specific, but considering the use cases, I thought it might make sense to expose a method to get all the positions in a faster way in `AICSImage` object. Looking forward to the feedback!

toloudis, 2023-03-23:
This is a good suggestion. It does seem silly that there's such a big performance difference.   Maybe get_mosaic_tile_position could be extended to accept a range of indices.

I'm curious to know what the dimensions were of the CZI file that took over 4 minutes in your test. (TCZYX and how many M tiles)

yfukai, 2023-03-24:
Hi, thanks @toloudis for your response! I think that fix will help those working on shading correction and stitching workflows. (Maybe the parameter `mosaic_tile_index` can default to `None` which returns all the indices, but I don't have a strong opinion on this.) The dimension for my image was `M=44 T=107 C=3 Z=1 Y=2048 X=2048` with the size of 153GB.

closed 2023-04-18 by SeanLeRoy