AllenCellModeling/aicsimageio #462: OME Zarr metadata files do not have newline characters at the end of files

galenlynch, 2023-03-01:
Low priority, low impact, but I noticed that .zarray, .zattrs, and other OME Zarre metadata files do not have a newline character at the end of files.

toloudis, 2023-03-02:
Interesting. I believe github.com/ome/ome-zarr-py (or something even lower level like zarr) is responsible for writing out the .zattrs data. 
Is there a particular way that you are reading back the data where this causes problems?

galenlynch, 2023-07-03:
No, it does not cause problems for me, I just noticed it.

SeanLeRoy, 2023-11-08:
This doesn't seem to be impacting anyone thus far and is largely a change that would need to be made by the third party library (`ome-zarr`) for the OME ZARR writer. I'll close this for now, but if this still seems important to anyone feel free to re-open it or better yet open an issue [over here](https://github.com/bioio-devs/bioio) in bioio which is the new repository/package soon to be replacing this package.

closed 2023-11-08 by SeanLeRoy