AllenCellModeling/aicsimageio #475: Do something about xarray version issue

SeanLeRoy, 2023-03-24:
Looks like `xarray` dropped support for python 3.8 in 2023.02.0 and our tests are now hitting that issue. What do we want to do here? Options:
* Restrict `xarray` to below that version for all python versions

in new repository (or sooner after the bugfixes) drop support for python3.8

toloudis, 2023-03-24:
@evamaxfield do you have any opinions on this? 
* xarray is still a foundational piece of aicsimageio, but one can use aicsimageio with only dask arrays and numpy arrays and never know xarray was in the mix. 
* And I would guess a lot of workflows using aicsimageio are out there on 3.8.  
* 3.8's EOL is scheduled for Oct 2024.

evamaxfield, 2023-03-24:
> xarray is still a foundational piece of aicsimageio, but one can use aicsimageio with only dask arrays and numpy arrays and never know xarray was in the mix.

Yep. I keep saying to myself that all of aicsimageio in some way can be seen as a fancy wrapper around xarray. Keeping it up-to-date I think is probably pretty important?

> And I would guess a lot of workflows using aicsimageio are out there on 3.8.

I am sure there are, from download stats, we see ~40 downloads of aicsimageio per day: 
[image]

From: https://pypistats.org/packages/aicsimageio

---

I would say if the xarray team made the decision to drop it, we can drop it too.

tlambert03, 2023-03-24:
bummer... imho that's a bit aggressive on xarray's part.

> 3.8's EOL is scheduled for Oct 2024. ( https://endoflife.date/python )

this is true, and I think it's the better support target for most python projects that aren't "too complex" ... but note that Numpy has an accelerated python EOL schedule (which is dropping py3.8 in about a month) and I suspect xarray is paying more attention to that: https://numpy.org/neps/nep-0029-deprecation_policy.html

for something trying to support lots of complicated wheels, I definitely get it... but otherwise I think it's too fast.

If you're going to make xarray a hard dependency though, you sorta have no option.

closed 2023-04-10 by SeanLeRoy