AllenCellModeling/aicsimageio #479: aicsimageio uses ome_types in a way that causes the emission of a warning

anntzer, 2023-04-03:
## System and Software
* aicsimageio Version: 4.9.4 or master
* Python Version: 3.11
* Operating System: macOS

## Description
aicsimageio calls ome_types.from_xml without setting the `parser` argument at https://github.com/AllenCellModeling/aicsimageio/blob/65226e28ae25eb57934c3a54ecfba92c4f2d10a7/aicsimageio/readers/ome_tiff_reader.py#L65-L71 but this now emits a warning on ome_types' side due to https://github.com/tlambert03/ome-types/blob/ae84abff0fa3371e69ce9d7f99d79a8d95c5d0c7/src/ome_types/_convenience.py#L53-L62 (from_xml immediately calls to_dict).

## Expected Behavior
No warning (hopefully, passing `parser` should be sufficient, or I guess you could also pin ome_types to an earlier version if you really prefer...)

## Reproduction
Download tubhiswt_C0.ome.tif from https://downloads.openmicroscopy.org/images/OME-TIFF/2016-06/tubhiswt-2D/ and try to open it with AICSImage(), with aicsimageio installed in a clean environment.

## Environment
Nothing special.

toloudis, 2023-04-03:
Coincidentally, keep an eye on #478 -- we may have this updated soon.

SeanLeRoy, 2023-04-18:
@toloudis Is this done with #478 now?

toloudis, 2023-04-18:
Yes I'm pretty sure but haven't verified this exact example yet

closed 2023-11-29 by SeanLeRoy