AllenCellModeling/aicsimageio #491: does not open the latest leica formats for .lif

ronnkling, 2023-04-20:
### Discussed in https://github.com/AllenCellModeling/aicsimageio/discussions/484

<div type='discussions-op-text'>

<sup>Originally posted by **ronnkling** April  9, 2023</sup>
This would be for LAS X version 3.5.0

ImageJ and Fiji do but I'm looking for a pure python app

I get the error
>>> from aicsimageio import AICSImage
>>> img = AICSImage("E:\Users\klingr\FLIM\Example_FLIM.lif")
  File "<stdin>", line 1
    img = AICSImage("E:\Users\klingr\FLIM\Example_FLIM.lif")
                                                                   ^
SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes in position 2-3: truncated \UXXXXXXXX escape</div>

ronnkling, 2023-06-06:
How do I tell what version I  am running? And did you try the image that I uploaded?

SeanLeRoy, 2023-06-06:
Yep, my tests were with the image you had uploaded in the discussion.

In the environment you run the python command to run your script you can run something like `pip show aicsimageio` and check the version printed or `pip freeze` to check out all the python packages installed currently. Alternatively (or in addition), you can create a new environment using `virtualenv` (or however you want) by running `python3.10 -m venv venv && source venv/bin/activate && pip install aicsimageio==4.11.0` (You'll need to deactivate your current virtual environment if using one and/or select a different python version if you don't have 3.10.

ronnkling, 2023-06-11:
I followed the instructions and got python 3.10.11 and did the pip install like above.  And I get the same error.  This is on a windows 10 machine running the latest anaconda with the python 10 environment activated.
[image]

psobolewskiPhD, 2023-06-12:
I think this might just be a Windows issue due to the backslash character used in paths interacting with Unicode escapes?
Perhaps it's sufficient to use `\\` or use `r` before the quoted string?
https://stackoverflow.com/questions/2953834/how-should-i-write-a-windows-path-in-a-python-string-literal

evamaxfield, 2023-06-12:
Ehhhh I was assuming it was a metadata decode issue. We have tests that run on windows that look just like this type of pathing.

psobolewskiPhD, 2023-06-12:
I dunno, I can't reproduce with the linked file on macOS 🤷 
[image]

toloudis, 2023-06-13:
@ronnkling From your screenshot this definitely looks like a path string problem with backslash.  try
```
img = AICSImage("E:\\Users\\klingr\\FLIM\\Example_FLIM.lif")
```
You definitely need to "escape" the backslashes by doubling them up.

SeanLeRoy, 2023-06-21:
Closing with #517 as a potential follow-up bug

closed 2023-06-21 by SeanLeRoy