AllenCellModeling/aicsimageio #495: Tifffile error when accessing dims attribute

jplumail, 2023-05-12:
## System and Software
* aicsimageio Version: 4.11.0
* Python Version: 3.10.11
* Operating System: Ubuntu on WSL2

## Description
When I open an TIFF image with `AICSImage` and ask for the `dims` attribute, I get an error: `TypeError: unexpected keyword argument: is_mmstack`.

Full Traceback:
```
Traceback (most recent call last):
  File "/home/plumail/aics-bug/bug.py", line 4, in <module>
    print(AICSImage(p).dims)
  File "/home/plumail/miniconda3/envs/aicsimageio2/lib/python3.10/site-packages/aicsimageio/aics_image.py", line 574, in dims
    dims=self.xarray_dask_data.dims, shape=self.shape
  File "/home/plumail/miniconda3/envs/aicsimageio2/lib/python3.10/site-packages/aicsimageio/aics_image.py", line 438, in xarray_dask_data
    and dimensions.DimensionNames.MosaicTile in self.reader.dims.order
  File "/home/plumail/miniconda3/envs/aicsimageio2/lib/python3.10/site-packages/aicsimageio/readers/reader.py", line 532, in dims
    self._dims = Dimensions(dims=self.xarray_dask_data.dims, shape=self.shape)
  File "/home/plumail/miniconda3/envs/aicsimageio2/lib/python3.10/site-packages/aicsimageio/readers/reader.py", line 359, in xarray_dask_data
    self._xarray_dask_data = self._read_delayed()
  File "/home/plumail/miniconda3/envs/aicsimageio2/lib/python3.10/site-packages/aicsimageio/readers/tiff_reader.py", line 459, in _read_delayed
    with TiffFile(open_resource, is_mmstack=False) as tiff:
  File "/home/plumail/miniconda3/envs/aicsimageio2/lib/python3.10/site-packages/tifffile/tifffile.py", line 3142, in __init__
    raise TypeError(f'unexpected keyword argument: {key}')
TypeError: unexpected keyword argument: is_mmstack
```

## Expected Behavior
It should return the Dims object.

## Reproduction
```bash
conda create -n aicsimageio-all-env python=3.10
conda activate aicsimageio-all-env
pip install aicsimageio[all]
```

Run 
```python
from aicsimageio import AICSImage
print(AICSImage("https://people.math.sc.edu/Burkardt/data/tif/at3_1m4_01.tif").dims)"
```

## Environment
The version of `tifffile` installed is `2022.4.8`.

When I install `aicsimageio` without `[all]`, I get `tifffile` version `2023.4.12` and the bug disappears.

jplumail, 2023-05-12:
This line https://github.com/AllenCellModeling/aicsimageio/blob/e0f9b0d47fab90526cde4a23f36ec5d7a9aff393/aicsimageio/readers/tiff_reader.py#L459 is causing trouble.

Tifffile requirements should be over `2023.3.15`: `is_mmstack` was added in this commit https://github.com/cgohlke/tifffile/commit/fc5f5a4eea5d510e7592a84f9c1017265e718cca

Currently the requirements are https://github.com/AllenCellModeling/aicsimageio/blob/e0f9b0d47fab90526cde4a23f36ec5d7a9aff393/setup.py#L111

and when you install `aicsimageio[all]`, it installs `bfio` which requires `tifffile` to be under a certain version https://github.com/AllenCellModeling/aicsimageio/blob/e0f9b0d47fab90526cde4a23f36ec5d7a9aff393/setup.py#L39

which causes the bug !

toloudis, 2023-05-13:
Thanks for the detailed report!  We are looking into this and will try to resolve it soon.

BrianWhitneyAI, 2023-05-25:
Hey, @jplumail! thanks for submitting this. As of tifffile 2023.3.15 the default reading of mmstack images has changed to require the flag is_mmstack and as a result of this, we had to adjust our tiff readers to have a minimum dependency of 2023.3.15.

Unfortunately, our installation of bfio with this version of tifffile breaks tests when dealing with tiled reading because of an issue with bfio backend. I've created an issue for that here: https://github.com/PolusAI/bfio/issues/42#issue-1726419814. 

Once that is resolved we can unpin tifffile for bfio and  aicsimageio[all] will work smoothly again, In the meantime you may need to manually decide which version to use depending on what reader you are looking to use. let me know if you have any questions!

jplumail, 2023-05-30:
I've been using `aicsimageio` without `[all]` since then. Thanks!

jplumail, 2023-09-25:
It seems that bfio is now fixed: https://github.com/PolusAI/bfio/issues/42

toloudis, 2023-09-25:
Thanks for the heads-up on this!

SeanLeRoy, 2023-11-20:
Closing now that `bfio` no longer specifies a `tifffile` version directly so `aicsimageio`'s ceiling no longer conflicts

jplumail, 2023-11-22:
The bug is still there !

I did a PR to fix it: #547

PR #547 (2024-03-06): Bugfix/Bfio unpin tifffile
## Description

resolves #495

bfio was fixed in version 2.3.1

## Pull request recommendations:
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [gh-<number>], adds tiff file format support_
- [ ] Provide relevant tests for your feature or bug fix.
- [ ] Provide or update documentation for any feature added by your pull request.

Thanks for contributing!