AllenCellModeling/aicsimageio #504: OmeZarrReader only has "public data or supply your own creds in env"

BrianWhitneyAI, 2023-05-25:
There may be problems with this... I think this currently only works if the image is public read? The `_path` is decoupled from the `_fs` itself and if the user needs to provide credentials, then this won't work.

UNLESS, the user has the credentials in the environment. It might be good to note that the ZarrReader needs env creds and not provided creds.

_Originally posted by @evamaxfield in https://github.com/AllenCellModeling/aicsimageio/pull/493#discussion_r1204547369_

BrianWhitneyAI, 2023-06-08:
In the current state, I confirmed that the OmeZarrReader can read from all three of the following. Local file, Public URI, Private bucket (S3) without additional credentials.

SeanLeRoy, 2023-06-12:
>Private bucket (S3) without additional credentials.

By this do you mean that it reads your aws credentials automatically or that it is somehow authenticating without any credentials at all?

BrianWhitneyAI, 2023-06-12:
> > Private bucket (S3) without additional credentials.
> 
> By this do you mean that it reads your aws credentials automatically or that it is somehow authenticating without any credentials at all?

That it read aws credentials automatically. I tested against the resources in the aicsimageio test resources folder without any issue or additional creds needed.

closed 2023-06-13 by BrianWhitneyAI