AllenCellModeling/aicsimageio #508: channel-z ordering problem when reading lif

TomPJWyatt, 2023-05-26:
## System and Software
* aicsimageio Version: 4.11.0 
* Python Version: 3.9.16
* Operating System: OS X Ventura 13.3.1

I have a lif with e.g. 5 channels and 27 z-slices, it opens fine as a hyperstack in fiji but when I open with aicsimageio the channels and z are all mixed up.

SeanLeRoy, 2023-05-26:
Hi @TomPJWyatt, do you have an example file we could test with? I'd also be interested in knowing how you are initializing the image, like if perhaps supplying the `LifReader` directly may change anything (like so)

```
from aicsimageio import AICSImage
from aicsimageio.readers import LifReader

img = AICSImage(filepath, reader=LifReader)
img.dims
img.shape

img = AICSImage(filepath)
img.dims
img.shape
```

TomPJWyatt, 2023-05-26:
Hi Sean, I have an example file but github won't allow uploading of lif files, any ideas?

I just tried your suggestiong with supplying the LifReader directly and the channels/z are still mixed up I'm afraid.

toloudis, 2023-05-26:
@TomPJWyatt Maybe you can use dropbox, or other cloud file storage that allows link sharing.  We can just download the file and then you could remove it once we confirm we got it.

SeanLeRoy, 2023-06-08:
I am able to reproduce this with your file; other LIF files I have on hand seem to not have this issue. I'm able to get it to read correctly using a change to the way the xy dimensions are ordered together, but it doesn't seem to solve the issue consistently nor am I sure how to flag files like yours as needing to be arranged differently (compared to the other LIF files I have for example) such that if that fix worked we could apply it when needed.

That is all to say this might not be an issue with `aicsimageio` per se. Was this file saved using the LAS X Stage Navigator? This _might_ be a known issue with `readlif` the underlying package responsible for reading LIF files in `aicsimageio` as described [here](https://github.com/nimne/readlif/issues/19#issuecomment-939978307). I'll keep poking around a bit longer to see if I can find something to assist this.

TomPJWyatt, 2023-06-12:
Hi @SeanLeRoy yes it was saved with LAS X Navigator. Thanks for the update!

SeanLeRoy, 2023-11-08:
Moving this issue over to [bioio-lif](https://github.com/bioio-devs/bioio-lif/issues/8). `aicsimageio` is moving into "maintenance" mode where only easy to implement bugfixes deemed critical (or community contributed) work will be done in `aicsimageio`. Instead of `aicsimageio`, we are creating a package soon to be released called bioio and plugins like the one aforementioned. [See the reason for this change here](https://github.com/AllenCellModeling/aicsimageio/discussions/372).

closed 2023-11-08 by SeanLeRoy