AllenCellModeling/aicsimageio #513: czi reader is very slow to get channel_names for a large file

toloudis, 2023-06-12:
## System and Software
* aicsimageio Version: 4.11.0
* Python Version: 3.10.x
* Operating System: Windows (reading from nfs)

## Description
Loading a large time series czi file (TCZYX = about 1000,1,150,2000,1000) simply calling img.channel_names takes well over a minute.  If this info can come from simple metadata then why do we have to do the slow code path for it?

## Expected Behavior
Getting channel names should be fairly quick if they are embedded in metadata that is quick to read/parse.  Getting the number of scenes is fast.

## Reproduction
AICS internal example: assay-dev\\MicroscopyData\\Leveille\\2023\\20230425\\20230425_L02-01_processed.czi

toloudis, 2023-07-07:
I just did a repro on a Windows machine connected to the onprem storage and the call to channel_names took about 3.3-3.6 seconds.  I think that is much more reasonable although still slow-ish.  I think we can close this out.

SeanLeRoy, 2023-07-25:
Closed per @toloudis update (also reproduced on my Mac & Linux)

closed 2023-07-25 by SeanLeRoy