AllenCellModeling/aicsimageio #515: dtype error converting Olympus VSI

dyf, 2023-06-13:
## System and Software
* aicsimageio Version: 4.10.0
* Python Version: 3.11.4
* Operating System: Centos 7

## Description

I am converting an Olympus Slide Scanner VSI file to OME-TIFF. The code is quite simple:

`AICSImage.open('/path/to/file.vsi').save(output_path)`

I get the following error:
```python
Traceback (most recent call last):
  File "/home/david.feng/projects/enhancer_conversion/convert_aics.py", line 7, in <module>
    img.save(output_path)
  File "/home/david.feng/micromamba/envs/aics/lib/python3.11/site-packages/aicsimageio/aics_image.py", line 966, in save
    OmeTiffWriter.save(
  File "/home/david.feng/micromamba/envs/aics/lib/python3.11/site-packages/aicsimageio/writers/ome_tiff_writer.py", line 247, in save
    ome_xml = OmeTiffWriter.build_ome(
              ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/david.feng/micromamba/envs/aics/lib/python3.11/site-packages/aicsimageio/writers/ome_tiff_writer.py", line 647, in build_ome
    img = OmeTiffWriter._build_ome_image(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/david.feng/micromamba/envs/aics/lib/python3.11/site-packages/aicsimageio/writers/ome_tiff_writer.py", line 498, in _build_ome_image
    type=utils.dtype_to_ome_type(data_dtype),
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/david.feng/micromamba/envs/aics/lib/python3.11/site-packages/aicsimageio/metadata/utils.py", line 537, in dtype_to_ome_type
    raise ValueError(f"Ome utils can't resolve pixel type: {npdtype.name}")
ValueError: Ome utils can't resolve pixel type: uint16
```

## Expected Behavior

File generated without error.

## Reproduction
*A minimal example that exhibits the behavior.*

```python
import aicsimageio 

slide_path = "/allen/aind/stage/enhancer_characterization/SlideScanner_619290/_03.vsi"
output_path = "/allen/aind/scratch/david.feng/test.ome.tiff"

img = aicsimageio.AICSImage(slide_path)
img.save(output_path)
```

dyf, 2023-06-14:
A little more info: I updated to `tifffile==2023.4.12`, `aicsimageio==4.11.0`, and `bioformats_jar==2020.5.27`, still no luck.

toloudis, 2023-06-14:
I opened the test file from S3 and it worked for me on Windows, python 3.10, tifffile==2023.4.12, aicsimageio==4.11.0, and bioformats_jar==2020.5.27.
I will give it a try from linux and see what happens.

toloudis, 2023-06-15:
```
(aicsimageio) danielt@dev-aics-dtp-001:~/git/aicsimageio/scripts$ ipython
Python 3.10.11 (main, Apr 20 2023, 19:02:41) [GCC 11.2.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.12.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import aicsimageio

In [2]: img = aicsimageio.AICSImage("s3://aind-msma-morphology-data/test_data/enhancer_characterization/SlideScanner_619
   ...: 290/_03.vsi")

In [3]: img.save("./test.ome.tiff")
/home/danielt/anaconda3/envs/aicsimageio/lib/python3.10/site-packages/ome_types/_convenience.py:112: FutureWarning: The default XML parser will be changing from 'xmlschema' to 'lxml' in version 0.4.0.  To silence this warning, please provide the `parser` argument, specifying either 'lxml' (to opt into the new behavior), or'xmlschema' (to retain the old behavior).
  d = to_dict(os.fspath(xml), parser=parser, validate=validate)

In [4]:
```

dyf, 2023-06-16:
@toloudis maybe a python 3.11 thing? let me downgrade to 3.10.

toloudis, 2023-06-22:
I will try one more thing: running from CentOS + use the aind mount, which is probably the last thing I didn't exactly reproduce.  Otherwise all attempts have been unable to repro (even using the aind mount from ubuntu).  The only other thing I can think to do is compare our `pip freeze` lists.

dyf, 2023-06-24:
@toloudis I just poked around here: 
https://github.com/AllenCellModeling/aicsimageio/blob/7e537c8ea650dda6c3cf3e3d2b53eb54fad3fd7f/aicsimageio/metadata/utils.py#L504-L538

In my current environment, when I point to the network filesystem, I notice that the `npdtype` variable that is coming into the function is not the correct type.

```python
print(npdtype, np.dtype(np.uint16))
# >u2 uint16
print(np.dtype(np.uint16) == npdtype)
# False
print(np.dtype(np.uint16) == np.dtype(npdtype))
# False
print(np.dtype(np.uint16) == npdtype.type)
# True
```

This smells a bit like what is being discuss here on SO: https://stackoverflow.com/questions/62646573/np-uint16-isnt-the-same-as-np-uint16

But I can't explain why it doesn't repro for you.

SeanLeRoy, 2023-11-29:
@dyf Is this still blocking you from using `aicsimageio`?  `aicsimageio` is moving into maintenance mode with `bioio` as its replacement package (same features, different installation) and are only applying critical bugfixes to `aicsimageio` due to this.  If it is still blocking you lets try again with `bioio` and make a new ticket linking this one.

closed 2023-11-29 by SeanLeRoy