AllenCellModeling/aicsimageio #517: same Example_FLIM.lif has 32 bit images in it that aicsimageio cannot read

ronnkling, 2023-06-17:
It handles the 8 and 16 bit OK
(base) PS E:\users\klingr\IDL_nih\readlif-0.6.5\readlif\testingLIFreader> conda activate py310
(py310) PS E:\users\klingr\IDL_nih\readlif-0.6.5\readlif\testingLIFreader> python
Python 3.10.11 | packaged by Anaconda, Inc. | (main, Apr 20 2023, 18:56:50) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> lazy_t0 = img.get_image_dask_data("CZYX", T=0)  # returns out-of-memory 4D dask array
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'img' is not defined
>>> lazy_t0 = img.get_image_dask_data("CZYX", T=0)  # returns out-of-memory 4D dask array
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'img' is not defined
>>> from aicsimageio import AICSImage
>>>  img = AICSImage('Example_FLIM.lif')
  File "<stdin>", line 1
    img = AICSImage('Example_FLIM.lif')
IndentationError: unexpected indent
>>> img = AICSImage('Example_FLIM.lif')
>>> img.scenes
('M1 F1', 'M1 F1/FLIM/Intensity', 'M1 F1/FLIM/Fast Flim', 'M1 F1/FLIM/Standard Deviation', 'M1 F1/FLIM/Pattern Matching Scatter Plot Channel 1', 'M1 F1/FLIM/Phasor Real', 'M1 F1/FLIM/Phasor Imaginary', 'M1 F1/FLIM/Phasor Intensity', 'M1 F1/FLIM/Phasor Real Filtered', 'M1 F1/FLIM/Phasor Imaginary Filtered', 'M1 F1/FLIM/Phasor Mask', 'M1 F1/FLIM/Phasor Plot', 'M1 F2', 'M1 F2/FLIM/Intensity', 'M1 F2/FLIM/Fast Flim', 'M1 F2/FLIM/Standard Deviation', 'M1 F2/FLIM/Pattern Matching Scatter Plot Channel 1', 'M1 F2/FLIM/Phasor Real', 'M1 F2/FLIM/Phasor Imaginary', 'M1 F2/FLIM/Phasor Intensity', 'M1 F2/FLIM/Phasor Real Filtered', 'M1 F2/FLIM/Phasor Imaginary Filtered', 'M1 F2/FLIM/Phasor Mask', 'M1 F2/FLIM/Phasor Plot', 'M1 F3 -- fiber moved', 'M1 F3 -- fiber moved/FLIM/Intensity', 'M1 F3 -- fiber moved/FLIM/Fast Flim', 'M1 F3 -- fiber moved/FLIM/Standard Deviation', 'M1 F3 -- fiber moved/FLIM/Pattern Matching Scatter Plot Channel 1', 'M1 F4-5', 'M1 F4-5/FLIM/Intensity', 'M1 F4-5/FLIM/Fast Flim', 'M1 F4-5/FLIM/Standard Deviation', 'M1 F4-5/FLIM/Pattern Matching Scatter Plot Channel 1', 'Series005', 'Series005/FLIM/Intensity', 'Series005/FLIM/Fast Flim', 'Series005/FLIM/Standard Deviation', 'Series005/FLIM/Pattern Matching Scatter Plot Channel 1', 'M1 F6', 'M1 F6/FLIM/Intensity', 'M1 F6/FLIM/Fast Flim', 'M1 F6/FLIM/Standard Deviation', 'M1 F6/FLIM/Pattern Matching Scatter Plot Channel 1', 'M1 F7-8-9', 'M1 F7-8-9/FLIM/Intensity', 'M1 F7-8-9/FLIM/Fast Flim', 'M1 F7-8-9/FLIM/Standard Deviation', 'M1 F7-8-9/FLIM/Pattern Matching Scatter Plot Channel 1', 'M2 F1-2-3', 'M2 F1-2-3/FLIM/Intensity', 'M2 F1-2-3/FLIM/Fast Flim', 'M2 F1-2-3/FLIM/Standard Deviation', 'M2 F1-2-3/FLIM/Pattern Matching Scatter Plot Channel 1', 'M2 F4-5', 'M2 F4-5/FLIM/Intensity', 'M2 F4-5/FLIM/Fast Flim', 'M2 F4-5/FLIM/Standard Deviation', 'M2 F4-5/FLIM/Pattern Matching Scatter Plot Channel 1', 'M2 F6-7', 'M2 F6-7/FLIM/Intensity', 'M2 F6-7/FLIM/Fast Flim', 'M2 F6-7/FLIM/Standard Deviation', 'M2 F6-7/FLIM/Pattern Matching Scatter Plot Channel 1', 'M2 F8', 'M2 F8/FLIM/Intensity', 'M2 F8/FLIM/Fast Flim', 'M2 F8/FLIM/Standard Deviation', 'M2 F8/FLIM/Pattern Matching Scatter Plot Channel 1', 'M2 F8/FLIM/Phasor Real', 'M2 F8/FLIM/Phasor Imaginary', 'M2 F8/FLIM/Phasor Intensity', 'M2 F8/FLIM/Phasor Real Filtered', 'M2 F8/FLIM/Phasor Imaginary Filtered', 'M2 F8/FLIM/Phasor Mask', 'M2 F8/FLIM/Phasor Plot', 'M2 F8/FLIM/Phasor Plot Region 1', 'M2 F9', 'M2 F9/FLIM/Intensity', 'M2 F9/FLIM/Fast Flim', 'M2 F9/FLIM/Standard Deviation', 'M2 F9/FLIM/Pattern Matching Scatter Plot Channel 1', 'M2 F9/FLIM/Phasor Real', 'M2 F9/FLIM/Phasor Imaginary', 'M2 F9/FLIM/Phasor Intensity', 'M2 F9/FLIM/Phasor Real Filtered', 'M2 F9/FLIM/Phasor Imaginary Filtered', 'M2 F9/FLIM/Phasor Mask', 'M2 F9/FLIM/Phasor Plot', 'M2 F10', 'M2 F10/FLIM/Intensity', 'M2 F10/FLIM/Fast Flim', 'M2 F10/FLIM/Standard Deviation', 'M2 F10/FLIM/Pattern Matching Scatter Plot Channel 1')
>>> img.set_scene(1)
>>> img.current_scene
'M1 F1/FLIM/Intensity'
>>> img.data
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\klingr\anaconda3\envs\py310\lib\site-packages\aicsimageio\aics_image.py", line 542, in data
    return self.xarray_data.data
  File "C:\Users\klingr\anaconda3\envs\py310\lib\site-packages\aicsimageio\aics_image.py", line 482, in xarray_data
    and dimensions.DimensionNames.MosaicTile in self.reader.dims.order
  File "C:\Users\klingr\anaconda3\envs\py310\lib\site-packages\aicsimageio\readers\reader.py", line 532, in dims
    self._dims = Dimensions(dims=self.xarray_dask_data.dims, shape=self.shape)
  File "C:\Users\klingr\anaconda3\envs\py310\lib\site-packages\aicsimageio\readers\reader.py", line 359, in xarray_dask_data
    self._xarray_dask_data = self._read_delayed()
  File "C:\Users\klingr\anaconda3\envs\py310\lib\site-packages\aicsimageio\readers\lif_reader.py", line 484, in _read_delayed
    image_data = self._create_dask_array(lif, dims)
  File "C:\Users\klingr\anaconda3\envs\py310\lib\site-packages\aicsimageio\readers\lif_reader.py", line 317, in _create_dask_array
    sample_plane = np.asarray(selected_scene.get_frame())
  File "C:\Users\klingr\anaconda3\envs\py310\lib\site-packages\readlif\reader.py", line 380, in get_frame
    return self._get_item(item_requested)
  File "C:\Users\klingr\anaconda3\envs\py310\lib\site-packages\readlif\reader.py", line 182, in _get_item
    raise ValueError("Unknown bit-depth, please submit a bug report"
ValueError: Unknown bit-depth, please submit a bug report on Github
>>>

ronnkling, 2023-06-21:
is there a plan as to when this will be fixed?

SeanLeRoy, 2023-06-22:
I believe [the issue you created](https://github.com/nimne/readlif/issues/47) in the `readlif` repository is the main path forward, however since it seems that package hasn't been developed on in 2 years it is unknown when/if it will be fixed. Our other options are to use another python LIF reader, [perhaps this one](https://github.com/yangyushi/explore_lif) (or any you might know of), but I'm unsure if it necessarily be any better than `readlif`.

SeanLeRoy, 2023-11-08:
Moving this issue over to [bioio-lif](https://github.com/bioio-devs/bioio-lif). `aicsimageio` is moving into "maintenance" mode where only easy to implement bugfixes deemed critical (or community contributed) work will be done in `aicsimageio`. Instead of `aicsimageio`, we are creating a package soon to be released called bioio and plugins like the one aforementioned. [See the reason for this change here](https://github.com/AllenCellModeling/aicsimageio/discussions/372).

closed 2023-11-08 by SeanLeRoy