AllenCellModeling/aicsimageio #518: Latest version of aicsimageio cannot read some czi images

christinab12, 2023-06-22:
Hi there, 

## System and Software
* aicsimageio Version: 4.11.0
* Python Version: 3.9
* Operating System: MacOS

## Description
With the new version of aicsimageio I cannot open _some_ czi files, while others work (I still haven't figured out what distinguishes between the ones that open and the ones that don't).

## Expected Behavior
I'm using napari-aicsimageio for my own napari plugin, which is dependent on aicsimageio. With v.4.9.4 when I drag and drop these czi files they are read normally. 

## Reproduction
`pip install napari-organoid-counter`
Open napari. 
Drag and drop czi file.

Unfortunately, I cannot share the data, but this is the relevant part of the error:
`~/opt/anaconda3/envs/noc-test/lib/python3.9/site-packages/napari_aicsimageio/core.py in ?(path='../merged_stack_B2.czi', in_memory=None)
    238 
    239     # Open file and get data
--> 240     img = AICSImage(path)
    241 

~/opt/anaconda3/envs/noc-test/lib/python3.9/site-packages/aicsimageio/aics_image.py in ?(self=<class 'aicsimageio.aics_image.AICSImage'> instance, image='../merged_stack_B2.czi', reader=None, reconstruct_mosaic=True, fs_kwargs={}, **kwargs={})
    281 
    282         # Init and store reader
--> 283         self._reader = ReaderClass(image, fs_kwargs=fs_kwargs, **kwargs)
    284 

~/opt/anaconda3/envs/noc-test/lib/python3.9/site-packages/aicsimageio/readers/ome_tiff_reader.py in ?(self=<OmeTiffReader [Image-is-in-Memory: False]>, image='../merged_stack_B2.czi', chunk_dims=['Z', 'Y', 'X', 'S'], clean_metadata=True, fs_kwargs={}, **kwargs={})
    200                 # many series
    201                 if tiff.is_micromanager and not isinstance(self._fs, LocalFileSystem):
--> 202                     log.warning(
    203                         "**Remote reading** (S3, GCS, HTTPS, etc.) of multi-image "

~/opt/anaconda3/envs/noc-test/lib/python3.9/site-packages/tifffile/tifffile.py in ?(self=<class 'tifffile.tifffile.TiffFile'> instance, arg=<fsspec.implementations.local.LocalFileOpener object>, mode=None, name=None, offset=None, size=None, _multifile=True, _useframes=None, _parent=None, **kwargs={'is_mmstack': False})
   3140                         setattr(self, key, bool(value))
   3141                 else:
-> 3142                     raise TypeError(f'unexpected keyword argument: {key}')
   3143 

TypeError: unexpected keyword argument: is_mmstack
`

toloudis, 2023-06-22:
the is_mmstack indicates that you probably need to install the latest `tifffile` version.

christinab12, 2023-06-27:
Thanks @toloudis @SeanLeRoy for your fast replies.

I see now, I think then the problem could then be that I am, in a previous step, saving this image as a tiff and then reading and converting to a czi file using the following code snippet:
```
from aicsimageio.writers.ome_tiff_writer import OmeTiffWriter
from aicsimageio import AICSImage
from aicsimageio import types

# code where merged_stack numpy array is created and xres, yres are computed

OmeTiffWriter.save(merged_stack, 
                            "merged_stack.tiff",
                            dim_order="YX",
                            physical_pixel_sizes=types.PhysicalPixelSizes(None, yres, xres))
AICSImage("merged_stack.tiff").save("merged_stack.czi")
```

I tried what you suggested above:
* For both ```4.9.4``` and ```4.11``` this:
```
from aicsimageio import AICSImage, readers
img=AICSImage("merged_stack.czi", reader=readers.CziReader)
```
gives the error ```aicsimageio.exceptions.UnsupportedFileFormatError: CziReader does not support the image```
* But in ```4.9.4``` this works:
```
from aicsimageio import AICSImage, readers
img=AICSImage("merged_stack_C4.czi")
```
* And in  ```4.11``` the same command gives again the ```mmstack``` error.
As you suggested @SeanLeRoy, raising to the latest ```tifffile``` version works, albeit the OME TIFF reader is being used to read the file instead of the CZI reader!

closed 2023-07-04 by christinab12